espressif/esp-box #6: [Schematic] Typo on the IMU MPN in the schematics

pedrominatel, 2021-12-06:
The part number in the schematics is wrong. The correct is ICM-42670-P and, not ICM-42607-P.

[image]

ESP-Mars, 2021-12-07:
Mass production used ICM-42607-P, ICM-42670-P and ICM-42607-P are pin to pin. The driver of it is developing now.

pedrominatel, 2021-12-07:
Thank you @ESP-Mars!

closed 2021-12-07 by pedrominatel

nishit-jain-esp, 2021-12-07:
If that is the case, then the name of the chip should be consistent on the first page of the [schematic](https://github.com/espressif/esp-box/blob/master/hardware/esp32_s3_box_v2.5/schematic/SCH_ESP32-S3-BOX_V2.5_20211011.pdf) and page 6. Also, the driver name in [code](https://github.com/espressif/esp-box/tree/master/components/i2c_devices/icm42670) should be the same as the actual chip being used.

leeebo, 2021-12-09:
Fixed #6 in [b41dbb307](https://github.com/espressif/esp-box/tree/b41dbb307dfebc06879695b55d25352cedd5e095)