espressif/esp-box #7: Porting from S3-Box LVGL Demo to custom board

tvanfossen, 2021-12-06:
Hi,

I have a custom LCD board with an ESP32S3 using LVGL. Previously, I had been building with an ESP32.

I was wondering what it would take to port the LVGL demos from the S3Box to this custom LCD.

The custom board uses an ST7789 display driver. Resolution is the same 320x240.

Pinout for my board is:
MOSI - 9
MISO - NC
SCLK - 19
CS - 47
DC - 16
RST - 17

I've compiled it, but cpu will not start as expected. I've also tried making the necessary changes from the original ESP32 project I've been working on, but to no avail (Boots as expected and starts on its main loop, but doesnt display to screen with the below pin changes)

I've adjusted, in the SDKCONFIG, the HMI Board Config such that:
HMI Board -> Custom HMI Board
Select LCD Interface -> LCD_INTERFACE_SPI
I2C GPIO Config (Unchanged)
SPI GPIO Config ->
    SPI CS -> 47
    SPI MISO -> 0 (NC?)
    SPI MOSI -> 9
    SPI SCLK -> 19
    
In esp32_s3_box.h:
#define GPIO_LCD_BL     (GPIO_NUM_NC)
#define GPIO_LCD_BL_ON  (1)
#define GPIO_LCD_CS     (GPIO_NUM_47)
#define GPIO_LCD_RST    (GPIO_NUM_17)
#define GPIO_LCD_DC     (GPIO_NUM_16)
#define GPIO_LCD_CLK    (GPIO_NUM_19)
#define GPIO_LCD_DIN    (GPIO_NUM_9)
#define GPIO_LCD_DOUT   (GPIO_NUM_NC)

Notably, when running the esp-box demo, I do not see the statement below printed.
(In doc)
[image]
(idf.py monitor)
[image]

When running the ESP32S3 on the previous repo I was using for esp32, I am able to pass this point without issue

Any suggestions would be much appreciated!

tvanfossen, 2021-12-06:
Ah didnt notice the differentiation in box vs custom board - will try setting things there

tvanfossen, 2021-12-06:
Still no luck with the values set in custom_board.h

tvanfossen, 2021-12-06:
Had a couple errors pop up around the inits - I disabled bsp_tp_init() (no touch on my board), and needed to set CONFIG_ESP_CONSOLE_UART_NUM to UART0 (was set to -1)

And the display is now working.

Real-YuZhe, 2021-12-08:
> Had a couple errors pop up around the inits - I disabled bsp_tp_init() (no touch on my board), and needed to set CONFIG_ESP_CONSOLE_UART_NUM to UART0 (was set to -1)
> 
> And the display is now working with one caveat.
> 
> Display is inverted, (color inversion is already set), in previous version of LVGL UIs, there was an option in the SDKCONFIG to set Portrait Inverted/Landscape Inverted as an option to fix this specific issue, but I dont see any equivalent setting for this set of drivers

1. Check for `esp_lcd_panel_invert_color(panel_handle, LCD_COLOR_INV);`
2. Check for `color_space` of `esp_lcd_panel_dev_config_t`.
3. Byte swap is required and make sure it's open in menuconfig.

tvanfossen, 2021-12-08:
All resolved, screen looks good and drivers are working as intended

Has any demo been done on the lottie lib in LVGL? Would love to see that running on an S3, but cant seem to compile with that option enabled

tvanfossen, 2021-12-08:
Closing this issue in favor of #8

closed 2021-12-08 by tvanfossen