espressif/esp-box #8: Compiling with Lottie Library enabled in LVGL configuration

tvanfossen, 2021-12-08:
### Describe the bug
Compiling examples/rlottie fails x

### To Reproduce
Compile latest LVGL on master with lottie library enabled in sdkconfig
lv_lib_rlottie added as a submodule under lvgl/lv_lib/lv_lib_rlottie

LVGL otherwise works as intended, but havent been able to get it to compile with the lottie option in the sdkconfig. Very interested in trying on the S3 specifically to see how large of a lottie animation can play on a 240x320 SPI TFT (ESP32 was not performant enough to play larger than 120x120 animations)

### Expected behavior
Will not compile

rlottie_capi.h is a dependency from the full rlottie lib (also in lv_lib_rlottie)

### Screenshots or video
[image]

tvanfossen, 2021-12-08:
Wanted to note that lv_lib_rlottie will compile and run on an ESP32

https://github.com/lvgl/lv_lib_rlottie/issues/3

tvanfossen, 2021-12-08:
@kisvegabor

tvanfossen, 2021-12-08:
https://docs.lvgl.io/master/libs/rlottie.html

Will try following this

tvanfossen, 2021-12-09:
[image]
[image]

Took some hacky fixes, but full screen (240x320) lottie file running on ESP32S3

closed 2021-12-09 by tvanfossen

X-Ryl669, 2021-12-09:
What is the difference so important between the ESP32 and the ESP32-S3 ? 

Is it just the LX7 vs LX6 architecture or is the LCD in your ESP-S3 board using multiple SPI lines (DIO/QIO/8IO) so transfers are faster ?
Or maybe you are using the DMA engine ?

tvanfossen, 2021-12-09:
Lottie files, per discussion in lv_lib_rlottie, consume a significant amount of memory in building the image buffer which is then used by rlottie to draw into (not discounting the amount of memory that then gets used by rlottie while drawing the buffer).

The ESP32-Wrover has 4MB of PSRAM - this is enough to be able to load a 240x320 lottie file, but is in no way performant (at or below 1 FPS).

LCD shown above vs early working lottie demos in lv_lib_rlottie is effectively the same. Single channel SPI interface to the display. A parallel interface (such as the one used in the ESP-BOX) could potentially get the lottie files running even smoother, but at the cost of IO (project dependent as to whether its affordable). I'd be intrigued to compare performance to an actual ESP-BOX devkit with the parallel interface

I expect the LX7 is fundamentally improved over LX6, but the specific part (my belief) the ESP32-S3 has over the ESP32 (depending on config) is an octal spi interface to PSRAM, enabling significantly faster comms than on the WROVER to the same block of PSRAM - I believe this is one of the key enabling differences between the ESP32 and S3 in getting a full screen lottie animation to play.

The ESP32, without PSRAM, doesnt have enough available memory to allocate the full buffer for a 240x320 lottie animation. 240x320x32/8 == ~300kb. DRAM segment ~256kb, before any chunks start to get eaten by freertos.

Another possibility - rlottie uses a signifcant number of floating point calculations (vector graphics after all). ESP32, from my reading, uses a VFPU, documentation for the S3 seemed to point to this VFPU being improved over the ESP32, but I dont have hard numbers to back that up.

X-Ryl669, 2022-03-03:
For reference, here's a benchmark of ESP32s3 vs ESP32 on highly optimized code:
https://docs.espressif.com/projects/esp-dsp/en/latest/esp-dsp-benchmarks.html

For the PSRAM, if I follow what Espressif is saying, it shouldn't matter since the PSRAM loads 32kB in the CPU cache, and works from here. But, since the rlottie code is using a lot of classes containing pointers in their members, it might cross the 32kB boundary and cause the PSRAM to swap. Hard to know without measuring...

I'll try to use the IRAM area for the rendering buffer, since, at least this one, is allocated from LVGL, access are 32 bits aligned and we have plenty left usually. On my ESP32 system, I couldn't get more than 12 fps on such animation, and I can't transfer the whole screen to the SPI LCD that fast. Even scrolling is slow on my system.

tvanfossen, 2022-03-03:
Above link is accounting for 02 optimization, I have been unable to get rlottie to compile outside of debug compiler settings - is this something that you fixed in your fork of rlottie?

I have not measured if it goes over 32kb, but would not surprise me on larger animations especially (maybe this is where smaller animations were performing better than larger ones?)

[image]
This is  a small change I add to lv_rlottie to move the lottie buffer into SPIRAM, rather than use lv_mem_alloc. 
[image]

I assume the test on ESP32 with IRAM was using your partial rendering PR? I'd be curious to try this on an S3

Fwiw, I just got an S3 with 2mb psram and quad spi (vs 8mb psram and octal spi) yesterday. The octal spi part is performant and plays at 28-33 fps with 240x320 animations, the quad spi part is dropping to around 15-20fps (same code between the two, aside from the change in sdkconfig for quad spi vs octal spi). I haven't taken hard numbers, but I think this starts to support the improvement that octal vs quad vs single spi on PSRAM can have.

Also would note that both the octal/quad chips are both connecting to a LAN and able to hit http end points for those devices, something that I was unable to ever get running alongside lottie on the ESP32 due to memory issues

X-Ryl669, 2022-03-12:
I've run a simple test on a ESP32 (first version) with different rendering buffers (either SPIRAM, IRAM or DRAM) and here are the results:
```
Bench psram
---------------------
Value =   1359523, select =  0, mask = 0001.  Counts cycles.
                  Amount of cycles
Value =    268212, select =  2, mask = 8dff.  Successfully Retired Instructions.
                  JX instructions
                  CALLXn instructions
                  return instructions (RET, RETW, ...)
                  supervisor return instructions (RFDE, RFE, RFI, RFWO, RFWU)
                  Conditional branch instructions where execution
                  transfers to the target (aka. taken branch),
                   or loopgtz/loopnez instr where execution skips
                   the loop (aka. not-taken loop)
                  J instr
                  CALLn instr
                  Conditional branch instr where execution
                   falls through (aka. not-taken branch)
                  Loop instr where execution falls into loop (aka. taken loop)
                  Last inst of loop and execution transfers
                   to LBEG (aka. loopback taken)
                  Last inst of loop and execution falls 
                   through to LEND (aka. loopback fallthrough)
                  Non-branch instr (aka. non-CTI)
Value =     43605, select = 10, mask = 0004.  Load Instruction (Data Memory).
                  Load from local memory i.e. DataRAM, DataROM, InstRAM, InstROM
Value =     22912, select = 11, mask = 0004.  Store Instruction (Data Memory).
                  Store to local memory i.e. DataRAM, InstRAM
Value =    753508, select =  6, mask = 01ed.  Hold and Other Bubble cycles.
                  Processor domain PSO bubble
                  R hold caused by Data Cache miss(unused)
                  R hold caused by Store release
                  R hold caused by MEMW, EXTW or EXCW
                  R hold caused by Halt instruction (TX only)
                  CTI bubble (e.g. branch delay slot)
                  WAITI bubble i.e. a cycle spent in WaitI power down mode.
Value =     76206, select =  6, mask = 0010.  Hold and Other Bubble cycles.
                  R hold caused by register dependency
Value =         0, select =  1, mask = 0001.  Overflow of counter.
                  Overflow counter
Value =   1342710, select =  0, mask = 6c696857.  Counts cycles.
                  Amount of cycles
Bench iram
---------------------
Value =   1128851, select =  0, mask = 0001.  Counts cycles.
                  Amount of cycles
Value =    268098, select =  2, mask = 8dff.  Successfully Retired Instructions.
                  JX instructions
                  CALLXn instructions
                  return instructions (RET, RETW, ...)
                  supervisor return instructions (RFDE, RFE, RFI, RFWO, RFWU)
                  Conditional branch instructions where execution
                  transfers to the target (aka. taken branch),
                   or loopgtz/loopnez instr where execution skips
                   the loop (aka. not-taken loop)
                  J instr
                  CALLn instr
                  Conditional branch instr where execution
                   falls through (aka. not-taken branch)
                  Loop instr where execution falls into loop (aka. taken loop)
                  Last inst of loop and execution transfers
                   to LBEG (aka. loopback taken)
                  Last inst of loop and execution falls 
                   through to LEND (aka. loopback fallthrough)
                  Non-branch instr (aka. non-CTI)
Value =     43586, select = 10, mask = 0004.  Load Instruction (Data Memory).
                  Load from local memory i.e. DataRAM, DataROM, InstRAM, InstROM
Value =     22902, select = 11, mask = 0004.  Store Instruction (Data Memory).
                  Store to local memory i.e. DataRAM, InstRAM
[... 34 lines cut ...]
                  Last inst of loop and execution falls 
                   through to LEND (aka. loopback fallthrough)
                  Non-branch instr (aka. non-CTI)
Value =     43560, select = 10, mask = 0004.  Load Instruction (Data Memory).
                  Load from local memory i.e. DataRAM, DataROM, InstRAM, InstROM
Value =     22890, select = 11, mask = 0004.  Store Instruction (Data Memory).
                  Store to local memory i.e. DataRAM, InstRAM
Value =    563117, select =  6, mask = 01ed.  Hold and Other Bubble cycles.
                  Processor domain PSO bubble
                  R hold caused by Data Cache miss(unused)
                  R hold caused by Store release
                  R hold caused by MEMW, EXTW or EXCW
                  R hold caused by Halt instruction (TX only)
                  CTI bubble (e.g. branch delay slot)
                  WAITI bubble i.e. a cycle spent in WaitI power down mode.
Value =     76136, select =  6, mask = 0010.  Hold and Other Bubble cycles.
                  R hold caused by register dependency
Value =         0, select =  1, mask = 0001.  Overflow of counter.
                  Overflow counter
Value =    939577, select =  0, mask = 6c696857.  Counts cycles.
                  Amount of cycles
```

To sum up, on DRAM, the rendering is 30% faster than SPIRAM and IRAM is 15% faster than SPIRAM (I guess because of the overhead of some 8 bit accesses on the 32bits IRAM that trap the CPU and must be emulated).
This doesn't explain the difference you are observing.

X-Ryl669, 2022-03-12:
The difference reduces to 25% when accounting for color conversion (DRAM is only 25% faster than SPIRAM). Couldn't test IRAM here, since color conversion code is doing unaligned access that's not supported by IRAM.

The next step would be to have rlottie allocate all its structures in DRAM to test (a lot harder since it implies modifying the C++ allocator system wide not to use SPIRAM...).

X-Ryl669, 2022-03-12:
Ok, here's the last test, and it matches what you observed.

```
Heap summary for capabilities 0x00000400:
  At 0x3f800000 len 4194303 free 4192095 allocated 20 min_free 4192095
    largest_free_block 4128768 alloc_blocks 1 free_blocks 1 total_blocks 2
  Totals:
    free 4192095 allocated 20 min_free 4192095 largest_free_block 4128768
Heap summary for capabilities 0x00000802:
  At 0x3ffbb744 len 32767 free 32011 allocated 0 min_free 32011
    largest_free_block 31744 alloc_blocks 0 free_blocks 1 total_blocks 1
  At 0x3ffae6e0 len 6432 free 1328 allocated 4668 min_free 152
    largest_free_block 1024 alloc_blocks 51 free_blocks 1 total_blocks 52
  At 0x3ffb67b8 len 170056 free 83568 allocated 85528 min_free 83568
    largest_free_block 81920 alloc_blocks 43 free_blocks 1 total_blocks 44
  At 0x3ffe0440 len 15072 free 14636 allocated 0 min_free 14636
    largest_free_block 14336 alloc_blocks 0 free_blocks 1 total_blocks 1
  At 0x3ffe4350 len 113840 free 112948 allocated 0 min_free 112948
    largest_free_block 110592 alloc_blocks 0 free_blocks 1 total_blocks 1
  At 0x40091818 len 59368 free 58544 allocated 0 min_free 58516
    largest_free_block 57344 alloc_blocks 0 free_blocks 1 total_blocks 1
  Totals:
    free 303035 allocated 90196 min_free 301831 largest_free_block 110592
Heap summary for capabilities 0x00000804:
  At 0x3ffbb744 len 32767 free 32011 allocated 0 min_free 32011
    largest_free_block 31744 alloc_blocks 0 free_blocks 1 total_blocks 1
  At 0x3ffae6e0 len 6432 free 1328 allocated 4668 min_free 152
    largest_free_block 1024 alloc_blocks 51 free_blocks 1 total_blocks 52
  At 0x3ffb67b8 len 170056 free 83568 allocated 85528 min_free 83568
    largest_free_block 81920 alloc_blocks 43 free_blocks 1 total_blocks 44
  At 0x3ffe0440 len 15072 free 14636 allocated 0 min_free 14636
    largest_free_block 14336 alloc_blocks 0 free_blocks 1 total_blocks 1
  At 0x3ffe4350 len 113840 free 112948 allocated 0 min_free 112948
    largest_free_block 110592 alloc_blocks 0 free_blocks 1 total_blocks 1
  Totals:
    free 244491 allocated 90196 min_free 243315 largest_free_block 110592
Opened animation 61 frames 10
Bench psram
---------------------
Value =   4091424, select =  0, mask = 0001.  Counts cycles.
                  Amount of cycles
Value =    380758, select =  2, mask = 8dff.  Successfully Retired Instructions.
                  JX instructions
                  CALLXn instructions
                  return instructions (RET, RETW, ...)
                  supervisor return instructions (RFDE, RFE, RFI, RFWO, RFWU)
                  Conditional branch instructions where execution
                  transfers to the target (aka. taken branch),
                   or loopgtz/loopnez instr where execution skips
                   the loop (aka. not-taken loop)
                  J instr
                  CALLn instr
                  Conditional branch instr where execution
                   falls through (aka. not-taken branch)
                  Loop instr where execution falls into loop (aka. taken loop)
                  Last inst of loop and execution transfers
                   to LBEG (aka. loopback taken)
                  Last inst of loop and execution falls 
                   through to LEND (aka. loopback fallthrough)
                  Non-branch instr (aka. non-CTI)
Value =     48024, select = 10, mask = 0004.  Load Instruction (Data Memory).
                  Load from local memory i.e. DataRAM, DataROM, InstRAM, InstROM
Value =     39438, select = 11, mask = 0004.  Store Instruction (Data Memory).
                  Store to local memory i.e. DataRAM, InstRAM
Value =   2104364, select =  6, mask = 01ed.  Hold and Other Bubble cycles.
                  Processor domain PSO bubble
                  R hold caused by Data Cache miss(unused)
                  R hold caused by Store release
                  R hold caused by MEMW, EXTW or EXCW
                  R hold caused by Halt instruction (TX only)
[... 171 lines cut ...]
    largest_free_block 14336 alloc_blocks 0 free_blocks 1 total_blocks 1
  At 0x3ffe4350 len 113840 free 112948 allocated 0 min_free 112948
    largest_free_block 110592 alloc_blocks 0 free_blocks 1 total_blocks 1
  At 0x40091818 len 59368 free 58544 allocated 0 min_free 42160
    largest_free_block 57344 alloc_blocks 0 free_blocks 1 total_blocks 1
  Totals:
    free 227595 allocated 165636 min_free 211063 largest_free_block 110592
Heap summary for capabilities 0x00000804:
  At 0x3ffbb744 len 32767 free 32011 allocated 0 min_free 32011
    largest_free_block 31744 alloc_blocks 0 free_blocks 1 total_blocks 1
  At 0x3ffae6e0 len 6432 free 300 allocated 5696 min_free 152
    largest_free_block 0 alloc_blocks 74 free_blocks 0 total_blocks 74
  At 0x3ffb67b8 len 170056 free 9156 allocated 159940 min_free 9156
    largest_free_block 7680 alloc_blocks 289 free_blocks 2 total_blocks 291
  At 0x3ffe0440 len 15072 free 14636 allocated 0 min_free 14636
    largest_free_block 14336 alloc_blocks 0 free_blocks 1 total_blocks 1
  At 0x3ffe4350 len 113840 free 112948 allocated 0 min_free 112948
    largest_free_block 110592 alloc_blocks 0 free_blocks 1 total_blocks 1
  Totals:
    free 169051 allocated 165636 min_free 168903 largest_free_block 110592
```

In short, if the rlottie structures (result of parsing) are in SPIRAM, then rendering is 4x slower. Please ignore the IRAM advantage here, since there's no color conversion in this test for IRAM. 
So clearly, the output rendering buffer can be in SPIRAM with no big impact on the performance (25% slower), since it's pipelined IIUC. 
However, it's super important that the parser allocates in DRAM (in that case, the rendering speed is 3.3x faster)

X-Ryl669, 2022-03-12:
For information, parsing the file takes 26333964 cycles (so 21.7x longer than rendering)

tvanfossen, 2022-03-14:
This is fantastic info!!! Thanks for digging into this more, way more precise than what I was doing - I just received a quad spi PSRAM ESP32S3 last week that I am started to test on, to compare how much of a difference the speed of the PSRAM spi channel makes, octal vs quad vs single channel.

To your point of modifying rlotties allocator to be able for DRAM/IRAM/SPIRAM/wherever else it could go, at what point do you think its worthwhile for an official lottie plugin for LVGL, rather than relying upon rlottie and any nuance that comes with it? Presumably the main value would be in the embedded side, which is a niche of LVGL to begin with. Is something like a ThorVG port into LVGL a better route for rendering lottie than rlottie? Is lottie the ideal format for rendering pre-built animations from a file, as opposed to Rive, Gifs, etc. 

Re-invent the wheel vs square peg/round hole

X-Ryl669, 2022-03-14:
And for the other questions:

I'm working on improving rlottie to consume less memory. Right now, I'm changing the code to defer to another vector class that's not using exponential-grow to allocate (so when allocating 17 elements, you don't pay for 32 elements like standard STL). This is not a huge work and it should hit my memshrink branch tomorrow. It'll be slower than the classical std::vector unless the code is modified to pre-reserve the expected size (I'm also looking at this, but the rlottie code is a bit convoluted). I'm sure the overhead will worth it, since the benchmark above show that a 12kB JSON file consumes 74kB heap memory once parsed. The less memory it uses, the more compact it'll be in SPIRAM and it'll fit the 32KB cache better so it might be faster in the end (also, parsing is done once, but rendering is done for each frame).

Then there's the possibility to change the allocator. Unfortunately, in C++, if you override operator new, it's global for all program (that's one of the historical dumbness of C++), or you need to overload per class (which is impossible to do for rlottie, too much work).

Yet, I think I've found a solution that might help more people. My idea is to implement a `operator new` override that's using a thread local storage variable telling where to allocate. So the code would be something like this:

```cpp
render_rlottie() {

   set_custom_allocator(DRAM);
   lottie = lottie_animation_from_file(...);   
   set_custom_allocator(Default);
}

// Allocator
thread_local mode; // Either DRAM, IRAM, SPIRAM, whatever
void * operator new(size_t n) {
    switch (mode) {
      case Default: return ::malloc(n);
      case DRAM: return ::heap_cap_malloc(DRAM, n);
...
}
[...]
```

For LVGL, I think it'll need more work to implement a capability aware allocator (a smarter `lv_malloc`)
The idea would be to call `lv_rlottie_create()` with some intent to tell if it's going to be used once, if it's rotated, resized and so on. With that intent, we could decide what heap to use smartly. It makes sense to use SPIRAM for lottie files used as vector images only (no animation), or when rotation is required (in that case it's rendered to a complete buffer anyway). But for animation where rendering needs to happen multiple times per second, then we should prefer DRAM for the lottie internal structures, and SPIRAM for the rendering buffer (unless there's enough DRAM/IRAM for those too).

In short, for light interfaces with 3 animated icons, I think we can work with the DRAM heap (that's around 180kB) and gain a good 3 to 4x the performance instead of using the SPIRAM. 

Last possible solution to optimize more would be to change rlottie code to use hash instead of strings for its *few* `unordered_map` in order to avoid most useless heap allocation for `std::string`. Maybe we could gain another 1 to 2kB here. But it's not the low hanging fruit anymore...

I don't think ThorVG will be the panacea either. It has many advantages but a lot of cons for me:
1. There are too few software to create rive file and none are open source (unlike lottie format that's usable with [Glaxnimate](https://gitlab.com/Sythelux/glaxnimate), Synfig, Haiku, ...)
2. Code doesn't fit **true** embedded platform very well
3. SVG is larger than lottie format (XML vs JSON), so for simple vectorial icons stored in flash, lottie is better (and with my branch, it's not required to load the json file in the heap, it can be parsed from flash directly). Their binary mode (.thorvg format) is just about the same size as uncompressed lottie file.
4. lottie support in ThorVG isn't there. I doubt it'll happen soon
5. There's no cmake support, so it has to be written from scratch.

X-Ryl669, 2022-03-16:
I've tried with a vector class that's not using exponential growth and go these results:
```
Heap summary for capabilities 0x00000400:
  At 0x3f800000 len 4194303 free 4192095 allocated 20 min_free 4192095
    largest_free_block 4128768 alloc_blocks 1 free_blocks 1 total_blocks 2
  Totals:
    free 4192095 allocated 20 min_free 4192095 largest_free_block 4128768
Heap summary for capabilities 0x00000802:
  At 0x3ffbb744 len 32767 free 32011 allocated 0 min_free 32011
    largest_free_block 31744 alloc_blocks 0 free_blocks 1 total_blocks 1
  At 0x3ffae6e0 len 6432 free 1328 allocated 4668 min_free 152
    largest_free_block 1024 alloc_blocks 51 free_blocks 1 total_blocks 52
  At 0x3ffb67b8 len 170056 free 83568 allocated 85528 min_free 83568
    largest_free_block 81920 alloc_blocks 43 free_blocks 1 total_blocks 44
  At 0x3ffe0440 len 15072 free 14636 allocated 0 min_free 14636
    largest_free_block 14336 alloc_blocks 0 free_blocks 1 total_blocks 1
  At 0x3ffe4350 len 113840 free 112948 allocated 0 min_free 112948
    largest_free_block 110592 alloc_blocks 0 free_blocks 1 total_blocks 1
  At 0x40091818 len 59368 free 58544 allocated 0 min_free 58516
    largest_free_block 57344 alloc_blocks 0 free_blocks 1 total_blocks 1
  Totals:
    free 303035 allocated 90196 min_free 301831 largest_free_block 110592
Heap summary for capabilities 0x00000804:
  At 0x3ffbb744 len 32767 free 32011 allocated 0 min_free 32011
    largest_free_block 31744 alloc_blocks 0 free_blocks 1 total_blocks 1
  At 0x3ffae6e0 len 6432 free 1328 allocated 4668 min_free 152
    largest_free_block 1024 alloc_blocks 51 free_blocks 1 total_blocks 52
  At 0x3ffb67b8 len 170056 free 83568 allocated 85528 min_free 83568
    largest_free_block 81920 alloc_blocks 43 free_blocks 1 total_blocks 44
  At 0x3ffe0440 len 15072 free 14636 allocated 0 min_free 14636
    largest_free_block 14336 alloc_blocks 0 free_blocks 1 total_blocks 1
  At 0x3ffe4350 len 113840 free 112948 allocated 0 min_free 112948
    largest_free_block 110592 alloc_blocks 0 free_blocks 1 total_blocks 1
  Totals:
    free 244491 allocated 90196 min_free 243315 largest_free_block 110592
Bench psram
---------------------
Value =   4399859, select =  0, mask = 0001.  Counts cycles.
                  Amount of cycles
Value =    392966, select =  2, mask = 8dff.  Successfully Retired Instructions.
                  JX instructions
                  CALLXn instructions
                  return instructions (RET, RETW, ...)
                  supervisor return instructions (RFDE, RFE, RFI, RFWO, RFWU)
                  Conditional branch instructions where execution
                  transfers to the target (aka. taken branch),
                   or loopgtz/loopnez instr where execution skips
                   the loop (aka. not-taken loop)
                  J instr
                  CALLn instr
                  Conditional branch instr where execution
                   falls through (aka. not-taken branch)
                  Loop instr where execution falls into loop (aka. taken loop)
                  Last inst of loop and execution transfers
                   to LBEG (aka. loopback taken)
                  Last inst of loop and execution falls
                   through to LEND (aka. loopback fallthrough)
                  Non-branch instr (aka. non-CTI)
Value =     50446, select = 10, mask = 0004.  Load Instruction (Data Memory).
                  Load from local memory i.e. DataRAM, DataROM, InstRAM, InstROM
Value =     42212, select = 11, mask = 0004.  Store Instruction (Data Memory).
                  Store to local memory i.e. DataRAM, InstRAM
Value =   2634778, select =  6, mask = 01ed.  Hold and Other Bubble cycles.
                  Processor domain PSO bubble
                  R hold caused by Data Cache miss(unused)
                  R hold caused by Store release
                  R hold caused by MEMW, EXTW or EXCW
                  R hold caused by Halt instruction (TX only)
                  CTI bubble (e.g. branch delay slot)
                  WAITI bubble i.e. a cycle spent in WaitI power down mode.
[... 167 lines cut ...]
    largest_free_block 14336 alloc_blocks 0 free_blocks 1 total_blocks 1
  At 0x3ffe4350 len 113840 free 112948 allocated 0 min_free 112948
    largest_free_block 110592 alloc_blocks 0 free_blocks 1 total_blocks 1
  At 0x40091818 len 59368 free 58544 allocated 0 min_free 42160
    largest_free_block 57344 alloc_blocks 0 free_blocks 1 total_blocks 1
  Totals:
    free 216791 allocated 176440 min_free 169679 largest_free_block 110592
Heap summary for capabilities 0x00000804:
  At 0x3ffbb744 len 32767 free 20811 allocated 11200 min_free 9615
    largest_free_block 10752 alloc_blocks 1 free_blocks 2 total_blocks 3
  At 0x3ffae6e0 len 6432 free 300 allocated 5696 min_free 152
    largest_free_block 0 alloc_blocks 74 free_blocks 0 total_blocks 74
  At 0x3ffb67b8 len 170056 free 9552 allocated 159544 min_free 1360
    largest_free_block 8192 alloc_blocks 288 free_blocks 2 total_blocks 290
  At 0x3ffe0440 len 15072 free 14636 allocated 0 min_free 3444
    largest_free_block 14336 alloc_blocks 0 free_blocks 1 total_blocks 1
  At 0x3ffe4350 len 113840 free 112948 allocated 0 min_free 112948
    largest_free_block 110592 alloc_blocks 0 free_blocks 1 total_blocks 1
  Totals:
    free 158247 allocated 176440 min_free 127519 largest_free_block 110592
```

[... cut ...] , as it shouldn't behave like this. There are few places in the lottie code that use `std::copy_n(... std::back_inserter)` without reserving the memory area before and for these places, it should call `std::vector::push_back` repeatedly which, in turn reserve an exponential grown buffer. My implementation only grow to the minimum possible size so it should save memory but it's clearly no acting this way. Maybe I have some memory leaks here.

It's in my memShrinkVector branch if you want to test.

X-Ryl669, 2022-03-16:
Re-run the test with adding `Vvector::reserve` where appropriate, and I'm getting 165280 bytes allocation in DRAM vs 165636 with `std::vector`. Not sure it's worth it for 356 bytes saved in heap.

vednildip, 2024-12-31:
in  new  version  lvgl  9.00 onward  #include <rlottie_capi.h>  is missing  any  one  has  idea how  to  get  this header

espressif2022, 2025-01-02:
> in new version lvgl 9.00 onward #include <rlottie_capi.h> is missing any one has idea how to get this header

Hello, you should add a submodule here, please refer to [here](https://docs.lvgl.io/master/details/libs/rlottie.html#getting-animations)