espressif/esp-box #9: EN Extremely bad wake word recognition

StuartIanNaylor, 2021-12-21:
Hi downloaded https://github.com/espressif/esp-box/releases/download/v0.2.1/ESP-Box_Demo_EN_V0.2.1.bin and flashed.

I have spent over ten minutes several reboots to get only 3 recognitions with worry that my neighbours may be thinking I have gone mad.

If you do manage the wake word then the command turn on/off light works great.
Actually getting the wake word to activate is another matter.

I have a feeling it could be your dataset and now it actually has a english male voice it doesn't fit well or this firmware / model is not good.
I have no problems recording some samples if you tell me format and sampling rate...
Maybe community members could contribute if its merely a esp32-s3-box dataset accent problem.

s60sc, 2021-12-22:
I received my ESP Box today.
I've not been able to get the wake word to activate at all. I've tried both the 0.1.1 and 0.2.1 firmware, and also tried waking it with the voice commands in this demo [video](https://www.youtube.com/watch?v=n0jH1Brmee8) without success.

s60sc, 2021-12-24:
Tried a Skainet [example ](https://github.com/espressif/esp-skainet/tree/master/examples/en_speech_commands_recognition) which uses the wake word Alexa and this works fine.

StuartIanNaylor, 2021-12-26:
> Tried a Skainet [example ](https://github.com/espressif/esp-skainet/tree/master/examples/en_speech_commands_recognition) which uses the wake word Alexa and this works fine.
> 
> Compiled the ESP_Box demo to use wakeword model alexa8 instead of hiesp8 and now wakes easily.

Maybe the Hi_ESP_Q8 model as I did get it to work but learning how to position myself and control my speech for wake word recognition took considerable time.
But maybe your recompile as multinet in the supplied firmware seemed equally buggy as after saying 'turn white' anything was recognised as 'turn white'.

I need to get back up to speed with esp and install the IDF and get to grips with the framework so will also compile and test but currently the supplied firmware gives an extremely bad impression of the box capabilities which in use is near useless.

@s60sc how did you find the multinet commands?

https://github.com/espressif/esp-sr/blob/509cf82658cf2b645aa9ae89b6764beea48e7eae/include/esp_map.h#L23
Also did you change to 34mm as with the box mic arrangement?

s60sc, 2021-12-26:
The multiword model worked fine for me. I did not change the mic arrangement. I will do more work on the ESP-BOX in the new year. I have uploaded a version of the firmware with the Alexa wakeword in this [repository](https://github.com/s60sc/ESP-BOX_firmware_Alexa)

feizi, 2022-01-04:
Hi, we have updated a new "Hi,ESP" model by adding more training samples. 
You can try it by pulling the latest [esp-sr release/v1.0 branch](https://github.com/espressif/esp-sr/tree/release/v1.0).

s60sc, 2022-01-04:
Tried the skainet en_speech_commands_recognition example after updating the esp-sr component, installed on the ESP-Box.
The new hiesp8 model crashes the ESP-Box without even logging an exception. The alexa8 model works fine.

hiesp8 version:
```
model_name: hiesp8 model_data: /srmodel/hiesp8/wn8_data
HUFZIP MC Quantized wakeNet8: WakeNet8_v3h8_hiesp_5_0.59_0.60, mode:2, p:1, (Jan  4 2022 15:08:13)
```

alexa8 version:
```
model_name: alexa8 model_data: /srmodel/alexa8/wn8_data
HUFZIP MC Quantized wakeNet8: wakeNet8_v5h8_alexa_5_0.57_0.59, mode:2, p:1, (Jan  4 2022 15:08:13)
```

feizi, 2022-01-05:
have fixed the bug wakenet8_hiesp, pls try again.

s60sc, 2022-01-05:
Now works, and hiesp8 model is significantly improved.

Updated factory demo firmwares in this [repository](https://github.com/s60sc/ESP-BOX_firmware_Alexa)