espressif/esp-box #10: [After playing well for 1 day ...]PermissionError(13, 'A device attached to the system is not functioning)

BigJohnn, 2022-01-08:
Hi, 
    I have followed some tutorials, and do 
> idf.py -p COMX build flash

... things go well.
But one day later, when I open my PC again, trying to run 

> idf.py -p COMX monitor

I got

`serial.serialutil.SerialException:` Cannot configure port, something went wrong. Original message: PermissionError(13, 'A device attached to the system is not functioning.', None, 31)`

I have tried to change my COM number or uninstall&&reinstall the drive of the serial, but nothing works T.T
Any ideas? Thx.

Real-YuZhe, 2022-01-10:
Try manually putting the device into download mode and try again. To do this, keep GPIO0 low while toggling reset. On many development boards, the “Boot” button is connected to GPIO0, and you can press “Reset” button while holding “Boot”.

BigJohnn, 2022-01-10:
> 

Thank you very much! I'll try this later～

BigJohnn, 2022-01-10:
@Real-YuZhe It WORKS again! 多谢多谢~~

closed 2022-01-10 by BigJohnn