espressif/esp-box #11: 编译出厂工程(factory_demo)失败

Beanv587, 2022-01-10:
系统:win10 64位 企业版

idf用的v4.4 release

已经按照README应用了补丁.编译到最后link阶段就报错了
[1466/1468] Linking CXX executable factory_demo.elf
FAILED: factory_demo.elf
[image]

请问有解决方案吗?

BigJohnn, 2022-01-11:
看起来像是代码不够新?  # 我猜的

Beanv587, 2022-01-11:
@BigJohnn 
被你猜对了哈哈.以为刚clone下来的所有的都是最新的..已经解决 谢谢兄台

BigJohnn, 2022-01-11:
> @BigJohnn 我按兄台说的试了下,全部更新到最新是可以编译的.但是按下面的操作应用补丁后编译就这个错误.
> 
> For Windows users, you may need to do the following manually:
> 
> Change the working directory to the folder where esp-idf is located Run the following commands in cmd in sequence： git fetch origin git checkout 35b20cadce65ce79c14cf2018efc87c44d71ab21 git apply X:\path\to/esp-box\idf_patch\idf_patch.patch
> 
> 由于昨天应用了补丁,所以git submodule update --init --recursive更新后就能编译了....但是补丁也就回去了(PS. idf_patch\components\esp_phy目录的3个文件不管是否覆盖都能正常编译)

不需要用补丁，也许文档所谓的bug， 最新commit已经修复了。 反正我用的时候ok

Beanv587, 2022-01-11:
> > @BigJohnn我兄弟台说的试了下，全部更新到最新是可以按但可以编译的。按下面的操作应用程序后编译就这个错误。
> > 对于 Windows 用户，您可能需要手动执行以下操作：
> > 将工作目录切换到esp-idf所在文件夹在cmd中依次运行以下命令： git fetch origin git checkout 35b20cadce65ce79c14cf2018efc87c44d71ab21 git apply X:\path\to/esp-box\idf_patch\idf_patch.patch
> > 昨天应用了补丁，所以 git 子模块更新 --init 更新后导致编译了....但是替换值 --re (PS. id_patch\esp_phy 的多个文件是否覆盖了当前的 3 个组件正常编译)
> 
> 不需要的，可能文档，最新提交修复了。

明白了哈哈.谢谢兄台

closed 2022-01-11 by Beanv587