espressif/esp-box #15: 烧录factory_demo出错

MinjunHuang, 2022-03-21:
操作系统：macOS Mojave
ESP-IDF版本：4.4
问题描述：运行idf.py -p /dev/cu.usbmodem14701 flash 
  已确认cu.usbmodem14701是esp-box设备。
  出现如下错误，看信息是连接不上ESP_BOX:

esptool.py esp32s3 -p /dev/cu.usbmodem14701 -b 460800 --before=default_reset --after=hard_reset --no-stub write_flash --flash_mode dio --flash_freq 80m --flash_size 16MB 0x0 bootloader/bootloader.bin 0x20000 factory_demo.bin 0x8000 partition_table/partition-table.bin 0x16000 ota_data_initial.bin 0x3bd000 storage.bin 0x647000 model.bin
esptool.py v3.2-dev
Serial port /dev/cu.usbmodem14701
Connecting......................................

A fatal error occurred: Failed to connect to ESP32-S3: No serial data received.
For troubleshooting steps visit: https://github.com/espressif/esptool#troubleshooting
CMake Error at run_serial_tool.cmake:56 (message):
  /Users/rainbow/.espressif/python_env/idf4.4_py3.7_env/bin/python
  /Users/rainbow/WorkSpace/esp/esp-idf/components/esptool_py/esptool/esptool.py
  --chip esp32s3 failed

make[3]: *** [CMakeFiles/flash] Error 1
make[2]: *** [CMakeFiles/flash.dir/all] Error 2
make[1]: *** [CMakeFiles/flash.dir/rule] Error 2
make: *** [flash] Error 2
make failed with exit code 2

ESP-Mars, 2022-03-21:
请按住 Boot 键的同时按一下 Reset 键手动进入固件下载模式，通过串口下载固件。

MinjunHuang, 2022-03-21:
OK了，多谢哈！

closed 2022-03-21 by MinjunHuang