espressif/esp-box #16: 离线语音识别是否支持变量识别

Kingwulin, 2022-03-30:
想要是实现离线语音识别 “把空调温度调到xx摄氏度” xx 为变量值，取值范围为 0-99
目前esp-box是否能实现该功能

ESP-Mars, 2022-03-30:
可以实现，设定 100 条离线语音命令词，但命令次长度不宜过长。

Kingwulin, 2022-03-31:
> 可以实现，设定 100 条离线语音命令词，但命令次长度不宜过长。

就是说只能是固定的语音，不能存在中间有变量的识别对吧

ESP-Mars, 2022-04-01:
是的

Kingwulin, 2022-04-14:
> 是的

你好，这个在esp-box上有现成的例子吗

ESP-Mars, 2022-04-14:
Yes, please follow the instruction to download our Latest developing firmware https://github.com/espressif/esp-box/blob/master/docs/firmware_update.md
Access the user guide please click https://github.com/espressif/esp-box/blob/master/docs/getting_started.md

Kingwulin, 2022-04-14:
> Yes, please follow the instruction to download our Latest developing firmware https://github.com/espressif/esp-box/blob/master/docs/firmware_update.md Access the user guide please click https://github.com/espressif/esp-box/blob/master/docs/getting_started.md

你好，我这边是要实现自定义离线语音识别的，这个功能的demo代码在什么位置

Kingwulin, 2022-04-14:
你好，如果我要用esp32s3开发板实现离线语音识别，外接I2S数字麦克风模块，是否有类似的或者接近的demo可供参考

ESP-Mars, 2022-04-14:
您好，上面提供的应用固件就可以自定义的离线命令词。
或者您试一下单纯离线语音识别的 example https://github.com/espressif/esp-skainet/tree/dev/2.0/examples/cn_speech_commands_recognition

ESP-Mars, 2023-06-30:
Hi, if you don't have any questions, please close this issue. Thank you!

closed 2023-07-11 by ESP-Mars