espressif/esp-box #17: Matter Example App for esp-box?

oidebrett, 2022-04-04:
Hi,
Would it be possible for a matter (project chip) example app to be provided? I have been successfully prototyping with the esp32 development board and matter by following this example code in the matter open sourc library:
https://github.com/project-chip/connectedhomeip/tree/master/examples/all-clusters-app/esp32

I would like to try this code on the new esp-box but I suspect it may work but would not provide the Ai voice activation and UI that Is in the "out of the box" firmware?

If there were plans to provide a matter enabled esp-box sample application I would really appreciate that as it would safe loads of time and I suspect others would be interested too.

many thanks
Oide

ESP-Mars, 2022-04-06:
Yes, espressif will release an esp matter SDK in this month (no external link now), and a matter example will be added in the esp-box repository, please look forward to!

DanTup, 2023-06-23:
Is this example available now? I was unable to find it.

espressif2022, 2023-06-26:
> Is this example available now? I was unable to find it.

We will have a meeting with the matter team tonight to discuss how to release this example.

ESP-Mars, 2023-06-30:
We will clean and rebase a version of this hot demand example. Thanks for your patience.

DanTup, 2023-07-04:
Thanks! It's not urgent for me, I don't currently have an ESP-Box. I was mostly curious about the status of this because I was considering buying an ESP-Box and wanted to understand what capabilities it would have for controlling matter devices in future (I'd like to stop IoT things in my house relying on external services!).

ESP-Mars, 2023-08-17:
Hi @DanTup @oidebrett we have release a matter switch demo for user to enjoy! https://github.com/espressif/esp-box/tree/master/examples/matter_switch

closed 2023-09-04 by ESP-Mars