espressif/esp-box #19: esp32box 下的factory demo编译时提示错误

Kingwulin, 2022-04-21:
你好，编译example下factory demo的时候总是提示以下错误，请问如何解决
CMake Error at /home/king5/code/idfCode/esp-idf/tools/cmake/build.cmake:200 (message):
  Failed to resolve component 'rmaker_common'.
Call Stack (most recent call first):
  /home/king5/code/idfCode/esp-idf/tools/cmake/build.cmake:226 (__build_resolve_and_add_req)
  /home/king5/code/idfCode/esp-idf/tools/cmake/build.cmake:512 (__build_expand_requirements)
  /home/king5/code/idfCode/esp-idf/tools/cmake/project.cmake:378 (idf_build_process)
  CMakeLists.txt:22 (project)

idf 的版本是4.4的
ESP-IDF v4.4.1-64-g4b2098bc58-dirty

TDA-2030, 2022-04-21:
看起来是子仓库没有下载，你在esp-box目录下运行`git submodule update --init --recursive`可能就没问题了

Fightzzw, 2022-05-28:
运行这个指令没反应git submodule update --init --recursive。
还是确实好多头文件怎么办？

ESP-Mars, 2023-06-30:
Hi, if you don't have any questions, please close this issue. Thank you!

closed 2023-07-11 by ESP-Mars