espressif/esp-box #21: Interest in switching to mono mp3s for reduced programming size?

chmorgan, 2022-04-24:
For my application I've been looking at using mono mp3s for size, and because the es8311 codec is mono output only.

Would there be interest in a PR to switch the mp3s to mono, reduce the partition size to reduce programming, and the code to properly handle the mono mp3s? This part isn't too tricky but did require some test and debug and it could help the next person.

I'd like to get the ok that the idea made sense to the maintainers here.

alic-maker, 2022-04-25:
Yes, I also thought about storing mono MP3 files in the beginning to save flash space. I'm not using mono now because I don't know how to configure es8311 and I2S of ESP32S3 to play mono audio correctly, playing directly will cause the playback speed to be double normal. Taking some time to solve this problem should be in our development plan later.

chmorgan, 2022-04-25:
@alic-maker getting them to work is exactly what my pr would be as I’ve added support for mono here after a few hours of debugging. If I can get an admin for the project to “ok” the idea I’ll submit a pr. It’s clean, makes sense and it’s not complex, but don’t want to go through the effort without a green light in case no one is maintaining the repo actively and it could sit there for months or years.

ESP-Mars, 2022-04-25:
hi, we welcome the feature via PR, the feature of switching mp3 to mono is very useful to use and what you are going to do is meaningful, maintainer @alic-maker will maintain the repo all the time.
Espressif also has a [Contributions Guide](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/contribute/index.html) for anyone interested in contributing to our repositories.

chmorgan, 2022-04-25:
Oh gotcha, didn’t realize alic-maker was an admin. I’ll submit an initial pr today.

chmorgan, 2022-04-26:
@ESP-Mars @alic-maker opened https://github.com/espressif/esp-box/pull/22

lijunru-hub, 2022-12-05:
The box playback must provide two channels of audio data and only the left channel is valid. The other channel will be ignored. You need to manually patch the single channel to dual channel

chmorgan, 2023-03-05:
Closing as I've resolved mono support in playback.

closed 2023-03-05 by chmorgan