espressif/esp-box #23: Moving audio file management out of audio.c and into the ui for better reuse of audio logic?

chmorgan, 2022-05-04:
I find myself editing the esp-box mp3 example pretty heavily in order to support a new ui that wants to play a single file based on a user input.

If audio.c/.h would have an api like:
- play, pause, play_file

It would let me modify the ui to call play_file() based on button events, for example, and I think it could make the code a bit more reusable.

Other changes that would be made would be:
- event type for playback completion (so the caller can play the next song and maintain the functionality web ave today)
- move of file scanning and file list maintenance into the ui

I know it sounds like a bit of change, I'm having to make almost all of it here though, so if it was something of interest I could make it in the esp-box mp3 demo project and submit a PR.

Thoughts?

TDA-2030, 2022-05-04:
Moving file management out of audio.c is very good, but I don't think it's a good idea to put it into the ui. Maybe the example [factory_demo](https://github.com/espressif/esp-box/blob/master/examples/factory_demo/main/app/app_player.c) is closer to what you want.

chmorgan, 2022-05-04:
It looks like the audio task is still managing lists of files etc in the
code you linked to. For my use case I just want to play a single file and
then have the audio task stop, that’s what got me down the path of trying
to help improve slightly the flexibility of the example code, without of
course adding complexity. I think it’s possible to achieve both as we don’t
want examples that are complex.

How about a second approach like:

Ui -> audio file manager
 |
V
Audio task

Hope that ascii diagram looks ok.

This way we keep the file list out of the ui abs audio task and the ui can
take a “next” request, get the next file from the audio file manager, and
then make a call to an audio task function that queues up the message for
the task to play the song with a given name.

Chris

chmorgan, 2022-05-05:
Hello @TDA-2030 !

Did you see my msg above? If this looks ok I can refactor and submit a PR.

TDA-2030, 2022-05-05:
It seems to be a good idea. You can submit a PR and improve it together.

chmorgan, 2022-05-06:
@TDA-2030 in the case where we wanted to share the player code between factory and mp3_demo, how would/should that work? Should that be a component? I'll be working from the mp3_demo/ directory but trying to think of next steps after the adjustments there.

And I'd like to make it possible to use the code in a project I'm working on.

TDA-2030, 2022-05-06:
I think that putting the player code into a component will be a good choice.

chmorgan, 2023-03-05:
Moved to a component, closing as resolved.

closed 2023-03-05 by chmorgan