espressif/esp-box #24: I2C read error is not handled correctly in TT21100 driver

ratzori, 2022-05-05:
I2C read error is not handled correctly as `ret_val == ESP_OK` and `data_len > 0` is not checked here:
https://github.com/espressif/esp-box/blob/c46079095e4115282babbddb78179af547b6d548/components/i2c_devices/touch_panel/tt21100.c#L122

This will cause that `tt21100_read `will be called with `data_len == 0`
https://github.com/espressif/esp-box/blob/c46079095e4115282babbddb78179af547b6d548/components/i2c_devices/touch_panel/tt21100.c#L123

https://github.com/espressif/esp-box/blob/c46079095e4115282babbddb78179af547b6d548/components/i2c_devices/touch_panel/tt21100.c#L33

```
esp_err_t i2c_master_read(i2c_cmd_handle_t cmd_handle, uint8_t *data, size_t data_len, i2c_ack_type_t ack)
{
...
    ESP_RETURN_ON_FALSE(data_len > 0, ESP_ERR_INVALID_ARG, I2C_TAG, I2C_DATA_LEN_ERR_STR);
```

alic-maker, 2022-05-06:
You are right. It's strange that we haven't encountered `data_len=0` yet. Anyway, it'll be fixed soon.

ESP-Mars, 2023-06-30:
Hi, if you don't have any questions, please close this issue. Thank you!

closed 2023-07-11 by ESP-Mars