espressif/esp-box #26: AFE_FETCH_CHANNEL_VERIFIED 的具体含义是什么？

bronya5th, 2022-05-11:
在[app_sr.c](https://github.com/espressif/esp-box/blob/efbfa327324a3a797cfff3be7eb6ff0fbf803f5f/examples/factory_demo/main/app/app_sr.c)中，
[image]
AFE_FETCH_CHANNEL_VERIFIED感觉表示的是返回唤醒词识别成功，但在ESP-SR中显示的是返回音频通道确认，
[image]
这个音频通道确认是指什么音频通道确认？

WeldonWangwang, 2022-05-18:
Hi, @bronya5th 
此处的通道确认是因为在双麦算法中，2路麦克风数据经过前端信号处理后会分为 2 路处理后数据，其中一路是目标人声，一路是其他声音。当返回音频通道确认时，表明此时sr 算法已经找出 2 路中的目标人声所在通道，因此此时可以进行下一步语音识别，

ESP-Mars, 2023-06-30:
Hi, if you don't have any questions, please close this issue. Thank you!

closed 2023-07-02 by bronya5th