espressif/esp-box #27: 我无法使用内置的JTAG调试器调试

BeeeeeeLazy, 2022-05-14:
[image]
报错的界面如图片所示。
`launch.json`如下所示，程序可以正常的下载运行。
```
{
  "version": "0.2.0",
  "configurations": [
    {
      "name": "ESP32-S3 DEBUG",
      "type": "cppdbg",
      "request": "launch",
      "MIMode": "gdb",
      "miDebuggerPath": "${command:espIdf.getXtensaGdb}",
      "program": "${workspaceFolder}/build/${command:espIdf.getProjectName}.elf",
      "windows": {
        "program": "${workspaceFolder}\\build\\${command:espIdf.getProjectName}.elf"
      },
      "cwd": "${workspaceFolder}",
      "environment": [{ "name": "PATH", "value": "${config:idf.customExtraPaths}" }],
      "setupCommands": [
        { "text": "target remote :3333" },
        { "text": "set remote hardware-watchpoint-limit 2"},
        { "text": "mon reset halt" },
        { "text": "thb app_main" },
        { "text": "flushregs" }
      ],
      "externalConsole": false,
      "logging": {
        "engineLogging": true
      }
    }
  ]
}
```

BeeeeeeLazy, 2022-05-14:
openocd server 也可以打开。
我手头上有两块ESP32-S3的板子。一块是esp32-box，另外一块是自己画的。
两块尝试了均无法调试。

所以我觉得应该是我哪里的步骤出错了。

BeeeeeeLazy, 2022-05-14:
我看了一下openocd的输出信息，貌似它并没有起作用。
```
Info : only one transport option; autoselect 'jtag'
Info : esp_usb_jtag: VID set to 0x303a and PID to 0x1001
Info : esp_usb_jtag: capabilities descriptor set to 0x2000
Warn : Transport "jtag" was already selected
Info : Listening on port 6666 for tcl connections
Info : Listening on port 4444 for telnet connections
Info : esp_usb_jtag: Device found. Base speed 40000KHz, div range 1 to 255
Info : clock speed 40000 kHz
❌ Error: libus❌ b_bulk_write error: LIBUSB_ERROR_NOT_FOUND
❌ Error: libusb_open() fai❌ led with LIBUSB_ERROR_ACCESS
Error: libusb_open() failed with LIBUSB_ERROR_NOT_SUPPORTED
❌ Error: libusb_open() failed w❌ ith LIBUSB_ERROR_ACCESS
Error: libusb_open() failed with LIBUSB_ERROR_NOT_SUPPORTED
❌ Error: libusb_open() f❌ ailed with LIBUSB_ERROR_ACCESS
Error: libusb_open() failed with LIBUSB_ERROR_NOT_SUPPORTED
❌ Error: libusb_open() failed with LIBUSB_ERROR_ACCESS
Error: libusb_open() failed with LIBUSB_ERROR_NOT_SUPPORTED
❌ Error: libusb_open()❌  failed with LIBUSB_ERROR_ACCESS
Error: libusb_open() failed with LIBUSB_ERROR_NOT_SUPPORTED
❌ Error: libusb_open() failed wit❌ h LIBUSB_ERROR_ACCESS
Error: libusb_open() failed with LIBUSB_ERROR_NOT_SUPPORTED
❌ Error: esp_usb_jtag: dev❌ ice not found!
Error: esp_usb_jtag: failed to revive USB device!
Error: libusb_bulk_write error: LIBUSB_ERROR_NOT_FOUND
❌ Error: libusb_open() failed with LIBUSB_ERROR_ACCESS
Error: libusb_open() failed with LIBUSB_ERROR_NOT_SUPPORTED
❌ Error: libusb_open() failed wit❌ h LIBUSB_ERROR_ACCESS
Error: libusb_open() failed with LIBUSB_ERROR_NOT_SUPPORTED
❌ Error: libusb_open() fail❌ ed with LIBUSB_ERROR_ACCESS
Error: libusb_open() failed with LIBUSB_ERROR_NOT_SUPPORTED
❌ Error: libusb_open() failed ❌ with LIBUSB_ERROR_ACCESS
Error: libusb_open() failed with LIBUSB_ERROR_NOT_SUPPORTED
❌ Error: libusb_open() failed with L❌ IBUSB_ERROR_ACCESS
Error: libusb_open() failed with LIBUSB_ERROR_NOT_SUPPORTED
❌ Error: libusb_open() f❌ ailed with LIBUSB_ERROR_ACCESS
Error: libusb_open() failed with LIBUSB_ERROR_NOT_SUPPORTED
❌ Error: esp_usb_jtag: devi❌ ce not found!
Error: esp_usb_jtag: failed to revive USB device!
Error: libusb_bulk_write error: LIBUSB_ERROR_NOT_FOUND
❌ Error: libusb_open() failed with LIBUSB_ERROR_ACCESS
Error: libusb_open() failed with LIBUSB_ERROR_NOT_SUPPORTED
❌ Error: libusb_open()❌  failed with LIBUSB_ERROR_ACCESS
Error: libusb_open() failed with LIBUSB_ERROR_NOT_SUPPORTED
❌ Error: libusb_open() fai❌ led with LIBUSB_ERROR_ACCESS
Error: libusb_open() failed with LIBUSB_ERROR_NOT_SUPPORTED
❌ Error: libusb_open(❌ ) failed with LIBUSB_ERROR_ACCESS
Error: libusb_open() failed with LIBUSB_ERROR_NOT_SUPPORTED
❌ Error: libusb_open() fail❌ ed with LIBUSB_ERROR_ACCESS
Error: libusb_open() failed with LIBUSB_ERROR_NOT_SUPPORTED
❌ Error: libusb_open() failed with L❌ IBUSB_ERROR_ACCESS
Error: libusb_open() failed with LIBUSB_ERROR_NOT_SUPPORTED
❌ Error: esp_usb_jtag: ❌ device not found!
Error: esp_usb_jtag: failed to revive USB device!
Error: libusb_bulk_write error: LIBUSB_ERROR_NOT_FOUND
❌ Error: libusb_open() failed with LIBUSB_ERROR_ACCESS
Error: libusb_open() failed with LIBUSB_ERROR_NOT_SUPPORTED
❌ Error: libusb_open() ❌ failed with LIBUSB_ERROR_ACCESS
Error: libusb_open() failed with LIBUSB_ERROR_NOT_SUPPORTED
❌ Error: libusb_open() failed with❌  LIBUSB_ERROR_ACCESS
Error: libusb_open() failed with LIBUSB_ERROR_NOT_SUPPORTED
❌ Error: libusb_open() ❌ failed with LIBUSB_ERROR_ACCESS
Error: libusb_open() failed with LIBUSB_ERROR_NOT_SUPPORTED
❌ Error: libusb_open() fa❌ iled with LIBUSB_ERROR_ACCESS
Error: libusb_open() failed with LIBUSB_ERROR_NOT_SUPPORTED
❌ Error: libusb_open() f❌ ailed with LIBUSB_ERROR_ACCESS
Error: libusb_open() failed with LIBUSB_ERROR_NOT_SUPPORTED
❌ Error: esp_usb_jtag: device not f❌ ound!
Error: esp_usb_jtag: failed to revive USB device!
[... 106 lines cut ...]
Error: libusb_open() failed with LIBUSB_ERROR_NOT_SUPPORTED
❌ Error: libusb_open() failed❌  with LIBUSB_ERROR_ACCESS
Error: libusb_open() failed with LIBUSB_ERROR_NOT_SUPPORTED
❌ Error: esp_usb_jtag: device not found!
❌ Error: esp_usb_jtag: failed to revive USB device!
Error: libusb_bulk_write error: LIBUSB_ERROR_NOT_FOUND
❌ Error: libusb_open() failed with LIBUSB_ERROR_ACCESS
Error: libusb_open() failed with LIBUSB_ERROR_NOT_SUPPORTED
❌ Error: libusb_ope❌ n() failed with LIBUSB_ERROR_ACCESS
Error: libusb_open() failed with LIBUSB_ERROR_NOT_SUPPORTED
❌ Error: libusb_open() failed ❌ with LIBUSB_ERROR_ACCESS
Error: libusb_open() failed with LIBUSB_ERROR_NOT_SUPPORTED
❌ Error: libusb_open() failed with LIB❌ USB_ERROR_ACCESS
Error: libusb_open() failed with LIBUSB_ERROR_NOT_SUPPORTED
❌ Error: libusb_open() failed wi❌ th LIBUSB_ERROR_ACCESS
Error: libusb_open() failed with LIBUSB_ERROR_NOT_SUPPORTED
❌ Error: libusb_open() fail❌ ed with LIBUSB_ERROR_ACCESS
Error: libusb_open() failed with LIBUSB_ERROR_NOT_SUPPORTED
❌ Error: esp_usb_jtag: device no❌ t found!
Error: esp_usb_jtag: failed to revive USB device!
Error: missing data from bitq interface
Warn : target esp32s3.cpu1 examination failed
Info : starting gdb server for esp32s3.cpu0 on 3333
Info : Listening on port 3333 for gdb connections
```

TDA-2030, 2022-05-18:
可能是没有正确配置，https://docs.espressif.com/projects/esp-idf/zh_CN/latest/esp32s3/api-guides/jtag-debugging/configure-builtin-jtag.html 可能对你有帮助

ESP-Mars, 2023-06-30:
Hi, if you don't have any questions, please close this issue. Thank you!

closed 2023-07-11 by ESP-Mars