espressif/esp-box #32: Running esp-box examples on a ESP32-S3-DevKitM-1

armwdev, 2022-05-27:
I think it should be possible to run esp-box examples on S3 dev boards. I would like to reduce the partition sizes and remove some features such as playing music files, codec staff and even the AI voice recognition. 

I've tried to make these examples work on the ESP32-S3-DevKitM-1 dev board that has 8 MB SPI flash with LCD connected to GPIO but the device keeps resetting for the image_display example and shows invalid header: 0x00010001 in the terminal for the factory_demo example.

This should be possible so I can start a proper development using these examples as a starting point since the esp-box and the S3 dev boards has the same chip. 

Actually all I need is a working LVGL GUI with 3 physical buttons, wifi setup and ESP BOX mobile APP integration. By the way, how can we modify the ESP BOX mobile APP?

How can this be achieved? Any help would be appreciated.

TDA-2030, 2022-05-30:
ESP32-S3-DevKitM-1 has almost no peripherals, and removing functionality from `factory_demo` exactly doesn't seem like an easy task. If you only need LVGL and Rainmaker, then I think it might be easier for you to port the related components and code to your own projects. Plus, you'll be great to use our upcoming [Box-Lite](https://github.com/espressif/esp-box/blob/master/docs/hardware_overview/esp32_s3_box_lite/hardware_overview_for_lite.md) hardware.

The ESP-BOX APP is not open source. This app is for developers to experience and use.  It is developed based on the native ESP Rainmaker app (open source link https://github.com/espressif/esp-rainmaker-ios ）With the same functions, only the UI is changed and some optimization items are added.

ESP-Mars, 2022-06-08:
esp32-s3-box-lite now can be accessed through https://www.digikey.co.uk/en/products/detail/espressif-systems/ESP32-S3-BOX-LITE/15967391?s=N4IgTCBcDaIKYGcAOBmMBaBL0CMD2AHugDYCWALnCALoC%2BQA

armwdev, 2022-06-08:
Thanks for the reply. What do you think about my other concerns? I hope the upcoming ESP-IDF versions or a new hardware releases does not obsolete my current setup in couple of years so I can rely on this.

alic-maker, 2022-06-09:
> I also couldn't make my 3 years old esp-iot-solution project to work on this board. It was working perfectly on ESP32 but not sure what to change so I can make it work on ESP32-S3. It uses an old ESP-IDF version and no cmake at that time. I've tried the python conversion script but it failed. Basically everything fails if I try to make this work on the new ESP-IDF and boards.

Only newer ESP-IDF support S3 chips, however, newer versions of IDF are not compatible with older versions, so you can have a lot of problems using older code. You need to update your project code to use ESP32-S3. This [document](https://docs.espressif.com/projects/esp-idf/en/latest/esp32s3/migration-guides/index.html#esp-idf-5-0-migration-guides) is helpful if you are using IDF 5.0

ESP-Mars, 2022-06-09:
> I thought the Rainmaker is an open source solution but when I checked the details, I noticed it says about evaluation and pricing. Isn't there any community or free option of the Rainmaker or somehow I can host the server files myself instead of trusting a third party?

https://github.com/espressif/esp-rainmaker is our public community that supports user download general firmware and related open source App. We have built a public server on AWS for developers as public RainMaker cloud, each account will be assigned five free self-claiming node to experience.

If you want to host your files on your personal cloud, a formal contract should be signed, We will deploy a private cloud based on the your AWS account. Espressif ESP-RainMaker service you can understand it as a paid solution to deploy private cloud for customers based on customer’s AWS account. (https://blog.espressif.com/esp-rainmaker-d88167000285)

armwdev, 2022-06-11:
Ok thanks, closing this for now since I have different issue: https://github.com/espressif/esp-iot-solution/issues/192

closed 2022-06-11 by armwdev