espressif/esp-box #33: Demo V0.3.0 FW showing Beijing time instead of Lisbon time (utc+1)

PaulskPt, 2022-06-18:
After flashing the latest demo firmware (V0.3.0), ESP-IDF Ver: v4.4-264-g2bdea81b2A, SR LANG: English, Board: S3_BOX,
after installing the APP in an iPhone8Plus, after Pairing BlueTooth, the S3_BOX is showing Beijing time instead of Lisbon (Portugal) time (= utc +1).

[image]

PaulskPt, 2022-06-18:
Device info on APP

[image]

PaulskPt, 2022-06-18:
For me the issue is solved: from within a Ubuntu Linux session I re-flashed the file: ESP_BOX_Demo_EN_V0.3.0.bin. After flashing I reset the device. I saw the Lisbon time coming up, then for a second or two I saw the Beijing time, finally came back the Lisbon time.

TDA-2030, 2022-06-19:
Currently, the Time Zone is hardcoded in [app_sntp.c](https://github.com/espressif/esp-box/blob/master/examples/factory_demo/main/app/app_sntp.c#L56) rather than automatically grab from somewhere. It's not smart enough :joy:

PaulskPt, 2022-06-19:
Hi Li (@TDA-2030), thank you very much for your rapid reply. Excuse me, I can almost not believe this! Such a nice product the ESP32-S3-BOX is. Why you guys hard coded the time? Probably no time to spend on the clock. Sure, the speech recognition part had much more priority, I think. However, by now the ESP32-S3-BOX is maybe sold all over the world! So what all the customers in the rest of the world do with Beijing time?  I had to do a lot of effort to get my hands on one of these ESP32-S3-BOX'es. At first I ordered one with mouser.com,  U.S.A. However they told me they could not sell it to me because 'the manufacturer decided to sell the product only to OEM companies', which I am not (I am a retired individual). Then I saw that Adafruit Industries of New York, U.S.A. was advertising the ESP32-S2-BOX, so I ordered one box with them. I received it a few days ago. Since about a year I am experimenting with M5Stack units. I have various M5Stack products. I have a CORE-2 in which I programmed an 'NTP Synchronized Clock'.  (it is using an Arduino sketch). It uses the file that has the WiFi SSID and Password: ```secrets.py``` to indicate the TimeZone that you want the clock to be sychronized to. See: ```https://github.com/PaulskPt/M5Stack-NTP-Clock/tree/NTP_Client_w_sleep```. This sketch uses the touch screen to put the device into a sleep mode and after another touch to wake-up; a software touch button to force synchronize datetime from NTP-server and another software touch button to show the temperature of the microprocessor (just for fun). I think that it can be adapted to function in the ESP32-S3-BOX, or just extract needed parts from it to incorporate into the software of the ESP32-S3-BOX.
Greetings,
Paulus 

[image]

ESP-Mars, 2022-06-20:
Thanks for your interesting in the project. We wholeheartedly endorse your opinion and we will improve this features in the next iteration.

PaulskPt, 2022-06-22:
I managed to create, build and flash a patch to ap_sntp.c using conditional #ifdef USE_TIME_ASIA, USE_TIME_USA and USE_TIME_EURO. But this is just a temporary patch until Espressif comes with the promised improvement on setting a local timezone in the next iteration.