espressif/esp-box #36: Can't flash new firmware to ESP32-S3-BOX

WH3080, 2022-06-24:
Hello, I just received the ESP32-S3-BOX. Following the recommendations I was hoping I could upgrade the latest firmware according to these instructions: https://github.com/espressif/esp-box/blob/master/docs/firmware_update.md Although I have used this tool successfully multiple times on various other ESP devices, it does not work here. And yes, I also tried holding the boot button etc. It initializes the upload:

AP:  7CDFA1E1C781  STA:  7CDFA1E1C780
BT:  7CDFA1E1C782  ETHERNET:  7CDFA1E1C783

... and then I always get this error:

Uploading stub...
Running stub...
Stub running...

is stub and send flash finish

Is there any other thing I could try? It's not the cable or computer as I was also able to flash other firmware onto it (Tasmota via the Tasmota Web Installer for instance). 

Thank you for your help!

TDA-2030, 2022-06-25:
You can try `Disable download stub` in menuconfig like the following diagram.
[image]

WH3080, 2022-06-25:
Wich software is that?

WH3080, 2022-06-25:
I realized that I needed to install the ESP IDF and found menuconfig command to see this screen and follow your advice. It worked.

beginner140BB, 2022-11-11:
Is it possible to flash through espressif IDE I don,t get how to do that
When i have run the command (git clone --recursive https://github.com/espressif/esp-box.git) 
esp-box folder with required elements are created 
so how can i run modify code using the IDE.
i able to flash using CMD  commands.

espressif2022, 2022-11-11:
> Is it possible to flash through espressif IDE I don,t get how to do that When i have run the command (git clone --recursive https://github.com/espressif/esp-box.git) esp-box folder with required elements are created so how can i run modify code using the IDE. i able to flash using CMD commands.

you want to do some modification in box, and don't know how to install IDF? 
you can refer to this: [installation-step-by-step](https://docs.espressif.com/projects/esp-idf/en/release-v4.4/esp32s3/get-started/index.html#installation-step-by-step)

beginner140BB, 2022-11-11:
> you want to do some modification in box, and don't know how to install IDF? you can refer to this: [...]

currently working in espressif ide(windows) projects based on esp32
is it possible to modify the example code given inside the box
esp32 s3 board is already there.

beginner140BB, 2022-11-11:
does the location of the folder is wrong..?

> C:\Espressif\frameworks\esp-idf-v4.4.2\esp-box

espressif2022, 2022-11-11:
> 
IDF is the same, you should press cmd to change build target:
**idf.py set-target esp32s3**
so, what's the problem now?  I'm not very clear

beginner140BB, 2022-11-11:
> so, what's the problem now? I'm not very clear
Need to use ESPRESSIF IDE to flash the code and modify the code of esp-box other than using ESP-IDF command promt.
what should i do for that?
initial step followed
was (git clone --recursive https://github.com/espressif/esp-box.git)
then try to open the example for building it what may be the issue here.

[image]

espressif2022, 2022-11-11:
> [External: This email originated outside Espressif]

Sorry, I didn’t use this IDE before.
I asked my colleagues, this is the IDE made by the previous agent,
I am not sure whether it has been adapted to the latest chip of S3, this is not our official recommended compilation method, if it is a window environment, you can execute it here [window install step]<https://docs.espressif.com/projects/esp-idf/en/latest/esp32/get-started/windows-setup.html#get-started-windows-first-steps>
And linux is recommended, window is too slow;
***@***.*** ***@***.***
***@***.***

Best regards,

徐新

AE｜Espressif Systems

T: 021-61091062

www.espressif.com<http://espressif.com/>
#204, Block 2, 690 Bibo Road,

Zhang Jiang High-Tech Park, Shanghai, China

[image]<https://user-images.githubusercontent.com/83206454/201333081-eb728970-79eb-4cf0-9022-cee43e60ed1f.png>

beginner140BB, 2022-11-11:
> 

okay thanks, @espressif2022 
so for development side you guys are using external editor and for building you use esp idf cmd right.
so there is no possible way to remodify compile the code using ide

beginner140BB, 2022-11-11:
using the ide is more easy way thats  why i asked for that to modify code of esp-box using ide since currently we have been using esp32 based projects using the ide.

is there any other ide that can be used for that such as visual studio code.

espressif2022, 2022-11-11:
> 
sorry, you did it with ESP-IDF Eclipse Plugin? [ESP-IDF Eclipse Plugin](https://github.com/espressif/idf-eclipse-plugin/blob/master/README.md)

beginner140BB, 2022-11-11:
> sorry, you did it with ESP-IDF Eclipse Plugin? [ESP-IDF Eclipse Plugin](https://github.com/espressif/idf-eclipse-plugin/blob/master/README.md)

yes

beginner140BB, 2022-11-11:
> I am not sure whether it has been adapted to the latest chip of S3

[image]

espressif2022, 2022-11-11:
> 
can you build it ok?

> > I am not sure whether it has been adapted to the latest chip of S3
> 
> [image]

can you build the examples successfully?

beginner140BB, 2022-11-16:
> can you build the examples successfully?

yes,
need to know how to include box exampple code to build and flash using IDE rather than using ESP-IDF CMD

ESP-Mars, 2023-06-30:
Hi, if you don't have any questions, please close this issue. Thank you!

closed 2023-07-11 by ESP-Mars