espressif/esp-box #38: rebooting problem

Charlesguo11, 2022-07-06:
Quantized8 Multinet5: MN5Q8_v2_english_8_0.9_0.90, (Feb 17 2022 11:49:42)
I (13514) app_sr: cmd_number=15
I (13514) app_sr: New 15 command set to :[TkN nN jc LiT;SWgp nN jc LiT;SWgp eF jc LiT;TkN eF jc LiT;TkN RfD;TkN GRmN;TkN BLo;KcSTcMiZ KcLk;Sgl c Sel;PLd MYoZgK;NfKST Sel;PeZ PLdgl;RmqgST YkSfLF;RmMoV YkSfLF;KeL FeR hfLP;;]
I (13515) MN: ---------------------SPEECH COMMANDS---------------------
I (13516) MN: Command ID0, phrase ID0: TkN nN jc LiT
I (13516) MN: Command ID1, phrase ID1: SWgp nN jc LiT
I (13516) MN: Command ID2, phrase ID2: SWgp eF jc LiT
I (13517) MN: Command ID3, phrase ID3: TkN eF jc LiT
I (13517) MN: Command ID4, phrase ID4: TkN RfD
I (13517) MN: Command ID5, phrase ID5: TkN GRmN
I (13517) MN: Command ID6, phrase ID6: TkN BLo
I (13518) MN: Command ID7, phrase ID7: KcSTcMiZ KcLk
I (13518) MN: Command ID8, phrase ID8: Sgl c Sel
I (13518) MN: Command ID9, phrase ID9: PLd MYoZgK
I (13519) MN: Command ID10, phrase ID10: NfKST Sel
I (13519) MN: Command ID11, phrase ID11: PeZ PLdgl
I (13519) MN: Command ID12, phrase ID12: RmqgST YkSfLF
I (13520) MN: Command ID13, phrase ID13: RmMoV YkSfLF
I (13520) MN: Command ID14, phrase ID14: KeL FeR hfLP
I (13520) MN: ---------------------------------------------------------

I (13522) app_sr: audio_chunksize=1024, feed_channel=3

assert failed: tcpip_send_msg_wait_sem IDF/components/lwip/lwip/src/api/tcpip.c:455 (Invalid mbox)

Backtrace:0x40379e36:0x3fcc5e600x4038d1b5:0x3fcc5e80 0x403932dd:0x3fcc5ea0 0x420b1b43:0x3fcc5fc0 0x420bc995:0x3fcc5ff0 0x420bca1e:0x3fcc6010 0x420b11a4:0x3fcc6060 0x42008101:0x3fcc6080 0x420027d8:0x3fcc60d0 0x4038d671:0x3fcc60f0 

I have an esp-box,I use the chip for dual-threaded work, but the chip keeps rebooting after programming.
And the error is shown above.
How can I solve this problem?

TDA-2030, 2022-07-07:
It seems that you have not initialized the network protocol stack when using TCP/UDP. You should pay attention to the order in which the code runs

Charlesguo11, 2022-07-07:
Thanks! 
I used sleep(5) to stuck my second thread until the network stack is established.

closed 2022-07-07 by Charlesguo11