espressif/esp-box #39: mp3 player problem

Charlesguo11, 2022-07-07:
Hi!
I have an esp-box, I used two thread on this chip.
And I copied the function play_mp3() in my second thread.Then I used a flag to trigger it.
However, every time I trigger it,it would stuck like this:

I (39665) main: start to decode /spiffs/mp3/warning.mp3
I (39679) main: audio info: sr=48000, bit=16, ch=1

I think the problem might be this line: "i2s_channel_t channel = (nChans == 1) ? I2S_CHANNEL_MONO : I2S_CHANNEL_STEREO;"
How can I solve this question?Thanks!

Charlesguo11, 2022-07-08:
I think the problem is this line:
"i2s_set_clk(I2S_NUM_0, sample_rate, bits_cfg, channel);"

chmorgan, 2022-07-13:
Hi @Charlesguo11, let me push some code tonight. I’ve been working on a rewrite of the audio player and it may be that the issue is solved due to the rewrite. A number of issues with mp3 playback were resolved.

Charlesguo11, 2022-07-17:
@chmorgan Hi，it would be great if the problems were solved. But,when can i have the new version of it?
Thanks a lot!

chmorgan, 2022-07-17:
@Charlesguo11 https://github.com/chmorgan/esp-box/tree/audio_playlist_component

is the work in progress branch. There is still an issue with mp3-demo related to playback switching on the ui side, something gets out of sync and the UI isn't in the correct state. But the audio_player component itself, and the playlist components should be good to go.

Let me know what you think and how it works.

chmorgan, 2023-03-05:
Updated mp3 playback player has been merged with a number of fixes for differing sample rates and proper handling of the final bytes to decode.

Can an admin close this one for that and lack of activity?

closed 2023-03-06 by Charlesguo11