espressif/esp-box #40: Build problems Demo Software for S3-BOX in Eclipse

vojkob, 2022-08-03:
Hi,

I installed from sckratch new ESPIDF and EClipse plugin for ESP32, imported projects from Examples folder on ESP32-S3-BOX  GIT, but when I build whatever example I get this message during compiling:

CMake Error at D:/ESP32IDE/esp-idf-v4.4.1/tools/cmake/component.cmake:301 (message):
  Include directory
  'D:/ESP32PRojects/S3-BOX/esp-box-master/components/lvgl/lvgl/src' is not a
  directory.
Call Stack (most recent call first):
  D:/ESP32IDE/esp-idf-v4.4.1/tools/cmake/component.cmake:472 (__component_add_include_dirs)
  D:/ESP32PRojects/S3-BOX/esp-box-master/components/lvgl/CMakeLists.txt:27 (idf_component_register)

-- Configuring incomplete, errors occurred!
See also "D:/ESP32PRojects/S3-BOX/esp-box-master/examples/image_display/build/CMakeFiles/CMakeOutput.log".
cmake --build . -- -v
ninja: error: loading 'build.ninja': The system cannot find the file specified.

Build complete (0 errors, 0 warnings): D:\ESP32PRojects\S3-BOX\esp-box-master\examples\image_display\build
Total time taken to build the project: 12,567 ms

Does anybody have idea what can be the reason?

Thanks,
Vojko

igrr, 2022-08-03:
Hi Vojko, it looks like you have downloaded the .zip file of this repository from Github.

Unfortunately this won't work, since this project uses Git submodules, but Github doesn't include the contents of these submodules in the downloadable .zip files.

Instead, please use `git clone --recursive ...` command given in the quickstart section: https://github.com/espressif/esp-box#quick-start. This will clone the project along with the submodules.

vojkob, 2022-08-03:
Hi Ivan,

 

Uf, Yes, I’m stupid. Thanks, I will try with clone.

 

Thanks a lot,

Vojko

vojkob, 2022-08-04:
Hi,

 

I made clone on my disk, the compiling is running, but now I got typical “undefined symbol “ message, which should not occur on valid build. 

D:/ESP32PRojects/ESP32_S3_BOX/components/bsp/src/boards/esp32_s3_box_lite.c:52:25: error: 'GPIO_NUM_45' undeclared here (not in a function); did you mean 'GPIO_NUM_35'?

     .GPIO_LCD_BL =     (GPIO_NUM_45),

                         ^~~~~~~~~~~

                         GPIO_NUM_35

D:/ESP32PRojects/ESP32_S3_BOX/components/bsp/src/boards/esp32_s3_box_lite.c:55:25: error: 'GPIO_NUM_48' undeclared here (not in a function); did you mean 'GPIO_NUM_38'?

     .GPIO_LCD_RST =    (GPIO_NUM_48),

                         ^~~~~~~~~~~

                         GPIO_NUM_38

D:/ESP32PRojects/ESP32_S3_BOX/components/bsp/src/boards/esp32_s3_box_lite.c:105:29: error: 'GPIO_NUM_47' undeclared here (not in a function); did you mean 'GPIO_NUM_37'?

     .GPIO_I2S_LRCK =       (GPIO_NUM_47),

                             ^~~~~~~~~~~

                             GPIO_NUM_37

D:/ESP32PRojects/ESP32_S3_BOX/components/bsp/src/boards/esp32_s3_box_lite.c:116:29: error: 'GPIO_NUM_46' undeclared here (not in a function); did you mean 'GPIO_NUM_36'?

     .GPIO_PWR_CTRL =       (GPIO_NUM_46),

                             ^~~~~~~~~~~

                             GPIO_NUM_36

 

 

Thanks for reply,

 

Vojko

vojkob, 2022-08-04:
Hi,

 

Here is also list of problems. There are some additional unresolved definitions:

Description         Resource             Path       Location               Type

'GPIO_NUM_45' undeclared here (not in a function); did you mean 'GPIO_NUM_35'?   factory_demo                   line 47, external location: D:\ESP32PRojects\ESP32_S3_BOX\components\bsp\src\boards\esp32_s3_box.c C/C++ Problem

'GPIO_NUM_45' undeclared here (not in a function); did you mean 'GPIO_NUM_35'?   factory_demo                   line 52, external location: D:\ESP32PRojects\ESP32_S3_BOX\components\bsp\src\boards\esp32_s3_box_lite.c                C/C++ Problem

'GPIO_NUM_46' undeclared here (not in a function); did you mean 'GPIO_NUM_36'?   factory_demo                   line 111, external location: D:\ESP32PRojects\ESP32_S3_BOX\components\bsp\src\boards\esp32_s3_box.c                C/C++ Problem

'GPIO_NUM_46' undeclared here (not in a function); did you mean 'GPIO_NUM_36'?   factory_demo                   line 116, external location: D:\ESP32PRojects\ESP32_S3_BOX\components\bsp\src\boards\esp32_s3_box_lite.c                C/C++ Problem

'GPIO_NUM_47' undeclared here (not in a function); did you mean 'GPIO_NUM_37'?   factory_demo                   line 100, external location: D:\ESP32PRojects\ESP32_S3_BOX\components\bsp\src\boards\esp32_s3_box.c                C/C++ Problem

'GPIO_NUM_47' undeclared here (not in a function); did you mean 'GPIO_NUM_37'?   factory_demo                   line 105, external location: D:\ESP32PRojects\ESP32_S3_BOX\components\bsp\src\boards\esp32_s3_box_lite.c                C/C++ Problem

'GPIO_NUM_48' undeclared here (not in a function); did you mean 'GPIO_NUM_38'?   factory_demo                   line 50, external location: D:\ESP32PRojects\ESP32_S3_BOX\components\bsp\src\boards\esp32_s3_box.c C/C++ Problem

'GPIO_NUM_48' undeclared here (not in a function); did you mean 'GPIO_NUM_38'?   factory_demo                   line 55, external location: D:\ESP32PRojects\ESP32_S3_BOX\components\bsp\src\boards\esp32_s3_box_lite.c                C/C++ Problem

Symbol 'ESP_ERR_INVALID_ARG' could not be resolved                esp32_s3_box.c                /factory_demo/esp_idf_components/BOX/bsp/src/boards        line 192 Semantic Error

Symbol 'ESP_FAIL' could not be resolved              esp32_s3_box.c                /factory_demo/esp_idf_components/BOX/bsp/src/boards        line 165 Semantic Error

Symbol 'ESP_OK' could not be resolved esp32_s3_box.c                /factory_demo/esp_idf_components/BOX/bsp/src/boards        line 159 Semantic Error

Symbol 'ESP_OK' could not be resolved esp32_s3_box.c                /factory_demo/esp_idf_components/BOX/bsp/src/boards        line 166 Semantic Error

Symbol 'ESP_OK' could not be resolved esp32_s3_box.c                /factory_demo/esp_idf_components/BOX/bsp/src/boards        line 177 Semantic Error

Symbol 'ESP_OK' could not be resolved esp32_s3_box.c                /factory_demo/esp_idf_components/BOX/bsp/src/boards        line 195 Semantic Error

Symbol 'false' could not be resolved       esp32_s3_box.c                /factory_demo/esp_idf_components/BOX/bsp/src/boards        line 44   Semantic Error

Symbol 'GPIO_NUM_45' could not be resolved esp32_s3_box.c                /factory_demo/esp_idf_components/BOX/bsp/src/boards        line 47   Semantic Error

Symbol 'GPIO_NUM_46' could not be resolved esp32_s3_box.c                /factory_demo/esp_idf_components/BOX/bsp/src/boards        line 111 Semantic Error

Symbol 'GPIO_NUM_47' could not be resolved esp32_s3_box.c                /factory_demo/esp_idf_components/BOX/bsp/src/boards        line 100 Semantic Error

Symbol 'GPIO_NUM_48' could not be resolved esp32_s3_box.c                /factory_demo/esp_idf_components/BOX/bsp/src/boards        line 50   Semantic Error

Symbol 'NULL' could not be resolved      esp32_s3_box.c                /factory_demo/esp_idf_components/BOX/bsp/src/boards        line 145 Semantic Error

Symbol 'true' could not be resolved        esp32_s3_box.c                /factory_demo/esp_idf_components/BOX/bsp/src/boards        line 42   Semantic Error

Symbol 'true' could not be resolved        esp32_s3_box.c                /factory_demo/esp_idf_components/BOX/bsp/src/boards        line 43   Semantic Error

Type 'bool' could not be resolved             esp32_s3_box.c                /factory_demo/esp_idf_components/BOX/bsp/src/boards        line 162 Semantic Error

 

 

Regards,

Vojko

alic-maker, 2022-08-04:
Maybe the target is not set to `esp32s3`, so those GPIOs are not defined.

vojkob, 2022-08-04:
Hi,

 

Hm, I installed last ESP IDF tools, but I can not select ESP32-s3 in as target in Eclipse. Just ESP and ESP S2.

 

Regards,Vojko

vojkob, 2022-08-04:
Hi,

OK, I managed to change target in Eclipse to S3, after 5 minutes of compile time, it seems there are no errors. Thanks a lot.

Regards,
Vojko

vojkob, 2022-08-04:
Hi,

actually there are errors, but this Eclipse is very nasty IDE. Not show anything in Console, just in Problems section:
Description	Resource	Path	Location	Type
Symbol 'ADC1_CHANNEL_0' could not be resolved	esp32_s3_box.c	/factory_demo/esp_idf_components/BOX/bsp/src/boards	line 64	Semantic Error
Symbol 'ESP_ERR_INVALID_ARG' could not be resolved	esp32_s3_box.c	/factory_demo/esp_idf_components/BOX/bsp/src/boards	line 192	Semantic Error
Symbol 'ESP_FAIL' could not be resolved	esp32_s3_box.c	/factory_demo/esp_idf_components/BOX/bsp/src/boards	line 165	Semantic Error
Symbol 'ESP_OK' could not be resolved	esp32_s3_box.c	/factory_demo/esp_idf_components/BOX/bsp/src/boards	line 159	Semantic Error
Symbol 'ESP_OK' could not be resolved	esp32_s3_box.c	/factory_demo/esp_idf_components/BOX/bsp/src/boards	line 166	Semantic Error
Symbol 'ESP_OK' could not be resolved	esp32_s3_box.c	/factory_demo/esp_idf_components/BOX/bsp/src/boards	line 177	Semantic Error
Symbol 'ESP_OK' could not be resolved	esp32_s3_box.c	/factory_demo/esp_idf_components/BOX/bsp/src/boards	line 195	Semantic Error
Symbol 'false' could not be resolved	esp32_s3_box.c	/factory_demo/esp_idf_components/BOX/bsp/src/boards	line 44	Semantic Error
Symbol 'GPIO_NUM_0' could not be resolved	esp32_s3_box.c	/factory_demo/esp_idf_components/BOX/bsp/src/boards	line 15	Semantic Error
Symbol 'GPIO_NUM_1' could not be resolved	esp32_s3_box.c	/factory_demo/esp_idf_components/BOX/bsp/src/boards	line 114	Semantic Error
Symbol 'GPIO_NUM_10' could not be resolved	esp32_s3_box.c	/factory_demo/esp_idf_components/BOX/bsp/src/boards	line 78	Semantic Error
Symbol 'GPIO_NUM_11' could not be resolved	esp32_s3_box.c	/factory_demo/esp_idf_components/BOX/bsp/src/boards	line 75	Semantic Error
Symbol 'GPIO_NUM_12' could not be resolved	esp32_s3_box.c	/factory_demo/esp_idf_components/BOX/bsp/src/boards	line 77	Semantic Error
Symbol 'GPIO_NUM_13' could not be resolved	esp32_s3_box.c	/factory_demo/esp_idf_components/BOX/bsp/src/boards	line 74	Semantic Error
Symbol 'GPIO_NUM_14' could not be resolved	esp32_s3_box.c	/factory_demo/esp_idf_components/BOX/bsp/src/boards	line 76	Semantic Error
Symbol 'GPIO_NUM_15' could not be resolved	esp32_s3_box.c	/factory_demo/esp_idf_components/BOX/bsp/src/boards	line 104	Semantic Error
Symbol 'GPIO_NUM_16' could not be resolved	esp32_s3_box.c	/factory_demo/esp_idf_components/BOX/bsp/src/boards	line 103	Semantic Error
Symbol 'GPIO_NUM_17' could not be resolved	esp32_s3_box.c	/factory_demo/esp_idf_components/BOX/bsp/src/boards	line 102	Semantic Error
Symbol 'GPIO_NUM_18' could not be resolved	esp32_s3_box.c	/factory_demo/esp_idf_components/BOX/bsp/src/boards	line 69	Semantic Error
Symbol 'GPIO_NUM_2' could not be resolved	esp32_s3_box.c	/factory_demo/esp_idf_components/BOX/bsp/src/boards	line 101	Semantic Error
Symbol 'GPIO_NUM_39' could not be resolved	esp32_s3_box.c	/factory_demo/esp_idf_components/BOX/bsp/src/boards	line 97	Semantic Error
Symbol 'GPIO_NUM_4' could not be resolved	esp32_s3_box.c	/factory_demo/esp_idf_components/BOX/bsp/src/boards	line 51	Semantic Error
Symbol 'GPIO_NUM_5' could not be resolved	esp32_s3_box.c	/factory_demo/esp_idf_components/BOX/bsp/src/boards	line 49	Semantic Error
Symbol 'GPIO_NUM_6' could not be resolved	esp32_s3_box.c	/factory_demo/esp_idf_components/BOX/bsp/src/boards	line 53	Semantic Error
Symbol 'GPIO_NUM_7' could not be resolved	esp32_s3_box.c	/factory_demo/esp_idf_components/BOX/bsp/src/boards	line 52	Semantic Error
Symbol 'GPIO_NUM_8' could not be resolved	esp32_s3_box.c	/factory_demo/esp_idf_components/BOX/bsp/src/boards	line 70	Semantic Error
Symbol 'GPIO_NUM_9' could not be resolved	esp32_s3_box.c	/factory_demo/esp_idf_components/BOX/bsp/src/boards	line 79	Semantic Error
Symbol 'GPIO_NUM_NC' could not be resolved	esp32_s3_box.c	/factory_demo/esp_idf_components/BOX/bsp/src/boards	line 54	Semantic Error
Symbol 'GPIO_NUM_NC' could not be resolved	esp32_s3_box.c	/factory_demo/esp_idf_components/BOX/bsp/src/boards	line 80	Semantic Error
Symbol 'GPIO_NUM_NC' could not be resolved	esp32_s3_box.c	/factory_demo/esp_idf_components/BOX/bsp/src/boards	line 84	Semantic Error
Symbol 'GPIO_NUM_NC' could not be resolved	esp32_s3_box.c	/factory_demo/esp_idf_components/BOX/bsp/src/boards	line 85	Semantic Error
Symbol 'GPIO_NUM_NC' could not be resolved	esp32_s3_box.c	/factory_demo/esp_idf_components/BOX/bsp/src/boards	line 86	Semantic Error
Symbol 'GPIO_NUM_NC' could not be resolved	esp32_s3_box.c	/factory_demo/esp_idf_components/BOX/bsp/src/boards	line 87	Semantic Error
Symbol 'GPIO_NUM_NC' could not be resolved	esp32_s3_box.c	/factory_demo/esp_idf_components/BOX/bsp/src/boards	line 90	Semantic Error
Symbol 'GPIO_NUM_NC' could not be resolved	esp32_s3_box.c	/factory_demo/esp_idf_components/BOX/bsp/src/boards	line 91	Semantic Error
Symbol 'GPIO_NUM_NC' could not be resolved	esp32_s3_box.c	/factory_demo/esp_idf_components/BOX/bsp/src/boards	line 92	Semantic Error
Symbol 'GPIO_NUM_NC' could not be resolved	esp32_s3_box.c	/factory_demo/esp_idf_components/BOX/bsp/src/boards	line 93	Semantic Error
Symbol 'GPIO_NUM_NC' could not be resolved	esp32_s3_box.c	/factory_demo/esp_idf_components/BOX/bsp/src/boards	line 96	Semantic Error
Symbol 'NULL' could not be resolved	esp32_s3_box.c	/factory_demo/esp_idf_components/BOX/bsp/src/boards	line 145	Semantic Error
Symbol 'true' could not be resolved	esp32_s3_box.c	/factory_demo/esp_idf_components/BOX/bsp/src/boards	line 42	Semantic Error
Symbol 'true' could not be resolved	esp32_s3_box.c	/factory_demo/esp_idf_components/BOX/bsp/src/boards	line 43	Semantic Error
Type 'bool' could not be resolved	esp32_s3_box.c	/factory_demo/esp_idf_components/BOX/bsp/src/boards	line 162	Semantic Error

vojkob, 2022-08-04:
Hi,

I expected  the compilation and download will be quicker. I see no big advantage in using ESP-IDE over Arduino, since I;m not digging to deep into ESP hardware. Can You help me with necessary step, too make S3-BOX arduino enabled?

Thanks,
Vojko

Jason2866, 2022-09-01:
@vojkob ESP-IDF is not Arduino. Two different platforms.

ESP-Mars, 2023-06-30:
Hi, if you don't have any questions, please close this issue. Thank you!

vojkob, 2023-07-01:
I switch completely to Arduino Visual Studio Code, Platformio, I use some native libraries where necessary and system runs really good. Thanks for help anyway.

vojkob, 2023-07-01:
Thanks a lot.

closed 2023-07-01 by vojkob