espressif/esp-box #43: esp-box iot_button button_adc.c function name spelled wrong

PaulskPt, 2022-10-21:
When searching for something else I came across this:

in file: esp-box/components/iot_button/button_adc.c
lines: 169 - 181

```
static uint32_t get_adc_volatge(adc1_channel_t channel)
{
    uint32_t adc_reading = 0;
    //Multisampling
    for (int i = 0; i < NO_OF_SAMPLES; i++) {
        adc_reading += adc1_get_raw(channel);
    }
    adc_reading /= NO_OF_SAMPLES;
    //Convert adc_reading to voltage in mV
    uint32_t voltage = adc_reading; //esp_adc_cal_raw_to_voltage(adc_reading, &g_button.adc_chars);
    ESP_LOGV(TAG, "Raw: %d\tVoltage: %dmV", adc_reading, voltage);
    return voltage;
}
```

I guess the name of the function should be: `get_adc_voltage()` like the word `voltage` is used four times inside the function.
However, maybe the function is never used. I don't see it defined inside the file: button_adc.h.

espressif2022, 2022-10-21:
It's used for ESP32-S3-BOX-Lite;
the name get_adc_voltage, i'll changed it next time;
thanks!

PaulskPt, 2022-10-21:
You're welcome. We ccan close this issue.