espressif/esp-box #49: esp-box收到后确定是否正常

upupycl, 2022-11-17:
我在淘宝乐鑫店买的ESP-BOX。收到后，我有两个问题想知道。期待您的回复
I bought the esp-box in Lexin Store on Taobao. After receiving the goods, I have two questions to know. Looking forward to your reply
      
      1. 我给开发板上电之后，使用”Hi 乐鑫“唤醒了开发板，屏幕也显示唤醒成功了，但是扬声器并没有回复词”好的“，这是否是正常现象，我不确定这是不是扬声器有问题
      After I give the development board power, use "Hi Loxin" wake up the development board, the screen also shows the wake up success, but the speaker did not reply to the word "good", this is a normal phenomenon, I am not sure if this is the speaker has a problem
      2.我唤醒开发板之后说一个命令词，就得再次唤醒，但是我看说明书可以体验连续语音识别，出厂固件可以连续识别吗？怎么进入连续识别？还是说是要我自己代码开发固件烧进去才能连续识别？ 
      After I wake up the development board and say a command word, I have to wake up again, but I see the manual can experience continuous speech recognition, factory firmware can continuous recognition? How to enter continuous recognition? Or do I have to develop my own code to burn the firmware in order to continuously identify?

ESP-Mars, 2022-11-17:
1. After using "Hi 乐鑫” to wake up the device, the speaker will reply “我在”. For Continuous Speech Recognition, after waking up the device, you can say “打开电灯”, the device replies “好的”, then you continue say “关闭电灯”, the device replies “好的”, keep saying “调成红色”, the device replies “好的”. The system will automatically exit if there is no new command within 6 seconds and the device replies "我休息了，有需要再叫我”.
2. You can open the "Media Player" and try to play music to see if the speaker works properly.

If you still have problems, please contact after-sales service. Thanks.

ESP-Mars, 2022-11-17:
If you cannot find the "Media Player" function,  the device may not have been flashed the correct factory firmware, you can refer the guide to flash our factory firmware again https://github.com/espressif/esp-box/releases/tag/v0.3.0.

espressif2022, 2022-11-17:
> 我在淘宝乐鑫店买的ESP-BOX。收到后，我有两个问题想知道。期待您的回复 I bought the esp-box in Lexin Store on Taobao. After receiving the goods, I have two questions to know. Looking forward to your reply
>
> ```
>   1. 我给开发板上电之后，使用”Hi [...]

连续语音识别会有超时时间，超过6秒没声音，模型会自动回到唤醒模式，同时屏幕上会显示 “超时/timeout”，并同步播报“再见/see you next time”；
There will be a timeout period for continuous speech recognition. If there is no sound for more than 6 seconds, the model will automatically return to the wake-up mode, "超时/timeout" will be displayed on the screen, and "再见/see you next time" will be broadcast simultaneously

upupycl, 2022-11-17:
Ok.At first, thanks for your reply.
 I use idf.py -p com monitor look the log,i have found that the demo's version is v0.2.1, and then i upgrade the firmware to v0.3.0, now it can reply nomally,and can Continuous identification.

closed 2022-11-17 by upupycl

ESP-Mars, 2022-11-17:
Double check where did you see the Continuous Speech Recognition function? The paper manual?

upupycl, 2022-11-17:
yes,the paper manual have the cotent of Continuous Recognition at title 4.6

ESP-Mars, 2022-11-17:
Thanks for your reply! The factory may mix up the firmware version.