espressif/esp-box #50: sd card and mp3_demo

babadue, 2022-11-23:
Trying to incorporate sd card with the mp3_demo.   I could get the mp3_demo to play mp3 on the sd card with no problem.  Except the UI elements would not show up.  And vice versa if I got the UI working then the sd card will have issue.

The problem is down to these two initialization functions:
lv_port_init for LCD and spi_bus_initialize for the sd card.

Any hints what needed to be done?

lijunru-hub, 2022-12-05:
You shouldn't let lvgl_task keep hogging system resources. You should add vTaskDelay

babadue, 2022-12-05:
> You shouldn't let lvgl_task keep hogging system resources. You should add vTaskDelay

vTaskDelay was used.  The issue is that Box will not support SD card in SPI mode/protocol.  In MMC mode/protocol, it works fine in conjunction with active LCD.  Unless someone tells me I am wrong.

closed 2022-12-06 by babadue