espressif/esp-box #55: Algorithm example (AEC)

conferringCalm, 2022-12-26:
Hi, 
I am using a esp-box device to test live microphone feedback rejection. I have changed the original code of the algorithm example to output the audio processed through AEC directly to the speaker. But I am getting choppy/scratchy output. Is live AEC not supported by this board? Or am I doing something wrong?
Thanks

espressif2022, 2022-12-26:
Hello,
I'm not sure about the working mode you need, can you briefly describe the audio data flow you need?

conferringCalm, 2022-12-26:
I am following the AFE voice communication scenario. I am simply trying to get I2S feed from the dual MICs to be processed through the AEC algorithm and then write the audio data back to I2S to be output from the speaker.

espressif2022, 2022-12-26:
> I am following the AFE voice communication scenario. I am simply trying to get I2S feed from the dual MICs to be processed through the AEC algorithm and then write the audio data back to I2S to be output from the speaker.

I'm not sure if it's because you didn't read the AEC data, you can compare the following two points:

- In terms of hardware, SDOUT2 of this board is suspended and not connected, 

    [image]

- have you adjusted the register of ES7210_SDP_INTERFACE2_REG12, the settings here are not adjusted, i2s cannot read AEC data?

    `/* Force ADC1/2 output to SDOUT1 and ADC3/4 output to SDOUT2 */`
    ` ret |= es7210_write_reg(ES7210_SDP_INTERFACE2_REG12, 0x00);`

conferringCalm, 2022-12-26:
I checked the hardware and the SDOUT2 is not connected.

I am using the driver for es7210 given in the ADF v2.4.1. I see that the register of ES7210_SDP_INTERFACE2_REG12 has been set.
[image]

espressif2022, 2022-12-27:
> I checked the hardware and the SDOUT2 is not connected.
> 
> I am using the driver for es7210 given in the ADF v2.4.1. I see that the register of ES7210_SDP_INTERFACE2_REG12 has been set. [image]

so，when you read 4 channel, the channel 3( AEC) is null or wrong data ?

conferringCalm, 2022-12-29:
Hi, 
Sorry for late reply. I have stopped using the ADF and switched to esp-Skainet. Now AEC is working (although with some noise). The channel3 data is not wrong but I saw that _algorithm_data_gain_ in ADF is working bit differently than in esp-Skainet. I'm not sure but maybe that was where the problem was.
Anyways, thanks for the support.

closed 2022-12-29 by conferringCalm