espressif/esp-box #56: A Talking Clock in Chinese

babadue, 2023-01-07:
While going through those examples for the Box, I manage to copy and paste my way through a talking clock in Chinese.

[A Talking Clock in Chinese](https://github.com/babadue/talking_clock_chinese_esp32_s3_box)

espressif2022, 2023-01-09:
> While going through those examples for the Box, I manage to copy and paste my way through a talking clock in Chinese.
> 
> [A Talking Clock in Chinese](https://github.com/babadue/talking_clock_chinese_esp32_s3_box)

I watched the demo video, it is a pretty good idea, do you have any plans for the follow-up?

babadue, 2023-01-09:
> I watched the demo video, it is a pretty good idea, do you have any plans for the follow-up?

Probably add another feature using motion sensor and ESP32-CAM board.

babadue, 2023-01-22:
Ported over to IDF v5.0

ESP-Mars, 2023-06-30:
Hi, if you don't have any questions, please close this issue. Thank you!

closed 2023-07-11 by ESP-Mars