espressif/esp-box #57: Problem compiling the demo

ailunwowo, 2023-01-30:
Use esp-idf5.0 to compile the watering_demo project prompt: ninja: error: loading 'build.ninja': The system cannot find the specified file.
But when compiling the image_display project, no error will be reported. Why?

espressif2022, 2023-01-30:
`git submodule update --init --recursive`
please try this step first.

ailunwowo, 2023-01-30:
The result is still the same, nothing changes

espressif2022, 2023-01-30:
> The result is still the same, nothing changes

factory_demo is ok? 
can you provide your idf and project commit id?

ailunwowo, 2023-01-30:
Compilation failed

wujiangang, 2023-01-30:
Could you please provide the detailed build log via file?

ailunwowo, 2023-01-31:
Which folder in the project is the build log?

espressif2022, 2023-01-31:
`e.g. project_xx/build/log/idf_py_stdout_output_4294`
Could you provide idf and project commit id, I'll try it too.

ailunwowo, 2023-01-31:
`The system is: Windows - 10.0.19044 - AMD64
Compiling the C compiler identification source file "CMakeCCompilerId.c" succeeded.
Compiler: F:/LVGL_Gui_Guider/Gui-Guider/environment/mingw/bin/gcc.exe 
Build flags: 
Id flags:  

The output was:
0

Compilation of the C compiler identification source "CMakeCCompilerId.c" produced "a.exe"

The C compiler identification is GNU, found in "F:/ESP_Code/lv_demos/build/CMakeFiles/3.23.1/CompilerIdC/a.exe"

Compiling the CXX compiler identification source file "CMakeCXXCompilerId.cpp" succeeded.
Compiler: F:/LVGL_Gui_Guider/Gui-Guider/environment/mingw/bin/c++.exe 
Build flags: 
Id flags:  

The output was:
0

Compilation of the CXX compiler identification source "CMakeCXXCompilerId.cpp" produced "a.exe"

The CXX compiler identification is GNU, found in "F:/ESP_Code/lv_demos/build/CMakeFiles/3.23.1/CompilerIdCXX/a.exe"

Detecting C compiler ABI info compiled with the following output:
Change Dir: F:/ESP_Code/lv_demos/build/CMakeFiles/CMakeTmp

Run Build Command(s):F:/Espressif/tools/ninja/1.10.2/ninja.exe cmTC_2a57f && [1/2] Building C object CMakeFiles/cmTC_2a57f.dir/CMakeCCompilerABI.c.obj

Using built-in specs.

COLLECT_GCC=F:\LVGL_Gui_Guider\Gui-Guider\environment\mingw\bin\gcc.exe

Target: mingw32

Configured with: ../src/gcc-9.2.0/configure --build=x86_64-pc-linux-gnu --host=mingw32 --target=mingw32 --disable-win32-registry --with-arch=i586 --with-tune=generic --enable-static --enable-shared --enable-threads --enable-languages=c,c++,objc,obj-c++,fortran,ada --with-dwarf2 --disable-sjlj-exceptions --enable-version-specific-runtime-libs --enable-libgomp --disable-libvtv --with-libiconv-prefix=/mingw --with-libintl-prefix=/mingw --enable-libstdcxx-debug --disable-build-format-warnings --prefix=/mingw --with-gmp=/mingw --with-mpfr=/mingw --with-mpc=/mingw --with-isl=/mingw --enable-nls --with-pkgversion='MinGW.org GCC Build-20200227-1'

Thread model: win32

gcc version 9.2.0 (MinGW.org GCC Build-20200227-1) 

COLLECT_GCC_OPTIONS='-v' '-o' 'CMakeFiles/cmTC_2a57f.dir/CMakeCCompilerABI.c.obj' '-c' '-mtune=generic' '-march=i586'

 f:/lvgl_gui_guider/gui-guider/environment/mingw/bin/../libexec/gcc/mingw32/9.2.0/cc1.exe -quiet -v -iprefix f:\lvgl_gui_guider\gui-guider\environment\mingw\bin\../lib/gcc/mingw32/9.2.0/ F:/Espressif/tools/cmake/3.23.1/share/cmake-3.23/Modules/CMakeCCompilerABI.c -quiet -dumpbase CMakeCCompilerABI.c -mtune=generic -march=i586 -auxbase-strip CMakeFiles/cmTC_2a57f.dir/CMakeCCompilerABI.c.obj -version -o C:\Users\AILUNW~1\AppData\Local\Temp\ccA9ayF0.s

GNU C17 (MinGW.org GCC Build-20200227-1) version 9.2.0 (mingw32)

	compiled by GNU C version 9.2.0, GMP version 6.1.2, MPFR version 4.0.2, MPC version 1.1.0, isl version isl-0.21-GMP

GGC heuristics: --param ggc-min-expand=100 --param ggc-min-heapsize=131072

ignoring nonexistent directory "f:\lvgl_gui_guider\gui-guider\environment\mingw\bin\../lib/gcc/mingw32/9.2.0/../../../../mingw32/include"

ignoring duplicate directory "f:/lvgl_gui_guider/gui-guider/environment/mingw/lib/gcc/../../lib/gcc/mingw32/9.2.0/include"

ignoring nonexistent directory "/mingw/include"

ignoring duplicate directory "f:/lvgl_gui_guider/gui-guider/environment/mingw/lib/gcc/../../include"

ignoring duplicate directory "f:/lvgl_gui_guider/gui-guider/environment/mingw/lib/gcc/../../lib/gcc/mingw32/9.2.0/include-fixed"

ignoring nonexistent directory "f:/lvgl_gui_guider/gui-guider/environment/mingw/lib/gcc/../../lib/gcc/mingw32/9.2.0/../../../../mingw32/include"

ignoring nonexistent directory "/mingw/include"

#include "..." search starts here:

#include <...> search starts here:

[... 446 lines cut ...]
    arg [-lshell32] ==> lib [shell32]
    arg [-luser32] ==> lib [user32]
    arg [-lkernel32] ==> lib [kernel32]
    arg [-lmingw32] ==> lib [mingw32]
    arg [-lmingwex] ==> lib [mingwex]
    arg [-lgcc_s] ==> lib [gcc_s]
    arg [-lgcc] ==> lib [gcc]
    arg [-lmoldname] ==> lib [moldname]
    arg [-lmsvcrt] ==> lib [msvcrt]
    arg [f:/lvgl_gui_guider/gui-guider/environment/mingw/bin/../lib/gcc/mingw32/9.2.0/crtend.o] ==> obj [f:/lvgl_gui_guider/gui-guider/environment/mingw/bin/../lib/gcc/mingw32/9.2.0/crtend.o]
  remove lib [msvcrt]
  remove lib [msvcrt]
  collapse obj [f:/lvgl_gui_guider/gui-guider/environment/mingw/bin/../lib/gcc/mingw32/9.2.0/../../../crt2.o] ==> [F:/LVGL_Gui_Guider/Gui-Guider/environment/mingw/lib/crt2.o]
  collapse obj [f:/lvgl_gui_guider/gui-guider/environment/mingw/bin/../lib/gcc/mingw32/9.2.0/crtbegin.o] ==> [F:/LVGL_Gui_Guider/Gui-Guider/environment/mingw/lib/gcc/mingw32/9.2.0/crtbegin.o]
  collapse obj [f:/lvgl_gui_guider/gui-guider/environment/mingw/bin/../lib/gcc/mingw32/9.2.0/crtend.o] ==> [F:/LVGL_Gui_Guider/Gui-Guider/environment/mingw/lib/gcc/mingw32/9.2.0/crtend.o]
  collapse library dir [f:/lvgl_gui_guider/gui-guider/environment/mingw/bin/../lib/gcc/mingw32/9.2.0] ==> [F:/LVGL_Gui_Guider/Gui-Guider/environment/mingw/lib/gcc/mingw32/9.2.0]
  collapse library dir [f:/lvgl_gui_guider/gui-guider/environment/mingw/bin/../lib/gcc] ==> [F:/LVGL_Gui_Guider/Gui-Guider/environment/mingw/lib/gcc]
  collapse library dir [f:/lvgl_gui_guider/gui-guider/environment/mingw/bin/../lib/gcc/mingw32/9.2.0/../../../../mingw32/lib] ==> [F:/LVGL_Gui_Guider/Gui-Guider/environment/mingw/mingw32/lib]
  collapse library dir [f:/lvgl_gui_guider/gui-guider/environment/mingw/bin/../lib/gcc/mingw32/9.2.0/../../..] ==> [F:/LVGL_Gui_Guider/Gui-Guider/environment/mingw/lib]
[... cut ...]
  implicit objs: [F:/LVGL_Gui_Guider/Gui-Guider/environment/mingw/lib/crt2.o;F:/LVGL_Gui_Guider/Gui-Guider/environment/mingw/lib/gcc/mingw32/9.2.0/crtbegin.o;F:/LVGL_Gui_Guider/Gui-Guider/environment/mingw/lib/gcc/mingw32/9.2.0/crtend.o]
  implicit dirs: [F:/LVGL_Gui_Guider/Gui-Guider/environment/mingw/lib/gcc/mingw32/9.2.0;F:/LVGL_Gui_Guider/Gui-Guider/environment/mingw/lib/gcc;F:/LVGL_Gui_Guider/Gui-Guider/environment/mingw/mingw32/lib;F:/LVGL_Gui_Guider/Gui-Guider/environment/mingw/lib]
  implicit fwks: []

`this is the log

ailunwowo, 2023-01-31:
this is the log
[CMakeOutput.log](https://github.com/espressif/esp-box/files/10542204/CMakeOutput.log)

espressif2022, 2023-01-31:
> this is the log [CMakeOutput.log](https://github.com/espressif/esp-box/files/10542204/CMakeOutput.log)

You build the project in window, I've tried it, the result is ok;
[image]

ailunwowo, 2023-01-31:
> > this is the log [CMakeOutput.log](https://github.com/espressif/esp-box/files/10542204/CMakeOutput.log)
> 
> You build the project in window, I've tried it, the result is ok; [image]

This is a project that could be compiled before, but now I don’t know why it can’t be compiled.

espressif2022, 2023-01-31:
> > > this is the log [CMakeOutput.log](https://github.com/espressif/esp-box/files/10542204/CMakeOutput.log)
> > 
> > 
> > You build the project in window, I've tried it, the result is ok; [image]
> 
> This is a project that could be compiled before, but now I don’t know why it can’t be compiled.

What's the difference, and is your workspace clean?

ailunwowo, 2023-01-31:
> What's the difference, and is your workspace clean?

It's clean, I didn't do anything, just haven't compiled code for half a month, that's it

espressif2022, 2023-01-31:
git clone a new esp-box project to see whether it is a project problem or an esp-idf environment, and then analyze it?

ailunwowo, 2023-01-31:
> git clone a new esp-box project to see whether it is a project problem or an esp-idf environment, and then analyze it?

It seems to be an environmental problem, because the file **F:/Espressif/frameworks/esp-idf-v4.4.3/esp-idf/tools/cmake/project.cmake** was not found. It seems that the name of my esp-idf file is wrong, causing I didn’t find it. After I changed it, it can be compiled normally.

I want to ask you another question, how to modify the working time of the voice when using the voice recognition of esp32box? That is, after a few seconds, it will automatically sleep if it does not recognize the command

espressif2022, 2023-01-31:
> It seems to be an environmental problem, because the file **F:/Espressif/frameworks/esp-idf-v4.4.3/esp-idf/tools/cmake/project.cmake** was not found. It seems that the name of my esp-idf file is [...]

Sorry, the interface of  working time has not been released.

ailunwowo, 2023-01-31:
> 

ok thank you

ailunwowo, 2023-02-03:
Sorry to bother you, I have a similar problem here again. It was fine a few days ago, but now this problem occurs again
[CMakeOutput.log](https://github.com/espressif/esp-box/files/10574673/CMakeOutput.log)

espressif2022, 2023-02-03:
> Sorry to bother you, I have a similar problem here again. It was fine a few days ago, but now this problem occurs again [CMakeOutput.log](https://github.com/espressif/esp-box/files/10574673/CMakeOutput.log)

could you provide build log "idf_py_stdout_output_XX"

ailunwowo, 2023-02-03:
> 

i didn't find this file

espressif2022, 2023-02-03:
> > 
> 
> i didn't find this file

What appears on the command line after you `press idf.py build`.It should be the same with log file `idf_py_stdout_output_XX`

closed 2023-03-21 by ailunwowo