espressif/esp-box #59: On the Routine Error Reporting Problem of Compiling ESP box with ESP IDF 5.0

ailunwowo, 2023-03-21:
**I am using the master branch. The following introduction says that IDF 5.0 is already supported, but I downloaded and compiled it and still reported errors. Why is this? The following are error prompts and log logs**

`-- Found Git: D:/Espressif/tools/idf-git/2.34.2/cmd/git.exe (found version "2.34.1.windows.1") 
-- Component directory D:/ESP32S3/esp-box-master/components/esp-rainmaker does not contain a CMakeLists.txt file. No component will be added
-- Component directory D:/Espressif/frameworks/esp-idf-v5.0.1/examples/common_components/env_caps does not contain a CMakeLists.txt file. No component will be added
-- The C compiler identification is GNU 11.2.0
-- The ASM compiler identification is GNU
-- Found assembler: D:/Espressif/tools/xtensa-esp32s3-elf/esp-2022r1-11.2.0/xtensa-esp32s3-elf/bin/xtensa-esp32s3-elf-gcc.exe
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: D:/Espressif/tools/xtensa-esp32s3-elf/esp-2022r1-11.2.0/xtensa-esp32s3-elf/bin/xtensa-esp32s3-elf-gcc.exe - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Project is not inside a git repository, or git repository has no commits; will not use 'git describe' to determine PROJECT_VER.
-- Building ESP-IDF components for target esp32s3
CMake Error at D:/Espressif/frameworks/esp-idf-v5.0.1/tools/cmake/build.cmake:519 (message):
  Traceback (most recent call last):

    File "runpy.py", line 194, in _run_module_as_main
    File "runpy.py", line 87, in _run_code
    File "D:\Espressif\python_env\idf5.0_py3.8_env\lib\site-packages\idf_component_manager\prepare_components\__main__.py", line 6, in <module>
      main()
    File "D:\Espressif\python_env\idf5.0_py3.8_env\lib\site-packages\idf_component_manager\prepare_components\prepare.py", line 124, in main
      args.func(args)
    File "D:\Espressif\python_env\idf5.0_py3.8_env\lib\site-packages\idf_component_manager\prepare_components\prepare.py", line 27, in prepare_dep_dirs
      ComponentManager(
    File "D:\Espressif\python_env\idf5.0_py3.8_env\lib\site-packages\idf_component_manager\core.py", line 62, in wrapper
      return func(self, *args, **kwargs)
    File "D:\Espressif\python_env\idf5.0_py3.8_env\lib\site-packages\idf_component_manager\core.py", line 438, in prepare_dep_dirs
      downloaded_component_paths, downloaded_component_version_dict = download_project_dependencies(
    File "D:\Espressif\python_env\idf5.0_py3.8_env\lib\site-packages\idf_component_manager\dependencies.py", line 106, in download_project_dependencies
      if is_solve_required(project_requirements, solution):
    File "D:\Espressif\python_env\idf5.0_py3.8_env\lib\site-packages\idf_component_manager\dependencies.py", line 76, in is_solve_required
      if project_requirements.manifest_hash != solution.manifest_hash\
    File "D:\Espressif\python_env\idf5.0_py3.8_env\lib\site-packages\idf_component_tools\manifest\manifest.py", line 310, in manifest_hash
      manifest_hashes = [manifest.manifest_hash for manifest in self.manifests]
    File "D:\Espressif\python_env\idf5.0_py3.8_env\lib\site-packages\idf_component_tools\manifest\manifest.py", line 310, in <listcomp>
      manifest_hashes = [manifest.manifest_hash for manifest in self.manifests]
    File "D:\Espressif\python_env\idf5.0_py3.8_env\lib\site-packages\idf_component_tools\manifest\manifest.py", line 145, in manifest_hash
      serialized = self.serialize(serialize_default=False)  # type: ignore
    File "D:\Espressif\python_env\idf5.0_py3.8_env\lib\site-packages\idf_component_tools\serialization.py", line 63, in _serialize
      serialization_properties[prop] = serialize(getattr(self, prop), serialize_default)
    File "D:\Espressif\python_env\idf5.0_py3.8_env\lib\site-packages\idf_component_tools\serialization.py", line 31, in serialize
      return [serialize(v, serialize_default) for v in value]
    File "D:\Espressif\python_env\idf5.0_py3.8_env\lib\site-packages\idf_component_tools\serialization.py", line 31, in <listcomp>
      return [serialize(v, serialize_default) for v in value]
    File "D:\Espressif\python_env\idf5.0_py3.8_env\lib\site-packages\idf_component_tools\serialization.py", line 34, in serialize
      return value.serialize(serialize_default)
    File "D:\Espressif\python_env\idf5.0_py3.8_env\lib\site-packages\idf_component_tools\serialization.py", line 63, in _serialize
      serialization_properties[prop] = serialize(getattr(self, prop), serialize_default)
    File "D:\Espressif\python_env\idf5.0_py3.8_env\lib\site-packages\idf_component_tools\manifest\manifest.py", line 195, in version_spec
      return self.source.normalize_spec(self._version_spec)
    File "D:\Espressif\python_env\idf5.0_py3.8_env\lib\site-packages\idf_component_tools\sources\git.py", line 196, in normalize_spec
      commit_id = self._client.get_commit_id_by_ref(self.git_repo, self.cache_path(), ref)
    File "D:\Espressif\python_env\idf5.0_py3.8_env\lib\site-packages\idf_component_tools\git_client.py", line 41, in wrapper
      return func(self, *args, **kwargs)
    File "D:\Espressif\python_env\idf5.0_py3.8_env\lib\site-packages\idf_component_tools\git_client.py", line 72, in wrapper
      return func(self, *args, **kwargs)
    File "D:\Espressif\python_env\idf5.0_py3.8_env\lib\site-packages\idf_component_tools\git_client.py", line 159, in get_commit_id_by_ref
      ref = self.run(['ls-remote', '--exit-code', 'origin', 'HEAD'], cwd=bare_path)[:40]
    File "D:\Espressif\python_env\idf5.0_py3.8_env\lib\site-packages\idf_component_tools\git_client.py", line 183, in run
      raise GitCommandError(

  idf_component_tools.git_client.GitCommandError: 'git ls-remote --exit-code
  origin HEAD' failed with exit code 128

  fatal: unable to access 'https://github.com/espressif/esp-rainmaker.git/':
  Failed to connect to github.com port 443 after 21057 ms: Timed out

Call Stack (most recent call first):
  D:/Espressif/frameworks/esp-idf-v5.0.1/tools/cmake/project.cmake:440 (idf_build_process)
  CMakeLists.txt:14 (project)

-- Configuring incomplete, errors occurred!
See also "D:/ESP32S3/esp-box-master/examples/factory_demo/build/CMakeFiles/CMakeOutput.log".

 *  终端进程“C:\Windows\System32\WindowsPowerShell\v1.0\powershell.exe -Command cmake -G Ninja -DPYTHON_DEPS_CHECKED=1 -DESP_PLATFORM=1 -B d:\ESP32S3\esp-box-master\examples\factory_demo\build -S d:\ESP32S3\esp-box-master\examples\factory_demo”已终止，退出代码: 1。  `

**It prompts me when I compile again**
`
ninja: error: loading 'build.ninja': 系统找不到指定的文件。

 *  终端进程“C:\Windows\System32\WindowsPowerShell\v1.0\powershell.exe -Command ninja ”已终止，退出代码: 1。`
 
### **Here is the log**

[CMakeOutput.log](https://github.com/espressif/esp-box/files/11025257/CMakeOutput.log)

espressif2022, 2023-03-21:
Delete the invalid folder downloaded by submodule in the project, you can check it with git status. 
There is no rainmaker folder in component in the newest project.

ailunwowo, 2023-03-21:
> 

Can I just delete it? "At first, I pulled the project from git and it wasn't complete, so I later pulled the project branches one by one and compiled them in examples."

closed 2023-03-31 by ailunwowo