espressif/esp-box #61: esp-box编译报错

MrYanLK, 2023-03-28:
我在esp-idf4.4.2环境下编译esp-box的例程，报错如下，这是什么问题？

Adding "set-target"'s dependency "fullclean" to list of commands with default set of options.
Executing action: fullclean
Build directory '/home/ylk/esp/esp-box/examples/factory_demo/build' not found. Nothing to clean.
Executing action: set-target
Set Target to: esp32s3, new sdkconfig created. Existing sdkconfig renamed to sdkconfig.old.
Traceback (most recent call last):
  File "/home/ylk/esp_4_4_2/esp-idf/tools/idf.py", line 821, in <module>
    main()
  File "/home/ylk/esp_4_4_2/esp-idf/tools/idf.py", line 739, in main
    cli(sys.argv[1:], prog_name=PROG, complete_var='_IDF.PY_COMPLETE')
  File "/home/ylk/.espressif/python_env/idf4.4_py3.8_env/lib/python3.8/site-packages/click/core.py", line 1130, in __call__
    return self.main(*args, **kwargs)
  File "/home/ylk/.espressif/python_env/idf4.4_py3.8_env/lib/python3.8/site-packages/click/core.py", line 1055, in main
    rv = self.invoke(ctx)
  File "/home/ylk/.espressif/python_env/idf4.4_py3.8_env/lib/python3.8/site-packages/click/core.py", line 1689, in invoke
    return _process_result(rv)
  File "/home/ylk/.espressif/python_env/idf4.4_py3.8_env/lib/python3.8/site-packages/click/core.py", line 1626, in _process_result
    value = ctx.invoke(self._result_callback, value, **ctx.params)
  File "/home/ylk/.espressif/python_env/idf4.4_py3.8_env/lib/python3.8/site-packages/click/core.py", line 760, in invoke
    return __callback(*args, **kwargs)
  File "/home/ylk/esp_4_4_2/esp-idf/tools/idf.py", line 641, in execute_tasks
    task(ctx, global_args, task.action_args)
  File "/home/ylk/esp_4_4_2/esp-idf/tools/idf.py", line 217, in __call__
    self.callback(self.name, context, global_args, **action_args)
  File "/home/ylk/esp_4_4_2/esp-idf/tools/idf_py_actions/core_ext.py", line 181, in set_target
    ensure_build_directory(args, ctx.info_name, True)
  File "/home/ylk/esp_4_4_2/esp-idf/tools/idf_py_actions/tools.py", line 183, in ensure_build_directory
    os.makedirs(build_dir)
  File "/usr/lib/python3.8/os.py", line 223, in makedirs
    mkdir(name, mode)
PermissionError: [Errno 13] Permission denied: '/home/ylk/esp/esp-box/examples/factory_demo/build'

espressif2022, 2023-03-29:
检查下你的 factory_demo 工程下的 build 目录有没有生成？看着最后错误，你这生成 build 编译文件夹失败了，权限有问题。

espressif2022, 2023-03-29:
是拉取 esp-rainmaker 的组件失败了？
watering_demo 也是有该组件的，这个例程可以编译成功？

MrYanLK, 2023-03-29:
Hi 看起来是拉取esp-rainmaker组件失败了，我尝试过编译其他几个例程，watering_demo例程似乎也无法编译通过(log如下),这一次的log信息似乎有点又不一样了，昨天编译factory_demo例程时也出现过这种报错
至此，仅factory_demo和watering_demo例程无法编译通过。其他几个都可以。

root@DESKTOP-6TBOMGV:/home/ylk/esp/esp-box/examples/watering_demo#
root@DESKTOP-6TBOMGV:/home/ylk/esp/esp-box/examples/watering_demo#
root@DESKTOP-6TBOMGV:/home/ylk/esp/esp-box/examples/watering_demo# idf.py set-target esp32s3
Adding "set-target"'s dependency "fullclean" to list of commands with default set of options.
Executing action: fullclean
Build directory '/home/ylk/esp/esp-box/examples/watering_demo/build' not found. Nothing to clean.
Executing action: set-target
Set Target to: esp32s3, new sdkconfig created. Existing sdkconfig renamed to sdkconfig.old.
Running cmake in directory /home/ylk/esp/esp-box/examples/watering_demo/build
Executing "cmake -G Ninja -DPYTHON_DEPS_CHECKED=1 -DPYTHON=/root/.espressif/python_env/idf5.1_py3.8_env/bin/python -DESP_PLATFORM=1 -DIDF_TARGET=esp32s3 -DIDF_TARGET=esp32s3 -DCCACHE_ENABLE=0 /home/ylk/esp/esp-box/examples/watering_demo"...
-- Found Git: /usr/bin/git (found version "2.25.1")
-- The C compiler identification is GNU 12.2.0
-- The CXX compiler identification is GNU 12.2.0
-- The ASM compiler identification is GNU
-- Found assembler: /root/.espressif/tools/xtensa-esp32s3-elf/esp-12.2.0_20230208/xtensa-esp32s3-elf/bin/xtensa-esp32s3-elf-gcc
-- Check for working C compiler: /root/.espressif/tools/xtensa-esp32s3-elf/esp-12.2.0_20230208/xtensa-esp32s3-elf/bin/xtensa-esp32s3-elf-gcc
-- Check for working C compiler: /root/.espressif/tools/xtensa-esp32s3-elf/esp-12.2.0_20230208/xtensa-esp32s3-elf/bin/xtensa-esp32s3-elf-gcc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /root/.espressif/tools/xtensa-esp32s3-elf/esp-12.2.0_20230208/xtensa-esp32s3-elf/bin/xtensa-esp32s3-elf-g++
-- Check for working CXX compiler: /root/.espressif/tools/xtensa-esp32s3-elf/esp-12.2.0_20230208/xtensa-esp32s3-elf/bin/xtensa-esp32s3-elf-g++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Building ESP-IDF components for target esp32s3
CMake Error at /home/ylk/esp/esp-idf/tools/cmake/build.cmake:542 (message):
  Traceback (most recent call last):

    File "/usr/lib/python3.8/runpy.py", line 194, in _run_module_as_main
      return _run_code(code, main_globals, None,
    File "/usr/lib/python3.8/runpy.py", line 87, in _run_code
      exec(code, run_globals)
    File "/root/.espressif/python_env/idf5.1_py3.8_env/lib/python3.8/site-packages/idf_component_manager/prepare_components/__main__.py", line 6, in <module>
      main()
    File "/root/.espressif/python_env/idf5.1_py3.8_env/lib/python3.8/site-packages/idf_component_manager/prepare_components/prepare.py", line 124, in main
      args.func(args)
    File "/root/.espressif/python_env/idf5.1_py3.8_env/lib/python3.8/site-packages/idf_component_manager/prepare_components/prepare.py", line 27, in prepare_dep_dirs
      ComponentManager(
    File "/root/.espressif/python_env/idf5.1_py3.8_env/lib/python3.8/site-packages/idf_component_manager/core.py", line 62, in wrapper
      return func(self, *args, **kwargs)
    File "/root/.espressif/python_env/idf5.1_py3.8_env/lib/python3.8/site-packages/idf_component_manager/core.py", line 438, in prepare_dep_dirs
      downloaded_component_paths, downloaded_component_version_dict = download_project_dependencies(
    File "/root/.espressif/python_env/idf5.1_py3.8_env/lib/python3.8/site-packages/idf_component_manager/dependencies.py", line 106, in download_project_dependencies
      if is_solve_required(project_requirements, solution):
    File "/root/.espressif/python_env/idf5.1_py3.8_env/lib/python3.8/site-packages/idf_component_manager/dependencies.py", line 76, in is_solve_required
      if project_requirements.manifest_hash != solution.manifest_hash\
    File "/root/.espressif/python_env/idf5.1_py3.8_env/lib/python3.8/site-packages/idf_component_tools/manifest/manifest.py", line 310, in manifest_hash
      manifest_hashes = [manifest.manifest_hash for manifest in self.manifests]
    File "/root/.espressif/python_env/idf5.1_py3.8_env/lib/python3.8/site-packages/idf_component_tools/manifest/manifest.py", line 310, in <listcomp>
      manifest_hashes = [manifest.manifest_hash for manifest in self.manifests]
    File "/root/.espressif/python_env/idf5.1_py3.8_env/lib/python3.8/site-packages/idf_component_tools/manifest/manifest.py", line 145, in manifest_hash
      serialized = self.serialize(serialize_default=False)  # type: ignore
    File "/root/.espressif/python_env/idf5.1_py3.8_env/lib/python3.8/site-packages/idf_component_tools/serialization.py", line 63, in _serialize
      serialization_properties[prop] = serialize(getattr(self, prop), serialize_default)
    File "/root/.espressif/python_env/idf5.1_py3.8_env/lib/python3.8/site-packages/idf_component_tools/serialization.py", line 31, in serialize
      return [serialize(v, serialize_default) for v in value]
    File "/root/.espressif/python_env/idf5.1_py3.8_env/lib/python3.8/site-packages/idf_component_tools/serialization.py", line 31, in <listcomp>
      return [serialize(v, serialize_default) for v in value]
[... cut ...]
See also "/home/ylk/esp/esp-box/examples/watering_demo/build/CMakeFiles/CMakeOutput.log".
cmake failed with exit code 1, output of the command is in the /home/ylk/esp/esp-box/examples/watering_demo/build/log/idf_py_stderr_output_757 and /home/ylk/esp/esp-box/examples/watering_demo/build/log/idf_py_stdout_output_757
root@DESKTOP-6TBOMGV:/home/ylk/esp/esp-box/examples/watering_demo# ls
CMakeLists.txt  README.md  build  images  main  partitions.csv  sdkconfig.defaults  spiffs
root@DESKTOP-6TBOMGV:/home/ylk/esp/esp-box/examples/watering_demo#

espressif2022, 2023-03-29:
我这 git clone 试了下，应该是好的；

MrYanLK, 2023-03-29:
我在esp-box目录下执行 git clone https://github.com/espressif/esp-rainmaker.git，且下载成功了，但是再次编译factory_demo例程，还是报错，跟第一个log信息一致

root@DESKTOP-6TBOMGV:/home/ylk/esp/esp-box/examples/factory_demo# idf.py set-target esp32s3
Adding "set-target"'s dependency "fullclean" to list of commands with default set of options.
Executing action: fullclean
Build directory '/home/ylk/esp/esp-box/examples/factory_demo/build' not found. Nothing to clean.
Executing action: set-target
Set Target to: esp32s3, new sdkconfig created. Existing sdkconfig renamed to sdkconfig.old.
Running cmake in directory /home/ylk/esp/esp-box/examples/factory_demo/build
Executing "cmake -G Ninja -DPYTHON_DEPS_CHECKED=1 -DPYTHON=/root/.espressif/python_env/idf5.1_py3.8_env/bin/python -DESP_PLATFORM=1 -DIDF_TARGET=esp32s3 -DIDF_TARGET=esp32s3 -DCCACHE_ENABLE=0 /home/ylk/esp/esp-box/examples/factory_demo"...
-- Found Git: /usr/bin/git (found version "2.25.1")
-- Component directory /home/ylk/esp/esp-idf/examples/common_components/env_caps does not contain a CMakeLists.txt file. No component will be added
-- The C compiler identification is GNU 12.2.0
-- The CXX compiler identification is GNU 12.2.0
-- The ASM compiler identification is GNU
-- Found assembler: /root/.espressif/tools/xtensa-esp32s3-elf/esp-12.2.0_20230208/xtensa-esp32s3-elf/bin/xtensa-esp32s3-elf-gcc
-- Check for working C compiler: /root/.espressif/tools/xtensa-esp32s3-elf/esp-12.2.0_20230208/xtensa-esp32s3-elf/bin/xtensa-esp32s3-elf-gcc
-- Check for working C compiler: /root/.espressif/tools/xtensa-esp32s3-elf/esp-12.2.0_20230208/xtensa-esp32s3-elf/bin/xtensa-esp32s3-elf-gcc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /root/.espressif/tools/xtensa-esp32s3-elf/esp-12.2.0_20230208/xtensa-esp32s3-elf/bin/xtensa-esp32s3-elf-g++
-- Check for working CXX compiler: /root/.espressif/tools/xtensa-esp32s3-elf/esp-12.2.0_20230208/xtensa-esp32s3-elf/bin/xtensa-esp32s3-elf-g++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Building ESP-IDF components for target esp32s3
CMake Error at /home/ylk/esp/esp-idf/tools/cmake/build.cmake:542 (message):
  Traceback (most recent call last):

    File "/usr/lib/python3.8/runpy.py", line 194, in _run_module_as_main
      return _run_code(code, main_globals, None,
    File "/usr/lib/python3.8/runpy.py", line 87, in _run_code
      exec(code, run_globals)
    File "/root/.espressif/python_env/idf5.1_py3.8_env/lib/python3.8/site-packages/idf_component_manager/prepare_components/__main__.py", line 6, in <module>
      main()
    File "/root/.espressif/python_env/idf5.1_py3.8_env/lib/python3.8/site-packages/idf_component_manager/prepare_components/prepare.py", line 124, in main
      args.func(args)
    File "/root/.espressif/python_env/idf5.1_py3.8_env/lib/python3.8/site-packages/idf_component_manager/prepare_components/prepare.py", line 27, in prepare_dep_dirs
      ComponentManager(
    File "/root/.espressif/python_env/idf5.1_py3.8_env/lib/python3.8/site-packages/idf_component_manager/core.py", line 62, in wrapper
      return func(self, *args, **kwargs)
    File "/root/.espressif/python_env/idf5.1_py3.8_env/lib/python3.8/site-packages/idf_component_manager/core.py", line 438, in prepare_dep_dirs
      downloaded_component_paths, downloaded_component_version_dict = download_project_dependencies(
    File "/root/.espressif/python_env/idf5.1_py3.8_env/lib/python3.8/site-packages/idf_component_manager/dependencies.py", line 106, in download_project_dependencies
      if is_solve_required(project_requirements, solution):
    File "/root/.espressif/python_env/idf5.1_py3.8_env/lib/python3.8/site-packages/idf_component_manager/dependencies.py", line 76, in is_solve_required
      if project_requirements.manifest_hash != solution.manifest_hash\
    File "/root/.espressif/python_env/idf5.1_py3.8_env/lib/python3.8/site-packages/idf_component_tools/manifest/manifest.py", line 310, in manifest_hash
      manifest_hashes = [manifest.manifest_hash for manifest in self.manifests]
    File "/root/.espressif/python_env/idf5.1_py3.8_env/lib/python3.8/site-packages/idf_component_tools/manifest/manifest.py", line 310, in <listcomp>
      manifest_hashes = [manifest.manifest_hash for manifest in self.manifests]
    File "/root/.espressif/python_env/idf5.1_py3.8_env/lib/python3.8/site-packages/idf_component_tools/manifest/manifest.py", line 145, in manifest_hash
      serialized = self.serialize(serialize_default=False)  # type: ignore
    File "/root/.espressif/python_env/idf5.1_py3.8_env/lib/python3.8/site-packages/idf_component_tools/serialization.py", line 63, in _serialize
      serialization_properties[prop] = serialize(getattr(self, prop), serialize_default)
    File "/root/.espressif/python_env/idf5.1_py3.8_env/lib/python3.8/site-packages/idf_component_tools/serialization.py", line 31, in serialize
      return [serialize(v, serialize_default) for v in value]
    File "/root/.espressif/python_env/idf5.1_py3.8_env/lib/python3.8/site-packages/idf_component_tools/serialization.py", line 31, in <listcomp>
      return [serialize(v, serialize_default) for v in value]
    File "/root/.espressif/python_env/idf5.1_py3.8_env/lib/python3.8/site-packages/idf_component_tools/serialization.py", line 34, in serialize
[... cut ...]
CMakeLists.txt  README.md  build  main  partitions.csv  sdkconfig.defaults  spiffs
root@DESKTOP-6TBOMGV:/home/ylk/esp/esp-box/examples/factory_demo#
root@DESKTOP-6TBOMGV:/home/ylk/esp/esp-box/examples/factory_demo#
root@DESKTOP-6TBOMGV:/home/ylk/esp/esp-box/examples/factory_demo# ls ../../../
esp-adf  esp-at  esp-box  esp-box-master  esp-gitee-tools  esp-idf  esp-rainmaker
root@DESKTOP-6TBOMGV:/home/ylk/esp/esp-box/examples/factory_demo#
root@DESKTOP-6TBOMGV:/home/ylk/esp/esp-box/examples/factory_demo#

espressif2022, 2023-03-29:
[image]
你把手动 clone 的上述两个文件夹，拷贝到components 路径，先编译试试把，用这个commit ID：b863717a86e8a3bd987aac95670c0817233f2a6a
你这看着就是 github 链接失败了，但是你手动下载仓库又是好的；

MrYanLK, 2023-03-29:
我尝试在IDF4.4和5.1版本，将上述文件夹放在components目录下，编译还是会报同样的错误，我这边暂时没有5.0的环境。另外，这个 commit ID 如何切换？

espressif2022, 2023-03-29:
> 我尝试在IDF4.4和5.1版本，将上述文件夹放在components目录下，编译还是会报同样的错误，我这边暂时没有5.0的环境。另外，这个 commit ID 如何切换？

我看你是 linux 环境吧，在IDF工程 git checkout release/v5.0 就行了，commit ID 也同样，git checkout commit ID

MrYanLK, 2023-03-29:
唉，切换到5.0版本也是编译不过，同样的报错，要崩溃了。。。

root@DESKTOP-6TBOMGV:/home/ylk/esp/esp-box/examples/factory_demo#
root@DESKTOP-6TBOMGV:/home/ylk/esp/esp-box/examples/factory_demo# idf.py set-target esp32s3
Adding "set-target"'s dependency "fullclean" to list of commands with default set of options.
Executing action: fullclean
Build directory '/home/ylk/esp/esp-box/examples/factory_demo/build' not found. Nothing to clean.
Executing action: set-target
Set Target to: esp32s3, new sdkconfig created. Existing sdkconfig renamed to sdkconfig.old.
Running cmake in directory /home/ylk/esp/esp-box/examples/factory_demo/build
Executing "cmake -G Ninja -DPYTHON_DEPS_CHECKED=1 -DESP_PLATFORM=1 -DIDF_TARGET=esp32s3 -DIDF_TARGET=esp32s3 -DCCACHE_ENABLE=0 /home/ylk/esp/esp-box/examples/factory_demo"...
-- Found Git: /usr/bin/git (found version "2.25.1")
-- Component directory /home/ylk/esp/esp-idf/components/esp-rainmaker does not contain a CMakeLists.txt file. No component will be added
-- Component directory /home/ylk/esp/esp-idf/components/esp_coex does not contain a CMakeLists.txt file. No component will be added
CMake Warning at /home/ylk/esp/esp-idf/tools/cmake/git_submodules.cmake:52 (message):
  Git submodule components/bt/controller/lib_esp32 is out of date.  Run 'git
  submodule update --init --recursive' to fix.
Call Stack (most recent call first):
  /home/ylk/esp/esp-idf/tools/cmake/build.cmake:77 (git_submodule_check)
  /home/ylk/esp/esp-idf/tools/cmake/build.cmake:219 (__build_get_idf_git_revision)
  /home/ylk/esp/esp-idf/tools/cmake/idf.cmake:53 (__build_init)
  /home/ylk/esp/esp-idf/tools/cmake/project.cmake:12 (include)
  CMakeLists.txt:5 (include)

CMake Warning at /home/ylk/esp/esp-idf/tools/cmake/git_submodules.cmake:52 (message):
  Git submodule components/bt/controller/lib_esp32h2/esp32h2-bt-lib is out of
  date.  Run 'git submodule update --init --recursive' to fix.
Call Stack (most recent call first):
  /home/ylk/esp/esp-idf/tools/cmake/build.cmake:77 (git_submodule_check)
  /home/ylk/esp/esp-idf/tools/cmake/build.cmake:219 (__build_get_idf_git_revision)
  /home/ylk/esp/esp-idf/tools/cmake/idf.cmake:53 (__build_init)
  /home/ylk/esp/esp-idf/tools/cmake/project.cmake:12 (include)
  CMakeLists.txt:5 (include)

CMake Warning at /home/ylk/esp/esp-idf/tools/cmake/git_submodules.cmake:52 (message):
  Git submodule components/bt/host/nimble/nimble is out of date.  Run 'git
  submodule update --init --recursive' to fix.
Call Stack (most recent call first):
  /home/ylk/esp/esp-idf/tools/cmake/build.cmake:77 (git_submodule_check)
  /home/ylk/esp/esp-idf/tools/cmake/build.cmake:219 (__build_get_idf_git_revision)
  /home/ylk/esp/esp-idf/tools/cmake/idf.cmake:53 (__build_init)
  /home/ylk/esp/esp-idf/tools/cmake/project.cmake:12 (include)
  CMakeLists.txt:5 (include)

CMake Warning at /home/ylk/esp/esp-idf/tools/cmake/git_submodules.cmake:52 (message):
  Git submodule components/esp_phy/lib is out of date.  Run 'git submodule
  update --init --recursive' to fix.
Call Stack (most recent call first):
  /home/ylk/esp/esp-idf/tools/cmake/build.cmake:77 (git_submodule_check)
  /home/ylk/esp/esp-idf/tools/cmake/build.cmake:219 (__build_get_idf_git_revision)
  /home/ylk/esp/esp-idf/tools/cmake/idf.cmake:53 (__build_init)
  /home/ylk/esp/esp-idf/tools/cmake/project.cmake:12 (include)
  CMakeLists.txt:5 (include)

CMake Warning at /home/ylk/esp/esp-idf/tools/cmake/git_submodules.cmake:52 (message):
  Git submodule components/esp_wifi/lib is out of date.  Run 'git submodule
  update --init --recursive' to fix.
Call Stack (most recent call first):
  /home/ylk/esp/esp-idf/tools/cmake/build.cmake:77 (git_submodule_check)
  /home/ylk/esp/esp-idf/tools/cmake/build.cmake:219 (__build_get_idf_git_revision)
  /home/ylk/esp/esp-idf/tools/cmake/idf.cmake:53 (__build_init)
  /home/ylk/esp/esp-idf/tools/cmake/project.cmake:12 (include)
  CMakeLists.txt:5 (include)

CMake Warning at /home/ylk/esp/esp-idf/tools/cmake/git_submodules.cmake:52 (message):
  Git submodule components/ieee802154/lib is out of date.  Run 'git submodule
  update --init --recursive' to fix.
Call Stack (most recent call first):
  /home/ylk/esp/esp-idf/tools/cmake/build.cmake:77 (git_submodule_check)
  /home/ylk/esp/esp-idf/tools/cmake/build.cmake:219 (__build_get_idf_git_revision)
[... 111 lines cut ...]
    File "/root/.espressif/python_env/idf5.0_py3.8_env/lib/python3.8/site-packages/idf_component_tools/sources/git.py", line 196, in normalize_spec
      commit_id = self._client.get_commit_id_by_ref(self.git_repo, self.cache_path(), ref)
    File "/root/.espressif/python_env/idf5.0_py3.8_env/lib/python3.8/site-packages/idf_component_tools/git_client.py", line 41, in wrapper
      return func(self, *args, **kwargs)
    File "/root/.espressif/python_env/idf5.0_py3.8_env/lib/python3.8/site-packages/idf_component_tools/git_client.py", line 72, in wrapper
      return func(self, *args, **kwargs)
    File "/root/.espressif/python_env/idf5.0_py3.8_env/lib/python3.8/site-packages/idf_component_tools/git_client.py", line 159, in get_commit_id_by_ref
      ref = self.run(['ls-remote', '--exit-code', 'origin', 'HEAD'], cwd=bare_path)[:40]
    File "/root/.espressif/python_env/idf5.0_py3.8_env/lib/python3.8/site-packages/idf_component_tools/git_client.py", line 183, in run
      raise GitCommandError(

  idf_component_tools.git_client.GitCommandError: 'git ls-remote --exit-code
  origin HEAD' failed with exit code 128

[... cut ...]
  /home/ylk/esp/esp-idf/tools/cmake/project.cmake:447 (idf_build_process)
  CMakeLists.txt:14 (project)

-- Configuring incomplete, errors occurred!
See also "/home/ylk/esp/esp-box/examples/factory_demo/build/CMakeFiles/CMakeOutput.log".
cmake failed with exit code 1, output of the command is in the /home/ylk/esp/esp-box/examples/factory_demo/build/log/idf_py_stderr_output_18073 and /home/ylk/esp/esp-box/examples/factory_demo/build/log/idf_py_stdout_output_18073
root@DESKTOP-6TBOMGV:/home/ylk/esp/esp-box/examples/factory_demo#

espressif2022, 2023-03-29:
> GnuTLS recv error (-110): The TLS connection was non-properly terminated.

你把手动改拷贝到 components 的目录路径截图看下，拷贝过去后可以把 idf_component.yml 下 rainmaker 注释掉，我这边试了是 ok 的
[image]

MrYanLK, 2023-03-30:
是我搞错了，我以为放到idf下的components目录下呢，factory_demo例程下是没有component目录的，刚刚拷贝到了esp-box的components下，并且注释了 factory_demo/main/idf_component.yml 的 rainmaker，但是报了另一种错，第一次见这种报错，不知什么原因？

root@DESKTOP-6TBOMGV:/home/ylk/esp/esp-box/examples/factory_demo#
root@DESKTOP-6TBOMGV:/home/ylk/esp/esp-box/examples/factory_demo# idf.py set-target esp32s3
Adding "set-target"'s dependency "fullclean" to list of commands with default set of options.
Executing action: fullclean
Build directory '/home/ylk/esp/esp-box/examples/factory_demo/build' not found. Nothing to clean.
Executing action: set-target
Set Target to: esp32s3, new sdkconfig created. Existing sdkconfig renamed to sdkconfig.old.
Running cmake in directory /home/ylk/esp/esp-box/examples/factory_demo/build
Executing "cmake -G Ninja -DPYTHON_DEPS_CHECKED=1 -DPYTHON=/root/.espressif/python_env/idf4.4_py3.8_env/bin/python -DESP_PLATFORM=1 -DIDF_TARGET=esp32s3 -DIDF_TARGET=esp32s3 -DCCACHE_ENABLE=0 /home/ylk/esp/esp-box/examples/factory_demo"...
-- Found Git: /usr/bin/git (found version "2.25.1")
-- Component directory /home/ylk/esp/esp-idf/components/esp-rainmaker does not contain a CMakeLists.txt file. No component will be added
-- Component directory /home/ylk/esp/esp-box/components/esp-rainmaker does not contain a CMakeLists.txt file. No component will be added
-- Component directory /home/ylk/esp/esp-idf/examples/common_components/env_caps does not contain a CMakeLists.txt file. No component will be added
-- The C compiler identification is GNU 12.2.0
-- The CXX compiler identification is GNU 12.2.0
-- The ASM compiler identification is GNU
-- Found assembler: /root/.espressif/tools/xtensa-esp32s3-elf/esp-12.2.0_20230208/xtensa-esp32s3-elf/bin/xtensa-esp32s3-elf-gcc
-- Check for working C compiler: /root/.espressif/tools/xtensa-esp32s3-elf/esp-12.2.0_20230208/xtensa-esp32s3-elf/bin/xtensa-esp32s3-elf-gcc
-- Check for working C compiler: /root/.espressif/tools/xtensa-esp32s3-elf/esp-12.2.0_20230208/xtensa-esp32s3-elf/bin/xtensa-esp32s3-elf-gcc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /root/.espressif/tools/xtensa-esp32s3-elf/esp-12.2.0_20230208/xtensa-esp32s3-elf/bin/xtensa-esp32s3-elf-g++
-- Check for working CXX compiler: /root/.espressif/tools/xtensa-esp32s3-elf/esp-12.2.0_20230208/xtensa-esp32s3-elf/bin/xtensa-esp32s3-elf-g++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Building ESP-IDF components for target esp32s3
Processing 20 dependencies:
[1/20] chmorgan/esp-audio-player (1.0.4)
[2/20] chmorgan/esp-file-iterator (1.0.0)
[3/20] chmorgan/esp-libhelix-mp3 (1.0.3)
[4/20] espressif/button (2.5.1)
[5/20] espressif/cmake_utilities (0.4.3)
[6/20] espressif/es7210 (1.0.1)
[7/20] espressif/es8311 (1.0.0)
[8/20] espressif/esp-box (2.2.1)
[9/20] espressif/esp-dsp (1.3.0)
[10/20] espressif/esp-sr (1.2.1)
[11/20] espressif/esp_lcd_touch (1.0.3)
[12/20] espressif/esp_lcd_touch_tt21100 (1.0.6)
[13/20] espressif/esp_lvgl_port (1.1.0)
[14/20] espressif/json_generator (1.1.1)
[15/20] espressif/json_parser (1.0.0)
[16/20] espressif/led_strip (2.0.0)
[17/20] espressif/qrcode (0.1.0)
[18/20] espressif/rmaker_common (1.3.0)
[19/20] idf (5.1.0)
[20/20] lvgl/lvgl (8.3.0)
CMake Error at /home/ylk/esp/esp-idf/tools/cmake/component.cmake:224 (message):
  CMake Warning (dev) at build_properties.temp.cmake:8:

    Syntax Warning in cmake code at column 47

    Argument not separated from preceding token by whitespace.

  Call Stack (most recent call first):

    /home/ylk/esp/esp-idf/tools/cmake/scripts/component_get_requirements.cmake:3 (include)

  This warning is for project developers.  Use -Wno-dev to suppress it.

  CMake Error at
  /home/ylk/esp/esp-idf/examples/common_components/protocol_examples_tapif_io/CMakeLists.txt:8
  (message):

    This component is currently only supported for linux target

  Call Stack (most recent call first):

    /home/ylk/esp/esp-idf/tools/cmake/scripts/component_get_requirements.cmake:106 (include)
    /home/ylk/esp/esp-idf/tools/cmake/scripts/component_get_requirements.cmake:124 (__component_get_requirements)

Call Stack (most recent call first):
  /home/ylk/esp/esp-idf/tools/cmake/build.cmake:576 (__component_get_requirements)
  /home/ylk/esp/esp-idf/tools/cmake/project.cmake:453 (idf_build_process)
  CMakeLists.txt:14 (project)

-- Configuring incomplete, errors occurred!
See also "/home/ylk/esp/esp-box/examples/factory_demo/build/CMakeFiles/CMakeOutput.log".
cmake failed with exit code 1, output of the command is in the /home/ylk/esp/esp-box/examples/factory_demo/build/log/idf_py_stderr_output_1955 and /home/ylk/esp/esp-box/examples/factory_demo/build/log/idf_py_stdout_output_1955
root@DESKTOP-6TBOMGV:/home/ylk/esp/esp-box/examples/factory_demo#
root@DESKTOP-6TBOMGV:/home/ylk/esp/esp-box/examples/factory_demo# ls ../../components/
bsp  esp-rainmaker
root@DESKTOP-6TBOMGV:/home/ylk/esp/esp-box/examples/factory_demo#

espressif2022, 2023-03-30:
你的子仓库 是不是没拉下来？git submodule update --init --recursive

MrYanLK, 2023-03-31:
按照教程我在esp-gitee-tools 目录下执行./submodule-update.sh  ../esp-box拉取子仓库，但是并没有下载任何东西也没报错，就很奇怪。
我将 idf_component.yml git hub 的仓库 改成 gitee的后，输入账户密码但是报了404错误码

root@DESKTOP-6TBOMGV:/home/ylk/esp/esp-box/examples/factory_demo#
root@DESKTOP-6TBOMGV:/home/ylk/esp/esp-box/examples/factory_demo# idf.py set-target esp32s3
Adding "set-target"'s dependency "fullclean" to list of commands with default set of options.
Executing action: fullclean
Build directory '/home/ylk/esp/esp-box/examples/factory_demo/build' not found. Nothing to clean.
Executing action: set-target
Set Target to: esp32s3, new sdkconfig created. Existing sdkconfig renamed to sdkconfig.old.
Running cmake in directory /home/ylk/esp/esp-box/examples/factory_demo/build
Executing "cmake -G Ninja -DPYTHON_DEPS_CHECKED=1 -DPYTHON=/root/.espressif/python_env/idf5.1_py3.8_env/bin/python -DESP_PLATFORM=1 -DIDF_TARGET=esp32s3 -DIDF_TARGET=esp32s3 -DCCACHE_ENABLE=0 /home/ylk/esp/esp-box/examples/factory_demo"...
-- Found Git: /usr/bin/git (found version "2.25.1")
-- Component directory /home/ylk/esp/esp-idf/components/esp-rainmaker does not contain a CMakeLists.txt file. No component will be added
-- The C compiler identification is GNU 12.2.0
-- The CXX compiler identification is GNU 12.2.0
-- The ASM compiler identification is GNU
-- Found assembler: /root/.espressif/tools/xtensa-esp32s3-elf/esp-12.2.0_20230208/xtensa-esp32s3-elf/bin/xtensa-esp32s3-elf-gcc
-- Check for working C compiler: /root/.espressif/tools/xtensa-esp32s3-elf/esp-12.2.0_20230208/xtensa-esp32s3-elf/bin/xtensa-esp32s3-elf-gcc
-- Check for working C compiler: /root/.espressif/tools/xtensa-esp32s3-elf/esp-12.2.0_20230208/xtensa-esp32s3-elf/bin/xtensa-esp32s3-elf-gcc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /root/.espressif/tools/xtensa-esp32s3-elf/esp-12.2.0_20230208/xtensa-esp32s3-elf/bin/xtensa-esp32s3-elf-g++
-- Check for working CXX compiler: /root/.espressif/tools/xtensa-esp32s3-elf/esp-12.2.0_20230208/xtensa-esp32s3-elf/bin/xtensa-esp32s3-elf-g++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Building ESP-IDF components for target esp32s3
**Username for 'https://gitee.com': yanlaikun
Password for 'https://yanlaikun@gitee.com':**
CMake Error at /home/ylk/esp/esp-idf/tools/cmake/build.cmake:542 (message):
  Traceback (most recent call last):

    File "/usr/lib/python3.8/runpy.py", line 194, in _run_module_as_main
      return _run_code(code, main_globals, None,
    File "/usr/lib/python3.8/runpy.py", line 87, in _run_code
      exec(code, run_globals)
    File "/root/.espressif/python_env/idf5.1_py3.8_env/lib/python3.8/site-packages/idf_component_manager/prepare_components/__main__.py", line 6, in <module>
      main()
    File "/root/.espressif/python_env/idf5.1_py3.8_env/lib/python3.8/site-packages/idf_component_manager/prepare_components/prepare.py", line 124, in main
      args.func(args)
    File "/root/.espressif/python_env/idf5.1_py3.8_env/lib/python3.8/site-packages/idf_component_manager/prepare_components/prepare.py", line 27, in prepare_dep_dirs
      ComponentManager(
    File "/root/.espressif/python_env/idf5.1_py3.8_env/lib/python3.8/site-packages/idf_component_manager/core.py", line 62, in wrapper
      return func(self, *args, **kwargs)
    File "/root/.espressif/python_env/idf5.1_py3.8_env/lib/python3.8/site-packages/idf_component_manager/core.py", line 438, in prepare_dep_dirs
      downloaded_component_paths, downloaded_component_version_dict = download_project_dependencies(
    File "/root/.espressif/python_env/idf5.1_py3.8_env/lib/python3.8/site-packages/idf_component_manager/dependencies.py", line 106, in download_project_dependencies
      if is_solve_required(project_requirements, solution):
    File "/root/.espressif/python_env/idf5.1_py3.8_env/lib/python3.8/site-packages/idf_component_manager/dependencies.py", line 76, in is_solve_required
      if project_requirements.manifest_hash != solution.manifest_hash\
    File "/root/.espressif/python_env/idf5.1_py3.8_env/lib/python3.8/site-packages/idf_component_tools/manifest/manifest.py", line 310, in manifest_hash
      manifest_hashes = [manifest.manifest_hash for manifest in self.manifests]
    File "/root/.espressif/python_env/idf5.1_py3.8_env/lib/python3.8/site-packages/idf_component_tools/manifest/manifest.py", line 310, in <listcomp>
      manifest_hashes = [manifest.manifest_hash for manifest in self.manifests]
    File "/root/.espressif/python_env/idf5.1_py3.8_env/lib/python3.8/site-packages/idf_component_tools/manifest/manifest.py", line 145, in manifest_hash
      serialized = self.serialize(serialize_default=False)  # type: ignore
    File "/root/.espressif/python_env/idf5.1_py3.8_env/lib/python3.8/site-packages/idf_component_tools/serialization.py", line 63, in _serialize
      serialization_properties[prop] = serialize(getattr(self, prop), serialize_default)
    File "/root/.espressif/python_env/idf5.1_py3.8_env/lib/python3.8/site-packages/idf_component_tools/serialization.py", line 31, in serialize
      return [serialize(v, serialize_default) for v in value]
    File "/root/.espressif/python_env/idf5.1_py3.8_env/lib/python3.8/site-packages/idf_component_tools/serialization.py", line 31, in <listcomp>
[... cut ...]
  /home/ylk/esp/esp-idf/tools/cmake/project.cmake:453 (idf_build_process)
  CMakeLists.txt:14 (project)

-- Configuring incomplete, errors occurred!
See also "/home/ylk/esp/esp-box/examples/factory_demo/build/CMakeFiles/CMakeOutput.log".
cmake failed with exit code 1, output of the command is in the /home/ylk/esp/esp-box/examples/factory_demo/build/log/idf_py_stderr_output_11747 and /home/ylk/esp/esp-box/examples/factory_demo/build/log/idf_py_stdout_output_11747
root@DESKTOP-6TBOMGV:/home/ylk/esp/esp-box/examples/factory_demo#

MrYanLK, 2023-03-31:
Hi，我重新下载了esp-box，将idf_component.yml git hub 的仓库 改成[ gitee](https://gitee.com/EspressifSystems/esp-rainmaker.git) 的后可以编译通过了，非常感谢！

ESP-Mars, 2023-04-03:
Hello, if your issue has been resolved, please close the issue. If there are any updates, you can open it again. Thank you.

MrYanLK, 2023-04-03:
好的

closed 2023-04-03 by MrYanLK

ESP-Mars, 2023-04-03:
@MrYanLK Please also update this issue https://github.com/espressif/esp-box/issues/62 thanks!