espressif/esp-box #63: 我编译一个我以前可以编译没问题的工程可是为什么换一台电脑后就编译不了呢

ailunwowo, 2023-03-30:
我clone了工程在编译的，但就是报错
-- Found Git: C:/Espressif/tools/idf-git/2.34.2/cmd/git.exe (found version "2.34.1.windows.1") 
-- The C compiler identification is GNU 8.4.0
-- The CXX compiler identification is GNU 8.4.0
-- The ASM compiler identification is GNU
-- Found assembler: C:/Espressif/tools/xtensa-esp32s3-elf/esp-2021r2-patch5-8.4.0/xtensa-esp32s3-elf/bin/xtensa-esp32s3-elf-gcc.exe
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: C:/Espressif/tools/xtensa-esp32s3-elf/esp-2021r2-patch5-8.4.0/xtensa-esp32s3-elf/bin/xtensa-esp32s3-elf-gcc.exe - skipped
-- Detecting C compile features       
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: C:/Espressif/tools/xtensa-esp32s3-elf/esp-2021r2-patch5-8.4.0/xtensa-esp32s3-elf/bin/xtensa-esp32s3-elf-g++.exe - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Project is not inside a git repository, or git repository has no commits; will not use 'git describe' to determine PROJECT_VER.
-- Building ESP-IDF components for target esp32s3
CMake Error at C:/Espressif/frameworks/esp-idf-v4.4.3/tools/cmake/build.cmake:452 (message):
  Traceback (most recent call last):

    File "runpy.py", line 194, in _run_module_as_main
    File "runpy.py", line 87, in _run_code
    File "C:\Espressif\python_env\idf4.4_py3.8_env\lib\site-packages\idf_component_manager\prepare_components\__main__.py", line 6, in <module>     
      main()
    File "C:\Espressif\python_env\idf4.4_py3.8_env\lib\site-packages\idf_component_manager\prepare_components\prepare.py", line 124, in main        
      args.func(args)
    File "C:\Espressif\python_env\idf4.4_py3.8_env\lib\site-packages\idf_component_manager\prepare_components\prepare.py", line 27, in prepare_dep_dirs
      ComponentManager(
    File "C:\Espressif\python_env\idf4.4_py3.8_env\lib\site-packages\idf_component_manager\core.py", line 62, in wrapper
      return func(self, *args, **kwargs)
    File "C:\Espressif\python_env\idf4.4_py3.8_env\lib\site-packages\idf_component_manager\core.py", line 438, in prepare_dep_dirs
      downloaded_component_paths, downloaded_component_version_dict = download_project_dependencies(
    File "C:\Espressif\python_env\idf4.4_py3.8_env\lib\site-packages\idf_component_manager\dependencies.py", line 106, in download_project_dependencies
      if is_solve_required(project_requirements, solution):
    File "C:\Espressif\python_env\idf4.4_py3.8_env\lib\site-packages\idf_component_manager\dependencies.py", line 76, in is_solve_required
      if project_requirements.manifest_hash != solution.manifest_hash\
    File "C:\Espressif\python_env\idf4.4_py3.8_env\lib\site-packages\idf_component_tools\manifest\manifest.py", line 310, in manifest_hash
      manifest_hashes = [manifest.manifest_hash for manifest in self.manifests]
    File "C:\Espressif\python_env\idf4.4_py3.8_env\lib\site-packages\idf_component_tools\manifest\manifest.py", line 310, in <listcomp>
      manifest_hashes = [manifest.manifest_hash for manifest in self.manifests]
    File "C:\Espressif\python_env\idf4.4_py3.8_env\lib\site-packages\idf_component_tools\manifest\manifest.py", line 145, in manifest_hash
      serialized = self.serialize(serialize_default=False)  # type: ignore
    File "C:\Espressif\python_env\idf4.4_py3.8_env\lib\site-packages\idf_component_tools\serialization.py", line 63, in _serialize
      serialization_properties[prop] = serialize(getattr(self, prop), serialize_default)
    File "C:\Espressif\python_env\idf4.4_py3.8_env\lib\site-packages\idf_component_tools\serialization.py", line 31, in serialize
      return [serialize(v, serialize_default) for v in value]
    File "C:\Espressif\python_env\idf4.4_py3.8_env\lib\site-packages\idf_component_tools\serialization.py", line 31, in <listcomp>
      return [serialize(v, serialize_default) for v in value]
    File "C:\Espressif\python_env\idf4.4_py3.8_env\lib\site-packages\idf_component_tools\serialization.py", line 34, in serialize
      return value.serialize(serialize_default)
    File "C:\Espressif\python_env\idf4.4_py3.8_env\lib\site-packages\idf_component_tools\serialization.py", line 63, in _serialize
      serialization_properties[prop] = serialize(getattr(self, prop), serialize_default)
    File "C:\Espressif\python_env\idf4.4_py3.8_env\lib\site-packages\idf_component_tools\manifest\manifest.py", line 195, in version_spec
      return self.source.normalize_spec(self._version_spec)
    File "C:\Espressif\python_env\idf4.4_py3.8_env\lib\site-packages\idf_component_tools\sources\git.py", line 196, in normalize_spec
      commit_id = self._client.get_commit_id_by_ref(self.git_repo, self.cache_path(), ref)
    File "C:\Espressif\python_env\idf4.4_py3.8_env\lib\site-packages\idf_component_tools\git_client.py", line 41, in wrapper
      return func(self, *args, **kwargs)
    File "C:\Espressif\python_env\idf4.4_py3.8_env\lib\site-packages\idf_component_tools\git_client.py", line 69, in wrapper
      self._update_bare_repo(*args, **kwargs)
    File "C:\Espressif\python_env\idf4.4_py3.8_env\lib\site-packages\idf_component_tools\git_client.py", line 63, in _update_bare_repo
      self.run(['fetch', 'origin'], cwd=bare_path)
    File "C:\Espressif\python_env\idf4.4_py3.8_env\lib\site-packages\idf_component_tools\git_client.py", line 183, in run
      raise GitCommandError(

  idf_component_tools.git_client.GitCommandError: 'git fetch origin' failed
  with exit code 128

  fatal: unable to access 'https://github.com/espressif/esp-rainmaker.git/':
  OpenSSL SSL_read: Connection was reset, errno 10054

Call Stack (most recent call first):
  C:/Espressif/frameworks/esp-idf-v4.4.3/tools/cmake/project.cmake:384 (idf_build_process)
  CMakeLists.txt:13 (project)

-- Configuring incomplete, errors occurred!
See also "D:/LiAiGuo/力/力控制+按键/watering_demo/build/CMakeFiles/CMakeOutput.log".

espressif2022, 2023-03-30:
idf_component.yml 先将 esp_rainmaker 的 Github 的地址 换成下面的 https://gitee.com/EspressifSystems/esp-rainmaker.git

ailunwowo, 2023-03-30:
idf_component.yml这个文件在什么地方？是把这个文件里的esp_rainmaker 地址换成您说的这个是吗？

espressif2022, 2023-03-30:
> idf_component.yml这个文件在什么地方？是把这个文件里的esp_rainmaker 地址换成您说的这个是吗？

工程 example 文件夹 main 主文件夹下 `examples/factory_demo/main`

ailunwowo, 2023-03-30:
还是报错，请问这是什么原因？

-- Found Git: C:/Espressif/tools/idf-git/2.34.2/cmd/git.exe (found version "2.34.1.windows.1") 
-- The C compiler identification is GNU 8.4.0
-- The CXX compiler identification is GNU 8.4.0
-- The ASM compiler identification is GNU
-- Found assembler: C:/Espressif/tools/xtensa-esp32s3-elf/esp-2021r2-patch5-8.4.0/xtensa-esp32s3-elf/bin/xtensa-esp32s3-elf-gcc.exe
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: C:/Espressif/tools/xtensa-esp32s3-elf/esp-2021r2-patch5-8.4.0/xtensa-esp32s3-elf/bin/xtensa-esp32s3-elf-gcc.exe - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: C:/Espressif/tools/xtensa-esp32s3-elf/esp-2021r2-patch5-8.4.0/xtensa-esp32s3-elf/bin/xtensa-esp32s3-elf-g++.exe - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Project is not inside a git repository, or git repository has no commits; will not use 'git describe' to determine PROJECT_VER.
-- Building ESP-IDF components for target esp32s3
Skipping optional dependency: espressif/led_strip
Skipping optional dependency: espressif/qrcode
Solving dependencies requirements
CMake Error at C:/Espressif/frameworks/esp-idf-v4.4.3/tools/cmake/build.cmake:452 (message):
  ERROR: Cannot establish a connection to the component registry.  Are you
  connected to the internet?

Call Stack (most recent call first):
  C:/Espressif/frameworks/esp-idf-v4.4.3/tools/cmake/project.cmake:384 (idf_build_process)
  CMakeLists.txt:14 (project)

-- Configuring incomplete, errors occurred!
See also "D:/LiAiGuo/ESP32S3/esp-box-master/examples/factory_demo/build/CMakeFiles/CMakeOutput.log".

 *  终端进程“C:\Windows\System32\WindowsPowerShell\v1.0\powershell.exe -Command cmake -G Ninja -DPYTHON_DEPS_CHECKED=1 -DESP_PLATFORM=1 -B d:\LiAiGuo\ESP32S3\esp-box-master\examples\factory_demo\build -S d:\LiAiGuo\ESP32S3\esp-box-master\examples\factory_demo”已终止，退出代码: 1。

ailunwowo, 2023-03-30:
> > idf_component.yml这个文件在什么地方？是把这个文件里的esp_rainmaker 地址换成您说的这个是吗？
> 
> 工程 example 文件夹 main 主文件夹下 `examples/factory_demo/main`

我按照您说的改了

ailunwowo, 2023-03-30:
> > idf_component.yml这个文件在什么地方？是把这个文件里的esp_rainmaker 地址换成您说的这个是吗？
> 
> 工程 example 文件夹 main 主文件夹下 `examples/factory_demo/main`

请问有什么好的解决办法吗？我环境都重新装几遍了都编译不了，之前都可以的，现在却不行了，我自己这个工程是依附这个example的都还没搞完，真的跪求解决方法

espressif2022, 2023-03-30:
> > > idf_component.yml这个文件在什么地方？是把这个文件里的esp_rainmaker 地址换成您说的这个是吗？
> > 
> > 
> > 工程 example 文件夹 main 主文件夹下 `examples/factory_demo/main`
> 
> 请问有什么好的解决办法吗？我环境都重新装几遍了都编译不了，之前都可以的，现在却不行了，我自己这个工程是依附这个example的都还没搞完，真的跪求解决方法

我细看了下你的 log，你的 IDF 是不是 4.4 版本的，包管理器组件拉不到。需要 IDF5.0 版本

wujiangang, 2023-03-30:
> 还是报错，请问这是什么原因？
>
> -- Found Git: C:/Espressif/tools/idf-git/2.34.2/cmd/git.exe (found version "2.34.1.windows.1") -- The C compiler identification is GNU 8.4.0 -- The CXX compiler identification is GNU [...]

看上去你是基于 IDF v4.4.3，esp-box master 上只能支持 v5.0 及以上。

ailunwowo, 2023-03-30:
> 

好的，那我更新一下，在重新拉取编译试试，那我以前的那个工程是基于4.4写的，会受影响吗？

ailunwowo, 2023-03-31:
> 看上去你是基于 IDF v4.4.3，esp-box master 上只能支持 v5.0 及以上。

我重新下载了esp-idf5.0环境编译master 的factory_demo 还是会有一些报错找不到文件或者函数子类的这种[{
	"resource": "/d:/LiAiGuo/ESP32S3/esp-box-master/examples/factory_demo/main/main.c",
	"owner": "C/C++",
	"code": "1696",
	"severity": 8,
	"message": "无法打开 源 文件 \"bsp_storage.h\"",
	"source": "C/C++",
	"startLineNumber": 16,
	"startColumn": 1,
	"endLineNumber": 16,
	"endColumn": 25
}]
这是为什么呢？

ailunwowo, 2023-03-31:
> > > 
> > 
> > 
> > 好的，那我更新一下，在重新拉取编译试试，那我以前的那个工程是基于4.4写的，会受影响吗？
> 
> 大部分都可以，有些 API 会受影响，具体看你用了哪些功能。 或者你可以将工程回退到 TAG v0.3.0，这样可以继续使用 IDF v4.4.3 环境继续。

请问怎么切换espidf版本编译呢？

espressif2022, 2023-03-31:
> > > > 
> > > 
> > > 
> > > 好的，那我更新一下，在重新拉取编译试试，那我以前的那个工程是基于4.4写的，会受影响吗？
> > 
> > 
> > 大部分都可以，有些 API 会受影响，具体看你用了哪些功能。 或者你可以将工程回退到 TAG v0.3.0，这样可以继续使用 IDF v4.4.3 环境继续。
> 
> 请问怎么切换espidf版本编译呢？

我不是很确定你现有的环境，
如果你现在没有 ESP-IDF v4.4 的版本需要安装，下载离线安装包，https://dl.espressif.com/dl/esp-idf/?idf=4.4
如果你想将 现有 ESP-BOX 的工程切回到匹配的 v0.3.0 的 tag，你可以进你 git clone 的 ESP-BOX 的工程目录，git checkout v0.3.0

espressif2022, 2023-03-31:
> 我重新下载了esp-idf5.0环境编译master 的factory_demo 还是会有一些报错找不到文件或者函数子类的这种[{ "resource": "/d:/LiAiGuo/ESP32S3/esp-box-master/examples/factory_demo/main/main.c", "owner": "C/C++", "code": "1696", "severity": 8, [...]

你这个有点难分析，我用你同样环境在 window 用ESP-IDF v5.0 的离线安装包，拉取最新的 ESP-BOX 仓库，编译时ok的。
首先这个文件是有的么，你的项目仓库是不是有 dirty 的修改部分？

ailunwowo, 2023-03-31:
> 你这个有点难分析，我用你同样环境在 window 用ESP-IDF v5.0 的离线安装包，拉取最新的 ESP-BOX 仓库，编译时ok的。 首先这个文件是有的么，你的项目仓库是不是有 dirty 的修改部分？

编译目前没问题了，就是代码改动的有点大，我要重新适配一下，请问那个旋转屏幕的选项在代码的什么地方？

espressif2022, 2023-03-31:
`esp_lvgl_port.c/lvgl_port_add_disp` disp_ctx->rotation 的若干变量；
所以上面编译不过是啥问题？

ailunwowo, 2023-03-31:
> `esp_lvgl_port.c/lvgl_port_add_disp` disp_ctx->rotation 的若干变量； 所以上面编译不过是啥问题？

又又出毛病了的😭我就想启动一下SDKconfig，启动不成功我就取消了，然后在编译就报错了，为什么😭😭😭

-- Found Git: C:/Espressif/tools/idf-git/2.34.2/cmd/git.exe (found version "2.34.1.windows.1") 
-- The C compiler identification is GNU 11.2.0
-- The CXX compiler identification is GNU 11.2.0
-- The ASM compiler identification is GNU
-- Found assembler: F:/ESP5.0/tools/xtensa-esp32s3-elf/esp-2022r1-11.2.0/xtensa-esp32s3-elf/bin/xtensa-esp32s3-elf-gcc.exe
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: F:/ESP5.0/tools/xtensa-esp32s3-elf/esp-2022r1-11.2.0/xtensa-esp32s3-elf/bin/xtensa-esp32s3-elf-gcc.exe - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: F:/ESP5.0/tools/xtensa-esp32s3-elf/esp-2022r1-11.2.0/xtensa-esp32s3-elf/bin/xtensa-esp32s3-elf-g++.exe - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Project is not inside a git repository, or git repository has no commits; will not use 'git describe' to determine PROJECT_VER.
-- Building ESP-IDF components for target esp32s3
Processing 7 dependencies:
[1/7] espressif/es8311 (1.0.0)
[2/7] espressif/esp-box (2.3.3)
[3/7] espressif/esp_lcd_touch (1.0.3)
[4/7] espressif/esp_lcd_touch_tt21100 (1.0.6)
[5/7] espressif/esp_lvgl_port (1.1.0)
[6/7] idf (5.0.1)
[7/7] lvgl/lvgl (8.3.0)
CMake Error at F:/ESP5.0/esp-idf/tools/cmake/build.cmake:519 (message):
  ERROR: Some components (espressif/esp-box, espressif/esp_lvgl_port,
  lvgl/lvgl) in the "managed_components" directory were modified on the disk
  since the last run of the CMake.  Content of this directory is managed
  automatically.

  If you want to keep the changes, you can move the directory with the
  component to the "components"directory of your project.

  I.E.  for "espressif__esp-box" run:

  mv
  D:\LiAiGuo\ESP32S3\esp-box-master\examples\lv_demos\managed_components\espressif__esp-box
  D:\LiAiGuo\ESP32S3\esp-box-master\examples\lv_demos\components\espressif__esp-box

  Or, if you want to discard the changes remove the ".component_hash" file
  from the component's directory.

  I.E.  for "espressif__esp-box" run:

  rm
  D:\LiAiGuo\ESP32S3\esp-box-master\examples\lv_demos\managed_components\espressif__esp-box\.component_hash

Call Stack (most recent call first):
  F:/ESP5.0/esp-idf/tools/cmake/project.cmake:440 (idf_build_process)
  CMakeLists.txt:12 (project)

-- Configuring incomplete, errors occurred!
See also "D:/LiAiGuo/ESP32S3/esp-box-master/examples/lv_demos/build/CMakeFiles/CMakeOutput.log".

 *  终端进程“C:\Windows\System32\WindowsPowerShell\v1.0\powershell.exe -Command cmake -G Ninja -DPYTHON_DEPS_CHECKED=1 -DESP_PLATFORM=1 -S d:\LiAiGuo\ESP32S3\esp-box-master\examples\lv_demos -B d:\LiAiGuo\ESP32S3\esp-box-master\examples\lv_demos\build”已终止，退出代码: 1。

ailunwowo, 2023-03-31:
> `esp_lvgl_port.c/lvgl_port_add_disp` disp_ctx->rotation 的若干变量； 所以上面编译不过是啥问题？

包括我启动一次sdkconfig后在编译也会报错误

espressif2022, 2023-03-31:
> 启动一下SDKconfig

设施社么操作，menuconfig？报错是社么？

ailunwowo, 2023-03-31:
> 

就是vscode里那个menuconfig吗，报错就是我上面发的那些

ailunwowo, 2023-03-31:
> > 启动一下SDKconfig
> 
> 设施社么操作，menuconfig？报错是社么？

而且有时候会启动不了这个menuconfig

ailunwowo, 2023-03-31:
> > 启动一下SDKconfig
> 
> 设施社么操作，menuconfig？报错是社么？

还有一个问题就是我是用的ESPBOX_Lite 然后我编译WATERING_DEMO 这个工程会提示这个报错是为啥？我啥也没改
E (827) ES7210: es7210_write_reg(144): error while writing register
E (827) ES7210: es7210_config_codec(289): i2c communication error while writing ES7210_RESET_REG00
E (827) ES7210: es7210_write_reg(144): error while writing register
E (827) ES7210: es7210_config_volume(339): i2c communication error while writing ES7210_ADC1_DIRECT_DB_REG1B
E (827) ES8311: es8311_init(329): I2C read/write error

espressif2022, 2023-03-31:
> * WindowsPowerShell

启动一下SDKconfig，是指你运行 idf.py menuconfig 吗？
你不是用的 WindowsPowerShell, `vscode里那个menuconfig` 是？
最新的 master 暂时不支持 Box_Lite

ailunwowo, 2023-03-31:
> > * WindowsPowerShell
> 
> 启动一下SDKconfig，是指你运行 idf.py menuconfig 吗？ 你不是用的 WindowsPowerShell, `vscode里那个menuconfig` 是？ 最新的 master 暂时不支持 Box_Lite

啊？不支持Box_Lite?那好吧

ailunwowo, 2023-03-31:
> > * WindowsPowerShell
> 
> 启动一下SDKconfig，是指你运行 idf.py menuconfig 吗？ 你不是用的 WindowsPowerShell, `vscode里那个menuconfig` 是？ 最新的 master 暂时不支持 Box_Lite

那我应该用那个版本？我是去年十二月下载的master分支，那我还是改回到4.4吧

ailunwowo, 2023-03-31:
> > * WindowsPowerShell
> 
> 启动一下SDKconfig，是指你运行 idf.py menuconfig 吗？ 你不是用的 WindowsPowerShell, `vscode里那个menuconfig` 是？ 最新的 master 暂时不支持 Box_Lite

如果我要加语音识别的话可以加吗？我不依附那个example，我移一个过去可以吗

espressif2022, 2023-03-31:
> > > * WindowsPowerShell
> > 
> > 
> > 启动一下SDKconfig，是指你运行 idf.py menuconfig 吗？ 你不是用的 WindowsPowerShell, `vscode里那个menuconfig` 是？ 最新的 master 暂时不支持 Box_Lite
> 
> 如果我要加语音识别的话可以加吗？我不依附那个example，我移一个过去可以吗

https://components.espressif.com/components/espressif/esp-sr 这是语音组件
https://github.com/espressif/esp-skainet 语音识别应用仓库

ailunwowo, 2023-04-01:
> > > > * WindowsPowerShell
> > > 
> > > 
> > > 启动一下SDKconfig，是指你运行 idf.py menuconfig 吗？ 你不是用的 WindowsPowerShell, `vscode里那个menuconfig` 是？ 最新的 master 暂时不支持 Box_Lite
> > 
> > 
> > 如果我要加语音识别的话可以加吗？我不依附那个example，我移一个过去可以吗
> 
> https://components.espressif.com/components/espressif/esp-sr 这是语音组件 https://github.com/espressif/esp-skainet 语音识别应用仓库

好的，感谢，我还有一个问题我现在换回我以前电脑用的4.4环境，工程也是之前的，刚开始能编译也能改没任何问题，但是为什么menuconfig我更改了一个语音唤醒词后就编译不成功，报这些错是为啥？
-- Found Git: F:/EPS/tools/idf-git/2.30.1/cmd/git.exe (found version "2.30.1.windows.1") 
-- The C compiler identification is GNU 8.4.0
-- The CXX compiler identification is GNU 8.4.0
-- The ASM compiler identification is GNU
-- Found assembler: F:/Espressif/tools/xtensa-esp32s3-elf/esp-2021r2-patch5-8.4.0/xtensa-esp32s3-elf/bin/xtensa-esp32s3-elf-gcc.exe
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: F:/Espressif/tools/xtensa-esp32s3-elf/esp-2021r2-patch5-8.4.0/xtensa-esp32s3-elf/bin/xtensa-esp32s3-elf-gcc.exe - skipped
-- Detecting C compile features       
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: F:/Espressif/tools/xtensa-esp32s3-elf/esp-2021r2-patch5-8.4.0/xtensa-esp32s3-elf/bin/xtensa-esp32s3-elf-g++.exe - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Project is not inside a git repository, or git repository has no commits; will not use 'git describe' to determine PROJECT_VER.
-- Building ESP-IDF components for target esp32s3
CMake Error at F:/Espressif/esp-idf/tools/cmake/build.cmake:452 (message):
  Traceback (most recent call last):

    File "runpy.py", line 194, in _run_module_as_main
    File "runpy.py", line 87, in _run_code
    File "f:\Espressif\python_env\idf4.4_py3.8_env\lib\site-packages\idf_component_manager\prepare_components\__main__.py", line 6, in <module>
      main()
    File "f:\Espressif\python_env\idf4.4_py3.8_env\lib\site-packages\idf_component_manager\prepare_components\prepare.py", line 124, in main
      args.func(args)
    File "f:\Espressif\python_env\idf4.4_py3.8_env\lib\site-packages\idf_component_manager\prepare_components\prepare.py", line 27, in prepare_dep_dirs
      ComponentManager(
    File "f:\Espressif\python_env\idf4.4_py3.8_env\lib\site-packages\idf_component_manager\core.py", line 62, in wrapper
      return func(self, *args, **kwargs)
    File "f:\Espressif\python_env\idf4.4_py3.8_env\lib\site-packages\idf_component_manager\core.py", line 438, in prepare_dep_dirs
      downloaded_component_paths, downloaded_component_version_dict = download_project_dependencies(
    File "f:\Espressif\python_env\idf4.4_py3.8_env\lib\site-packages\idf_component_manager\dependencies.py", line 106, in download_project_dependencies
      if is_solve_required(project_requirements, solution):
    File "f:\Espressif\python_env\idf4.4_py3.8_env\lib\site-packages\idf_component_manager\dependencies.py", line 76, in is_solve_required
      if project_requirements.manifest_hash != solution.manifest_hash\
    File "f:\Espressif\python_env\idf4.4_py3.8_env\lib\site-packages\idf_component_tools\manifest\manifest.py", line 310, in manifest_hash
      manifest_hashes = [manifest.manifest_hash for manifest in self.manifests]
    File "f:\Espressif\python_env\idf4.4_py3.8_env\lib\site-packages\idf_component_tools\manifest\manifest.py", line 310, in <listcomp>
      manifest_hashes = [manifest.manifest_hash for manifest in self.manifests]
    File "f:\Espressif\python_env\idf4.4_py3.8_env\lib\site-packages\idf_component_tools\manifest\manifest.py", line 145, in manifest_hash
      serialized = self.serialize(serialize_default=False)  # type: ignore
    File "f:\Espressif\python_env\idf4.4_py3.8_env\lib\site-packages\idf_component_tools\serialization.py", line 63, in _serialize
      serialization_properties[prop] = serialize(getattr(self, prop), serialize_default)
    File "f:\Espressif\python_env\idf4.4_py3.8_env\lib\site-packages\idf_component_tools\serialization.py", line 31, in serialize
      return [serialize(v, serialize_default) for v in value]
    File "f:\Espressif\python_env\idf4.4_py3.8_env\lib\site-packages\idf_component_tools\serialization.py", line 31, in <listcomp>
      return [serialize(v, serialize_default) for v in value]
    File "f:\Espressif\python_env\idf4.4_py3.8_env\lib\site-packages\idf_component_tools\serialization.py", line 34, in serialize
      return value.serialize(serialize_default)
    File "f:\Espressif\python_env\idf4.4_py3.8_env\lib\site-packages\idf_component_tools\serialization.py", line 63, in _serialize
      serialization_properties[prop] = serialize(getattr(self, prop), serialize_default)
    File "f:\Espressif\python_env\idf4.4_py3.8_env\lib\site-packages\idf_component_tools\manifest\manifest.py", line 195, in version_spec
      return self.source.normalize_spec(self._version_spec)
    File "f:\Espressif\python_env\idf4.4_py3.8_env\lib\site-packages\idf_component_tools\sources\git.py", line 196, in normalize_spec
      commit_id = self._client.get_commit_id_by_ref(self.git_repo, self.cache_path(), ref)
    File "f:\Espressif\python_env\idf4.4_py3.8_env\lib\site-packages\idf_component_tools\git_client.py", line 41, in wrapper
      return func(self, *args, **kwargs)
    File "f:\Espressif\python_env\idf4.4_py3.8_env\lib\site-packages\idf_component_tools\git_client.py", line 69, in wrapper
[... cut ...]
  OpenSSL SSL_read: Connection was reset, errno 10054

Call Stack (most recent call first):
  F:/Espressif/esp-idf/tools/cmake/project.cmake:384 (idf_build_process)
  CMakeLists.txt:13 (project)

-- Configuring incomplete, errors occurred!
See also "F:/ESP_Code/watering_demo/build/CMakeFiles/CMakeOutput.log".

 *  终端进程“C:\WINDOWS\System32\WindowsPowerShell\v1.0\powershell.exe -Command cmake -G Ninja -DPYTHON_DEPS_CHECKED=1 -DESP_PLATFORM=1 -S f:\ESP_Code\watering_demo -B f:\ESP_Code\watering_demo\build”已终止，退出代码: 1。

ailunwowo, 2023-04-01:
> idf_component.yml 先将 esp_rainmaker 的 Github 的地址 换成下面的 https://gitee.com/EspressifSystems/esp-rainmaker.git

在换成您这个地址后报这些错
[2003/2005] Linking CXX executable watering_demo.elf
FAILED: watering_demo.elf
cmd.exe /C "cd . && F:\Espressif\tools\xtensa-esp32s3-elf\esp-2021r2-patch5-8.4.0\xtensa-esp32s3-elf\bin\xtensa-esp32s3-elf-g++.exe -mlongcalls  @CMakeFiles\watering_demo.elf.rsp -o watering_demo.elf  && cd 
."
f:/espressif/tools/xtensa-esp32s3-elf/esp-2021r2-patch5-8.4.0/xtensa-esp32s3-elf/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld.exe: ../managed_components/espressif__esp-sr/lib/esp32s3/libesp_audio_processor.a(checks.cc.obj):(.literal._ZN3rtc12FatalMessageD2Ev+0x10): undefined reference to `std::__cxx11::basic_stringbuf<char, std::char_traits<char>, std::allocator<char> >::str() const &'
f:/espressif/tools/xtensa-esp32s3-elf/esp-2021r2-patch5-8.4.0/xtensa-esp32s3-elf/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld.exe: ../managed_components/espressif__esp-sr/lib/esp32s3/libesp_audio_processor.a(checks.cc.obj):(.literal.rtc_FatalMessage+0x0): undefined reference to `std::__cxx11::basic_ostringstream<char, std::char_traits<char>, std::allocator<char> >::basic_ostringstream()'
f:/espressif/tools/xtensa-esp32s3-elf/esp-2021r2-patch5-8.4.0/xtensa-esp32s3-elf/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld.exe: ../managed_components/espressif__esp-sr/lib/esp32s3/libesp_audio_processor.a(checks.cc.obj): in function `std::__cxx11::basic_ostringstream<char, std::char_traits<char>, std::allocator<char> >::str() const &':
/home/sunxiangyu/.espressif/tools/xtensa-esp32s3-elf/esp-2022r1-11.2.0/xtensa-esp32s3-elf/xtensa-esp32s3-elf/include/c++/11.2.0/sstream:917: undefined reference to `std::__cxx11::basic_stringbuf<char, std::char_traits<char>, std::allocator<char> >::str() const &'
f:/espressif/tools/xtensa-esp32s3-elf/esp-2021r2-patch5-8.4.0/xtensa-esp32s3-elf/bin/../lib/gcc/xtensa-esp32s3-elf/8.4.0/../../../../xtensa-esp32s3-elf/bin/ld.exe: ../managed_components/espressif__esp-sr/lib/esp32s3/libesp_audio_processor.a(checks.cc.obj): in function `rtc::FatalMessage::FatalMessage(char const*, int)':
/home/sunxiangyu/workspace/esp_sr_lib/build/../components/esp_audio_processor/rtc_base/checks.cc:69: undefined reference to `std::__cxx11::basic_ostringstream<char, std::char_traits<char>, std::allocator<char> >::basic_ostringstream()'
collect2.exe: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.

ailunwowo, 2023-04-01:
@wujiangang

ailunwowo, 2023-04-01:
求助

espressif2022, 2023-04-03:
现在报错是在 esp-sr ,拉取的 esp-sr 包是哪个版本，v1.1.0？
你当前的 esp-idf 和 工程的 commit ID 分别是多少？git status . 看下本地有啥修改。

ailunwowo, 2023-04-07:
我编译0.3.0的工厂demo也报错，esp-idf版本是4.4.4，这是为啥？

-- Found Git: C:/Espressif/tools/idf-git/2.34.2/cmd/git.exe (found version "2.34.1.windows.1") 
-- Not find RMAKER_PATH, default is D:/LiAiGuo/ESP32S3/esp-box-0.3.0/examples/factory_demo/../../components/esp-rainmaker
-- Component directory D:/LiAiGuo/ESP32S3/esp-box-0.3.0/components/esp-rainmaker does not contain a CMakeLists.txt file. No component will be added
-- Component directory D:/LiAiGuo/ESP32S3/esp-box-0.3.0/components/esp-rainmaker/components/esp-insights does not contain a CMakeLists.txt file. No component will be added
-- Component directory D:/LiAiGuo/ESP32S3/esp-box-0.3.0/components/esp-rainmaker/components/rmaker_common does not contain a CMakeLists.txt file. No component will be added
-- Component directory D:/LiAiGuo/ESP32S3/esp-box-0.3.0/examples/factory_demo/components/esp-rainmaker does not contain a CMakeLists.txt file. No component will 
be added
-- The C compiler identification is GNU 8.4.0
-- The CXX compiler identification is GNU 8.4.0
-- The ASM compiler identification is GNU
-- Found assembler: C:/Espressif/tools/xtensa-esp32s3-elf/esp-2021r2-patch5-8.4.0/xtensa-esp32s3-elf/bin/xtensa-esp32s3-elf-gcc.exe
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: C:/Espressif/tools/xtensa-esp32s3-elf/esp-2021r2-patch5-8.4.0/xtensa-esp32s3-elf/bin/xtensa-esp32s3-elf-gcc.exe - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: C:/Espressif/tools/xtensa-esp32s3-elf/esp-2021r2-patch5-8.4.0/xtensa-esp32s3-elf/bin/xtensa-esp32s3-elf-g++.exe - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Project is not inside a git repository, or git repository has no commits; will not use 'git describe' to determine PROJECT_VER.
-- Building ESP-IDF components for target esp32s3
Processing 1 dependencies:
[1/1] idf (4.4.4)
-- DEBUG: Use esp-modbus component folder: C:/Espressif/frameworks/esp-idf-v4.4.4/components/freemodbus.
CMake Error at C:/Espressif/frameworks/esp-idf-v4.4.4/tools/cmake/build.cmake:201 (message):
  Failed to resolve component 'rmaker_common'.
Call Stack (most recent call first):
  C:/Espressif/frameworks/esp-idf-v4.4.4/tools/cmake/build.cmake:227 (__build_resolve_and_add_req)
  C:/Espressif/frameworks/esp-idf-v4.4.4/tools/cmake/build.cmake:507 (__build_expand_requirements)
  C:/Espressif/frameworks/esp-idf-v4.4.4/tools/cmake/project.cmake:384 (idf_build_process)
  CMakeLists.txt:22 (project)

-- Configuring incomplete, errors occurred!
See also "D:/LiAiGuo/ESP32S3/esp-box-0.3.0/examples/factory_demo/build/CMakeFiles/CMakeOutput.log".

espressif2022, 2023-04-07:
> 

操作一下：git submodule update --init --recursive

log 提示你的文件夹是空的：
Component directory D:/LiAiGuo/ESP32S3/esp-box-0.3.0/components/esp-rainmaker does not contain a CMakeLists.txt file. No component will be added

ailunwowo, 2023-04-07:
[CMakeOutput.log](https://github.com/espressif/esp-box/files/11179100/CMakeOutput.log)

ailunwowo, 2023-04-07:
> > 
> 
> 操作一下：git submodule update --init --recursive
> 
> log 提示你的文件夹是空的： Component directory D:/LiAiGuo/ESP32S3/esp-box-0.3.0/components/esp-rainmaker does not contain a CMakeLists.txt file. No component will be added

执行命令后返回 fatal: not a git repository (or any of the parent directories): .git

espressif2022, 2023-04-07:
你工程有啥修改的东西，先检查一下，这个 tag 是我们量产的，submodule 的 commit 也是指定的，应该不会啊。
这个拉不下来还有可能，这个 fatal 我也有点不明白。

ailunwowo, 2023-04-07:
什么东西都没修改直接打开编译的

ailunwowo, 2023-04-08:
你们box和box-lite的功放和麦克风那部分是不是电路不太一样啊？

ailunwowo, 2023-04-08:
(1059) ES7210: es7210_write_reg(144): error while writing register
E (1059) ES7210: es7210_config_codec(289): i2c communication error while writing ES7210_RESET_REG00
E (1059) ES7210: es7210_write_reg(144): error while writing register
E (1059) ES7210: es7210_config_volume(339): i2c communication error while writing ES7210_ADC1_DIRECT_DB_REG1B
E (1059) ES8311: es8311_init(329): I2C read/write error

ailunwowo, 2023-04-10:
我换了一个esp-box来编译调试，代码库重新拉取了一遍，语音组件的地址也改了，idf版本是5.0，编译出厂demo，可为什么还是报错，我啥也没改，5.0自带的例程是可以编译成功的，这是为什么呢？

[1005/1347] Building CXX object esp-idf/chmorgan__esp-audio-player/CMakeFiles/__idf_chmorgan__esp-audio-player.dir/audio_player.cpp.obj
D:/LiAiGuo/esp-box/examples/factory_demo/managed_components/chmorgan__esp-audio-player/audio_player.cpp: In function 'esp_err_t audio_player_new(audio_player_config_t)':
D:/LiAiGuo/esp-box/examples/factory_demo/managed_components/chmorgan__esp-audio-player/audio_player.cpp:560:9: warning: type qualifiers ignored on cast result type [-Wignored-qualifiers]
  560 |         (TaskHandle_t * const)  NULL,
      |         ^
[1237/1347] Building C object esp-idf/bsp/CMakeFiles/__idf_bsp.dir/src/boards/esp32_s3_box.c.obj
D:/LiAiGuo/esp-box/components/bsp/src/boards/esp32_s3_box.c:88:61: warning: initialization of 'uint8_t (*)(void *)' {aka 'unsigned char (*)(void *)'} from incompatible pointer type '_Bool (*)(void *)' [-Wincompatible-pointer-types]
   88 |         .custom_button_config.button_custom_get_key_value = bsp_home_button_get,
      |                                                             ^~~~~~~~~~~~~~~~~~~
D:/LiAiGuo/esp-box/components/bsp/src/boards/esp32_s3_box.c:88:61: note: (near initialization for 'BOARD_BTN_ID_config[2].<anonymous>.custom_button_config.button_custom_get_key_value')
[1263/1347] Building C object esp-idf/espressif__esp_rainmaker/CMakeFiles/__idf_espressif__esp_rainmaker.dir/src/core/esp_rmaker_client_data.c.obj
FAILED: esp-idf/espressif__esp_rainmaker/CMakeFiles/__idf_espressif__esp_rainmaker.dir/src/core/esp_rmaker_client_data.c.obj
F:\ESP5.0\tools\xtensa-esp32s3-elf\esp-2022r1-11.2.0\xtensa-esp32s3-elf\bin\xtensa-esp32s3-elf-gcc.exe -DMBEDTLS_CONFIG_FILE=\"mbedtls/esp_config.h\" -ID:/LiAiGuo/esp-box/examples/factory_demo/build/config -ID:/LiAiGuo/esp-box/examples/factory_demo/managed_components/espressif__esp_rainmaker/include -ID:/LiAiGuo/esp-box/examples/factory_demo/managed_components/espressif__esp_rainmaker/src/core -ID:/LiAiGuo/esp-box/examples/factory_demo/managed_components/espressif__esp_rainmaker/src/ota -ID:/LiAiGuo/esp-box/examples/factory_demo/managed_components/espressif__esp_rainmaker/src/console -ID:/LiAiGuo/esp-box/examples/factory_demo/managed_components/espressif__esp_rainmaker/src/mqtt -IF:/ESP5.0/esp-idf/components/newlib/platform_include -IF:/ESP5.0/esp-idf/components/freertos/FreeRTOS-Kernel/include -IF:/ESP5.0/esp-idf/components/freertos/esp_additions/include/freertos -IF:/ESP5.0/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/include -IF:/ESP5.0/esp-idf/components/freertos/esp_additions/include -IF:/ESP5.0/esp-idf/components/esp_hw_support/include -IF:/ESP5.0/esp-idf/components/esp_hw_support/include/soc -IF:/ESP5.0/esp-idf/components/esp_hw_support/include/soc/esp32s3 -IF:/ESP5.0/esp-idf/components/esp_hw_support/port/esp32s3/. -IF:/ESP5.0/esp-idf/components/esp_hw_support/port/esp32s3/private_include -IF:/ESP5.0/esp-idf/components/heap/include -IF:/ESP5.0/esp-idf/components/log/include -IF:/ESP5.0/esp-idf/components/soc/include -IF:/ESP5.0/esp-idf/components/soc/esp32s3/. -IF:/ESP5.0/esp-idf/components/soc/esp32s3/include -IF:/ESP5.0/esp-idf/components/hal/esp32s3/include -IF:/ESP5.0/esp-idf/components/hal/include -IF:/ESP5.0/esp-idf/components/hal/platform_port/include -IF:/ESP5.0/esp-idf/components/esp_rom/include -IF:/ESP5.0/esp-idf/components/esp_rom/include/esp32s3 -IF:/ESP5.0/esp-idf/components/esp_rom/esp32s3 -IF:/ESP5.0/esp-idf/components/esp_common/include -IF:/ESP5.0/esp-idf/components/esp_system/include -IF:/ESP5.0/esp-idf/components/esp_system/port/soc -IF:/ESP5.0/esp-idf/components/esp_system/port/include/private -IF:/ESP5.0/esp-idf/components/xtensa/include -IF:/ESP5.0/esp-idf/components/xtensa/esp32s3/include -IF:/ESP5.0/esp-idf/components/esp_timer/include -IF:/ESP5.0/esp-idf/components/lwip/include -IF:/ESP5.0/esp-idf/components/lwip/include/apps -IF:/ESP5.0/esp-idf/components/lwip/include/apps/sntp -IF:/ESP5.0/esp-idf/components/lwip/lwip/src/include -IF:/ESP5.0/esp-idf/components/lwip/port/esp32/include -IF:/ESP5.0/esp-idf/components/lwip/port/esp32/include/arch -ID:/LiAiGuo/esp-box/examples/factory_demo/managed_components/espressif__rmaker_common/include -IF:/ESP5.0/esp-idf/components/esp_event/include -IF:/ESP5.0/esp-idf/components/protobuf-c/protobuf-c -ID:/LiAiGuo/esp-box/examples/factory_demo/managed_components/espressif__json_parser/upstream/include -ID:/LiAiGuo/esp-box/examples/factory_demo/managed_components/espressif__json_parser/upstream -ID:/LiAiGuo/esp-box/examples/factory_demo/managed_components/espressif__json_generator/include -IF:/ESP5.0/esp-idf/components/wifi_provisioning/include -IF:/ESP5.0/esp-idf/components/protocomm/include/common -IF:/ESP5.0/esp-idf/components/protocomm/include/security -IF:/ESP5.0/esp-idf/components/protocomm/include/transports -IF:/ESP5.0/esp-idf/components/bt/include/esp32s3/include -IF:/ESP5.0/esp-idf/components/bt/common/osi/include -IF:/ESP5.0/esp-idf/components/bt/common/api/include/api -IF:/ESP5.0/esp-idf/components/bt/common/btc/profile/esp/blufi/include -IF:/ESP5.0/esp-idf/components/bt/common/btc/profile/esp/include -IF:/ESP5.0/esp-idf/components/bt/host/nimble/nimble/nimble/host/include -IF:/ESP5.0/esp-idf/components/bt/host/nimble/nimble/nimble/host/services/ans/include -IF:/ESP5.0/esp-idf/components/bt/host/nimble/nimble/nimble/host/services/bas/include -IF:/ESP5.0/esp-idf/components/bt/host/nimble/nimble/nimble/host/services/ [... cut ...]
  133 |         mqtt_conn_params->client_cert_len = esp_rmaker_get_client_cert_len();
      |                           ^~~~~~~~~~~~~~~
      |                           client_cert
cc1.exe: some warnings being treated as errors
[1272/1347] Building C object esp-idf/espressif__esp_rainmaker/CMakeFiles/__idf_espressif__esp_rainmaker.dir/src/core/esp_rmaker_claim.c.obj
ninja: build stopped: subcommand failed.

 *  终端进程“C:\Windows\System32\WindowsPowerShell\v1.0\powershell.exe -Command ninja ”已终止，退出代码: 1。

ailunwowo, 2023-04-10:
> 你工程有啥修改的东西，先检查一下，这个 tag 是我们量产的，submodule 的 commit 也是指定的，应该不会啊。 这个拉不下来还有可能，这个 fatal 我也有点不明白。

有什么办法吗？编译不了

espressif2022, 2023-04-10:
> 

esp-rainmaker 有修改，暂时没匹配和 IDF5.0 的。
先参考 https://github.com/espressif/esp-box/issues/60，用fixed 的 rainmaker 版本号

box 和 box-lite 的 IC 是不一样的，具体参考硬件说明；

另外，上周 tag v0.3.0 编译失败是社么原因呢？

ailunwowo, 2023-04-10:
> > 
> 
> esp-rainmaker 有修改，暂时没匹配和 IDF5.0 的。 先参考 [https://github.com/espressif/esp-box/issues/60，用fixed](https://github.com/espressif/esp-box/issues/60%EF%BC%8C%E7%94%A8fixed) 的 rainmaker 版本号
> 
> box 和 box-lite 的 IC 是不一样的，具体参考硬件说明；
> 
> 另外，上周 tag v0.3.0 编译失败是社么原因呢？

好，就IC不一样吗？那这个demo都是可以用的吗？两个板子都可以用的吗，030那个问题还没解决
我说下我现在需要解决的问题吧
现在就是我以前基于4.4的watering_demo写了一段新功能，然后现在更新5.0后就编译不了了，甚至倒退回4.4也编译不了

espressif2022, 2023-04-10:
好，就IC不一样吗？那这个demo都是可以用的吗？两个板子都可以用的吗，030那个问题还没解决
- 不一样，代码驱动也是不同的，有区别

我说下我现在需要解决的问题吧
现在就是我以前基于4.4的watering_demo写了一段新功能，然后现在更新5.0后就编译不了了，甚至倒退回4.4也编译不了
-社么报错，是 ADC 接口么，看下报错 log？

ailunwowo, 2023-04-10:
> > 
> 
> esp-rainmaker 有修改，暂时没匹配和 IDF5.0 的。 先参考 [https://github.com/espressif/esp-box/issues/60，用fixed](https://github.com/espressif/esp-box/issues/60%EF%BC%8C%E7%94%A8fixed) 的 rainmaker 版本号
> 
> box 和 box-lite 的 IC 是不一样的，具体参考硬件说明；
> 
> 另外，上周 tag v0.3.0 编译失败是社么原因呢？

我更改了esp-rainmaker的地址后还是报错

-- Found Git: C:/Espressif/tools/idf-git/2.34.2/cmd/git.exe (found version "2.34.1.windows.1") 
-- The C compiler identification is GNU 11.2.0
-- The CXX compiler identification is GNU 11.2.0
-- The ASM compiler identification is GNU
-- Found assembler: F:/ESP5.0/tools/xtensa-esp32s3-elf/esp-2022r1-11.2.0/xtensa-esp32s3-elf/bin/xtensa-esp32s3-elf-gcc.exe
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: F:/ESP5.0/tools/xtensa-esp32s3-elf/esp-2022r1-11.2.0/xtensa-esp32s3-elf/bin/xtensa-esp32s3-elf-gcc.exe - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: F:/ESP5.0/tools/xtensa-esp32s3-elf/esp-2022r1-11.2.0/xtensa-esp32s3-elf/bin/xtensa-esp32s3-elf-g++.exe - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Building ESP-IDF components for target esp32s3
CMake Error at F:/ESP5.0/esp-idf/tools/cmake/build.cmake:519 (message):
  Traceback (most recent call last):

    File "runpy.py", line 194, in _run_module_as_main
    File "runpy.py", line 87, in _run_code
    File "f:\ESP5.0\python_env\idf5.0_py3.8_env\lib\site-packages\idf_component_manager\prepare_components\__main__.py", line 6, in <module>
      main()
    File "f:\ESP5.0\python_env\idf5.0_py3.8_env\lib\site-packages\idf_component_manager\prepare_components\prepare.py", line 124, in main
      args.func(args)
    File "f:\ESP5.0\python_env\idf5.0_py3.8_env\lib\site-packages\idf_component_manager\prepare_components\prepare.py", line 27, in prepare_dep_dirs        
      ComponentManager(
    File "f:\ESP5.0\python_env\idf5.0_py3.8_env\lib\site-packages\idf_component_manager\core.py", line 62, in wrapper
      return func(self, *args, **kwargs)
    File "f:\ESP5.0\python_env\idf5.0_py3.8_env\lib\site-packages\idf_component_manager\core.py", line 438, in prepare_dep_dirs
      downloaded_component_paths, downloaded_component_version_dict = download_project_dependencies(
    File "f:\ESP5.0\python_env\idf5.0_py3.8_env\lib\site-packages\idf_component_manager\dependencies.py", line 106, in download_project_dependencies        
      if is_solve_required(project_requirements, solution):
    File "f:\ESP5.0\python_env\idf5.0_py3.8_env\lib\site-packages\idf_component_manager\dependencies.py", line 76, in is_solve_required
      if project_requirements.manifest_hash != solution.manifest_hash\
    File "f:\ESP5.0\python_env\idf5.0_py3.8_env\lib\site-packages\idf_component_tools\manifest\manifest.py", line 310, in manifest_hash
      manifest_hashes = [manifest.manifest_hash for manifest in self.manifests]
    File "f:\ESP5.0\python_env\idf5.0_py3.8_env\lib\site-packages\idf_component_tools\manifest\manifest.py", line 310, in <listcomp>
      manifest_hashes = [manifest.manifest_hash for manifest in self.manifests]
    File "f:\ESP5.0\python_env\idf5.0_py3.8_env\lib\site-packages\idf_component_tools\manifest\manifest.py", line 145, in manifest_hash
      serialized = self.serialize(serialize_default=False)  # type: ignore
    File "f:\ESP5.0\python_env\idf5.0_py3.8_env\lib\site-packages\idf_component_tools\serialization.py", line 63, in _serialize
      serialization_properties[prop] = serialize(getattr(self, prop), serialize_default)
    File "f:\ESP5.0\python_env\idf5.0_py3.8_env\lib\site-packages\idf_component_tools\serialization.py", line 31, in serialize
      return [serialize(v, serialize_default) for v in value]
    File "f:\ESP5.0\python_env\idf5.0_py3.8_env\lib\site-packages\idf_component_tools\serialization.py", line 31, in <listcomp>
      return [serialize(v, serialize_default) for v in value]
    File "f:\ESP5.0\python_env\idf5.0_py3.8_env\lib\site-packages\idf_component_tools\serialization.py", line 34, in serialize
      return value.serialize(serialize_default)
    File "f:\ESP5.0\python_env\idf5.0_py3.8_env\lib\site-packages\idf_component_tools\serialization.py", line 63, in _serialize
      serialization_properties[prop] = serialize(getattr(self, prop), serialize_default)
    File "f:\ESP5.0\python_env\idf5.0_py3.8_env\lib\site-packages\idf_component_tools\manifest\manifest.py", line 195, in version_spec
      return self.source.normalize_spec(self._version_spec)
    File "f:\ESP5.0\python_env\idf5.0_py3.8_env\lib\site-packages\idf_component_tools\sources\git.py", line 196, in normalize_spec
      commit_id = self._client.get_commit_id_by_ref(self.git_repo, self.cache_path(), ref)
    File "f:\ESP5.0\python_env\idf5.0_py3.8_env\lib\site-packages\idf_component_tools\git_client.py", line 41, in wrapper
      return func(self, *args, **kwargs)
    File "f:\ESP5.0\python_env\idf5.0_py3.8_env\lib\site-packages\idf_component_tools\git_client.py", line 72, in wrapper
      return func(self, *args, **kwargs)
    File "f:\ESP5.0\python_env\idf5.0_py3.8_env\lib\site-packages\idf_component_tools\git_client.py", line 159, in get_commit_id_by_ref
      ref = self.run(['ls-remote', '--exit-code', 'origin', 'HEAD'], cwd=bare_path)[:40]
[... cut ...]
Call Stack (most recent call first):
  F:/ESP5.0/esp-idf/tools/cmake/project.cmake:440 (idf_build_process)
  CMakeLists.txt:14 (project)

-- Configuring incomplete, errors occurred!
See also "D:/LiAiGuo/esp-box/examples/factory_demo/build/CMakeFiles/CMakeOutput.log".

 *  终端进程“C:\Windows\System32\WindowsPowerShell\v1.0\powershell.exe -Command cmake -G Ninja -DPYTHON_DEPS_CHECKED=1 -DESP_PLATFORM=1 -B d:\LiAiGuo\esp-box\examples\factory_demo\build -S d:\LiAiGuo\esp-box\examples\factory_demo”已终止，退出代码: 1。

ailunwowo, 2023-04-10:
> * 口么

hal/efuse_hal.h: No such file or directory

[{
	"resource": "/c:/Espressif/frameworks/esp-idf-v4.4.4/components/bootloader_support/src/bootloader_clock_init.c",
	"owner": "cpp",
	"severity": 8,
	"message": "hal/efuse_hal.h: No such file or directory",
	"startLineNumber": 14,
	"startColumn": 10,
	"endLineNumber": 14,
	"endColumn": 10
}]

espressif2022, 2023-04-10:
你刚才 github 的 esp-rainmaker 不是正常拉取的么，换个 commit id 就失败了 ，换 gitee 的链接 https://gitee.com/EspressifSystems/esp-rainmaker.git 
你上述两次报错原因，完全是两种情况，
1 是 esp-rainmaker 调用接口和最新 ESP_IDF 没匹配
2 是仓库拉取失败了
你要不先自己简单试试，你是有一个 ok 的编译环境，就是 esp-rainmaker 这编译不过吧？

ailunwowo, 2023-04-10:
> https://gitee.com/EspressifSystems/esp-rainmaker.git

1 是 esp-rainmaker 调用接口和最新 ESP_IDF 没匹配
那这个怎么办呢，编译不过的话，那如果我用030的话是离线的应该没问题吧

ailunwowo, 2023-04-10:
> hal/efuse_hal.h: No such file or directory
>
> [{ "resource": "/c:/Espressif/frameworks/esp-idf-v4.4.4/components/bootloader_support/src/bootloader_clock_init.c", "owner": "cpp", "severity": 8, [...]

这个是啥原因呢？

espressif2022, 2023-04-10:
> > https://gitee.com/EspressifSystems/esp-rainmaker.git
> 
> 1 是 esp-rainmaker 调用接口和最新 ESP_IDF 没匹配 那这个怎么办呢，编译不过的话，那如果我用030的话是离线的应该没问题吧

不匹配的，可以调下。你现在不是拉不下。

espressif2022, 2023-04-10:
> 这个是啥原因呢？

你不是说是 5.0 的？咋又是 v4.4.4 ？

ailunwowo, 2023-04-10:
> 你不是说是 5.0 的？咋又是 v4.4.4 ？

不好意思看错了😥我以为你要我4.4的报错，下面是5.0的报错(不行的话我发工程给你你可以帮我编译一下试试吗？)
-- Found Git: C:/Espressif/tools/idf-git/2.34.2/cmd/git.exe (found version "2.34.1.windows.1") 
-- The C compiler identification is GNU 11.2.0
-- The CXX compiler identification is GNU 11.2.0
-- The ASM compiler identification is GNU
-- Found assembler: F:/ESP5.0/tools/xtensa-esp32s3-elf/esp-2022r1-11.2.0/xtensa-esp32s3-elf/bin/xtensa-esp32s3-elf-gcc.exe
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: F:/ESP5.0/tools/xtensa-esp32s3-elf/esp-2022r1-11.2.0/xtensa-esp32s3-elf/bin/xtensa-esp32s3-elf-gcc.exe - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: F:/ESP5.0/tools/xtensa-esp32s3-elf/esp-2022r1-11.2.0/xtensa-esp32s3-elf/bin/xtensa-esp32s3-elf-g++.exe - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Project is not inside a git repository, or git repository has no commits; will not use 'git describe' to determine PROJECT_VER.
-- Building ESP-IDF components for target esp32s3
Processing 13 dependencies:
[1/13] espressif/button (2.5.1)
[2/13] espressif/cmake_utilities (0.4.4)
[3/13] espressif/esp-dsp (1.3.0)
[4/13] espressif/esp-sr (1.3.0)
[5/13] espressif/esp_rainmaker (cc861bffc818c607e799f66c79923fd4e2d00447)
[6/13] espressif/esp_schedule (cc861bffc818c607e799f66c79923fd4e2d00447)
[7/13] espressif/esp_secure_cert_mgr (2.2.1)
[8/13] espressif/json_generator (1.1.1)
[9/13] espressif/json_parser (1.0.0)
[10/13] espressif/mdns (1.0.9)
[11/13] espressif/qrcode (0.1.0)
[12/13] espressif/rmaker_common (1.3.0)
[13/13] idf (5.0.1)
CMake Error at F:/ESP5.0/esp-idf/tools/cmake/build.cmake:245 (message):
  Failed to resolve component 'esp32s3'.
Call Stack (most recent call first):
  F:/ESP5.0/esp-idf/tools/cmake/build.cmake:287 (__build_resolve_and_add_req)
  F:/ESP5.0/esp-idf/tools/cmake/build.cmake:282 (__build_expand_requirements)
  F:/ESP5.0/esp-idf/tools/cmake/build.cmake:282 (__build_expand_requirements)
  F:/ESP5.0/esp-idf/tools/cmake/build.cmake:282 (__build_expand_requirements)
  F:/ESP5.0/esp-idf/tools/cmake/build.cmake:282 (__build_expand_requirements)
  F:/ESP5.0/esp-idf/tools/cmake/build.cmake:282 (__build_expand_requirements)
  F:/ESP5.0/esp-idf/tools/cmake/build.cmake:282 (__build_expand_requirements)
  F:/ESP5.0/esp-idf/tools/cmake/build.cmake:282 (__build_expand_requirements)
  F:/ESP5.0/esp-idf/tools/cmake/build.cmake:282 (__build_expand_requirements)
  F:/ESP5.0/esp-idf/tools/cmake/build.cmake:282 (__build_expand_requirements)
  F:/ESP5.0/esp-idf/tools/cmake/build.cmake:282 (__build_expand_requirements)
  F:/ESP5.0/esp-idf/tools/cmake/build.cmake:574 (__build_expand_requirements)
  F:/ESP5.0/esp-idf/tools/cmake/project.cmake:440 (idf_build_process)
  CMakeLists.txt:11 (project)

-- Configuring incomplete, errors occurred!
See also "C:/ST/watering_demo/build/CMakeFiles/CMakeOutput.log".

 *  终端进程“C:\Windows\System32\WindowsPowerShell\v1.0\powershell.exe -Command cmake -G Ninja -DPYTHON_DEPS_CHECKED=1 -DESP_PLATFORM=1 -S c:\ST\watering_demo -B c:\ST\watering_demo\build”已终止，退出代码: 1。

ailunwowo, 2023-04-10:
> 你不是说是 5.0 的？咋又是 v4.4.4 ？

能帮忙编译一下吗？

espressif2022, 2023-04-10:
我这是好的， 
麻烦把你 esp-idf 的 commit id，box 工程的 commit id 给下

ailunwowo, 2023-04-10:
> 我这是好的， 麻烦把你 esp-idf 的 commit id，box 工程的 commit id 给下

 esp-idf 的  a4afa44435ef4488d018399e1de50ad2ee964be8
box工程我可以直接发压缩包邮箱给你吗

espressif2022, 2023-04-10:
> > 我这是好的， 麻烦把你 esp-idf 的 commit id，box 工程的 commit id 给下
> 
> esp-idf 的 a4afa44435ef4488d018399e1de50ad2ee964be8 box工程我可以直接发压缩包邮箱给你吗

不用，你给我工程的 commit id 啊，我这保证你基础例程是 ok 的，我在准备一个 MR，写死 rainmaker 的版本号

ailunwowo, 2023-04-10:
> > > 我这是好的， 麻烦把你 esp-idf 的 commit id，box 工程的 commit id 给下
> > 
> > 
> > esp-idf 的 a4afa44435ef4488d018399e1de50ad2ee964be8 box工程我可以直接发压缩包邮箱给你吗
> 
> 不用，你给我工程的 commit id 啊，我这保证你基础例程是 ok 的，我在准备一个 MR，写死 rainmaker 的版本号

请问一下咋看？

espressif2022, 2023-04-10:
> > > > 我这是好的， 麻烦把你 esp-idf 的 commit id，box 工程的 commit id 给下
> > > 
> > > 
> > > esp-idf 的 a4afa44435ef4488d018399e1de50ad2ee964be8 box工程我可以直接发压缩包邮箱给你吗
> > 
> > 
> > 不用，你给我工程的 commit id 啊，我这保证你基础例程是 ok 的，我在准备一个 MR，写死 rainmaker 的版本号
> 
> 请问一下咋看？
进入工程目录， git log，第一条

ailunwowo, 2023-04-10:
> > > > > 我这是好的， 麻烦把你 esp-idf 的 commit id，box 工程的 commit id 给下
> > > > 
> > > > 
> > > > esp-idf 的 a4afa44435ef4488d018399e1de50ad2ee964be8 box工程我可以直接发压缩包邮箱给你吗
> > > 
> > > 
> > > 不用，你给我工程的 commit id 啊，我这保证你基础例程是 ok 的，我在准备一个 MR，写死 rainmaker 的版本号
> > 
> > 
> > 请问一下咋看？
> > 进入工程目录， git log，第一条

好的谢谢这是工程的，a83643cfb1d2ed1c0928457e072f3f831d1e51bc

espressif2022, 2023-04-10:
> > > > > > 我这是好的， 麻烦把你 esp-idf 的 commit id，box 工程的 commit id 给下
> > > > > 
> > > > > 
> > > > > esp-idf 的 a4afa44435ef4488d018399e1de50ad2ee964be8 box工程我可以直接发压缩包邮箱给你吗
> > > > 
> > > > 
> > > > 不用，你给我工程的 commit id 啊，我这保证你基础例程是 ok 的，我在准备一个 MR，写死 rainmaker 的版本号
> > > 
> > > 
> > > 请问一下咋看？
> > > 进入工程目录， git log，第一条
> 
> 好的谢谢这是工程的，a83643cfb1d2ed1c0928457e072f3f831d1e51bc

我咋搜不到这条 commit ？能都把前后的 log 发看下

[11 more replies]

closed 2023-07-11 by ESP-Mars