espressif/esp-box #64: watering_demo中存在idf版本冲突问题。

MeowStatus, 2023-04-06:
如题。
idf_component.yml中要求idf≥5.0，readme中要求idf版本为v4.4。
最终导致无论用5.0还是4.4都不能成功编译该例程。
其他例程中也有类似的问题，望尽快修复。

espressif2022, 2023-04-06:
在 ESP-IDF 5.0 上，是 esp-rainmaker 编译失败吗？

MeowStatus, 2023-04-06:
> 在 ESP-IDF 5.0 上，是 esp-rainmaker 编译失败吗？
是。
factory_demo中也是同样的问题。
相关log如下：
[image]

espressif2022, 2023-04-06:
稍等，验证有问题。我换个 version， 验证一下，一会回复

espressif2022, 2023-04-06:
> > 在 ESP-IDF 5.0 上，是 esp-rainmaker 编译失败吗？
> 
> 是。 factory_demo中也是同样的问题。 相关log如下： [image]

https://github.com/espressif/esp-box/issues/60 麻烦先参考这个 issue，验证 ok，谢谢。

MeowStatus, 2023-04-06:
> #60 麻烦先参考这个 issue，验证 ok，谢谢。

验证可行，感谢！
此issue您可关闭。

espressif2022, 2023-04-06:
> > #60 麻烦先参考这个 issue，验证 ok，谢谢。
> 
> 验证可行，感谢！ 此issue您可关闭。

请帮忙 close 下，我这好像没权限

closed 2023-04-06 by MeowStatus