espressif/esp-box #65: Unable to flash box after build and flash of switch example from esp-rainmaker

corneo, 2023-04-06:
I recently got a ESP32-S3-BOX.  I am (was?) interested in using rainmaker to control/monitor an app I am (was?) developing.  Following the rainmaker getting started doc, I built and flashed the switch example.  The build and flash went fine and the application worked, that is, I was able to turn the led on/off using the top button and the rainmaker app on IOS.  I also ran monitor and observed the log.  Now the trouble starts...

I want to flash another example, specifically, hello_world. The build works fine, but using the same process I used to flash the switch example, the flash tool is unable to connect.  I thought a 'factory-reset' would help resolve that.  Noting that switch was designed to factory-reset the chip by pressing and holding boot (maybe reset) on the board fore more than 10 seconds, I gave that a try. (?? I wonder what motivated the addition of this 'factory-reset'  capability. I haven't seen awfully many apps, but I have never seen this built-in. ??)  I observed the monitor/console while doing this.  I saw the message after three seconds allowing WiFi reset, then 7 seconds later, the factory-reset message.  I released the button and got a message saying the board would reset in 2 seconds.  It did reboot but there is no indication that it was reset to factory default; nothing changed.  And now, not remembering what I did since to change things, but the monitor will no longer connect.  Nothing I have tried so far, including following advice [here](https://github.com/espressif/esp-box/issues/36) about disabling esp stub.

So... how can I factory reset the chip to recover the box.  It's useless to me, I think, until I can make that happen.  Or... can you recommend a different option?  E.g. is there a way to reset the box with a particular strapping pin configuration, which I know VERY little about so far, but I have the sense that I might have/get to learn more about it if I am to get out of this pickle.

I did try to post this in the Espressif community for the box but the site was not responding when I tried to create a new item.  I might try that again today.

I am able to flash other kits using the same cable I used to originally flash the box, so it's not a cable issue.

espressif2022, 2023-04-07:
Is it impossible to flash the fw again?
Press and hold the boot button in the upper left corner, and then do power on, let the box enter the boot flash mode.
It may be the USB port occupied by your application.

corneo, 2023-04-07:
I seems impossible to flash the firmware; I have not been able to.  The idf.py monitor cannot connect to the box, although the usb port shows and disappears on my laptop when I plug-in and remove the box. It is the Samp port and cable I used to flash the switch code when it first worked. I have tried many combinations of button presses while both powered-up and powering-up.  It seems that the switch example is running because the top led toggles each time I press the top button.  Also... it connects to BT; I can see PROV_xxxxxx in bluetooth settings on my phone.

corneo, 2023-04-07:
To be (reasonably) certain that the USB cable I am using has somehow spontaneously failed, I just used it to flash a Heltec WiFi Kit 32 V3.  Works like a charm.

I also tried to flash only the boot loader [idf.py boot loader boot loader-flash].  The bootloader build appears to have worked fine but, of course, the bootloader-flash did not.

I was able to figure out some time ago how to turn on verbose logging during the flash, but I can't seem to figure that our again today.

Any advice about fixing this would certainly be appreciated.  Maybe it's just plain broken.  Is there a warranty or return policy?

corneo, 2023-04-07:
But wait... there's more... I can hold down the boot button and while doing so press the top button.  Amazing... the LED toggles just fine.  Same for the rest button.  Hold it down and the top button still toggles the LED.  It also toggles immediately after releasing either button.  It is as if those buttons are somehow disconnected.  I thought perhaps the flappy bits of plastic were not doing their job so I removed the cover and still no love.  I can feel the 'tactile feedback' the same on all buttons so I don't have any reason to believe they are not working.  It's time to get out the meter.  (;->)

Is there some combination of pins I could pull high or low, i.e. strapping pins (?)?

I have ordered a couple more of these, in case the one I have is just plain broken.  I certainly hope I don't discover that I have thrown good money after bad.

espressif2022, 2023-04-10:
[image]
I want to confirm, are you pressing the boot btn or the mute btn, the top btn is the mute button.
Long press the boot btn, press the reset btn to enter the boot programming mode.

corneo, 2023-04-16:
I think my [recent post](https://github.com/espressif/esp-box/issues/65#issuecomment-1500614450)answers [your question](https://github.com/espressif/esp-box/issues/65#issuecomment-1501310065).  I apologize for not seeing you question sooner and clarifying.  If I have missed and hence failed to respond to some aspect of your question please let me know.  Thank you for your time and assistance.

espressif2022, 2023-04-18:
> I think my [recent post](https://github.com/espressif/esp-box/issues/65#issuecomment-1500614450)answers [your question](https://github.com/espressif/esp-box/issues/65#issuecomment-1501310065). I [...]

So you still can't download your FW now, even the ESP_BOX is in boot mode?

corneo, 2023-04-18:
I was finally able to get the box to work again.  I discovered that the device can be flashed from the monitor

> So you still can't download your FW now, even the ESP_BOX is in boot mode?

I was unable to get the box into boot mode, as far as I could tell, using the boot button, nor any other button or combination of button sequences.  If it was going into boot mode it was not at all apparent.  In fact, never did the 'app' stop responding to toggling the top led with the top button.  I assume if the device went into boot mode that would work.

Nonetheless... today I decided to take another crack at it.  After a LOT of hunting around and trying stuff... I discovered that the device could be flashed from the monitor with Ctrl+T-Crtl+F.  I was able to flash a different example (blink, I think) using the monitor.  Then I was able to build and flash lv_demos (whatever the default option is) and it seems to be working great.

I have little confidence that flashing via the monitor is really what resolved the problem, because I tried quite a lot of things, and really didn't keep track of what I tried.  So... I'm working again.

If I learn more about it, i.e. recreate the issue and 'fix' it with a monitor flash, I'll try to remember to post that here.

corneo, 2023-04-18:
I think this is why the boot button did not work... from from https://github.com/espressif/esp-rainmaker/tree/master/examples/switch#readme ---

 `This example uses the BOOT button and RGB LED on the ESP32-S2-Saola-1/ESP32-C3-DevKitC board to demonstrate a switch.`

I built and flashed the switch example again and it worked.  I immediately tried again and it failed to flash.

I was having trouble getting the device to connect to the monitor.  I discovered that it I press the bottom-left-side button, and while holding it down press the upper-left-side-button, then release the lower-left-side button, then release the upper-left-side button... whew... the monitor displays:

```
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x15 (USB_UART_CHIP_RESET),boot:0x3 (DOWNLOAD(USB/UART0))
Saved PC:0x40041a76
waiting for download
```
 Then I can build/flash with Ctrl+T-Ctrl+F and it flashes OK...

OR, it seems, I can quit the monitor with Ctrl-] and then flash with idf.py flash.

So... I think I can get the box to flash now regardless, if I follow the specific sequence described above.

I wanted to re-flash the factory_demo but it will not build.  I get errors:

> /Users/tim/esp/esp-box/examples/factory_demo/managed_components/espressif__esp_rainmaker/src/core/esp_rmaker_client_data.c: In function 'esp_rmaker_get_client_cert_len':
/Users/tim/esp/esp-box/examples/factory_demo/managed_components/espressif__esp_rainmaker/src/core/esp_rmaker_client_data.c:78:12: error: implicit declaration of function 'esp_rmaker_factory_get_size'; did you mean 'esp_rmaker_factory_get'? [-Werror=implicit-function-declaration]
   78 |     return esp_rmaker_factory_get_size(ESP_RMAKER_CLIENT_CERT_NVS_KEY) + 1; /* +1 for NULL terminating byte */
      |            ^~~~~~~~~~~~~~~~~~~~~~~~~~~
      |            esp_rmaker_factory_get
/Users/tim/esp/esp-box/examples/factory_demo/managed_components/espressif__esp_rainmaker/src/core/esp_rmaker_client_data.c: In function 'esp_rmaker_get_mqtt_conn_params':
/Users/tim/esp/esp-box/examples/factory_demo/managed_components/espressif__esp_rainmaker/src/core/esp_rmaker_client_data.c:128:27: error: 'esp_rmaker_mqtt_conn_params_t' has no member named 'client_key_len'; did you mean 'client_key'?
  128 |         mqtt_conn_params->client_key_len = esp_rmaker_get_client_key_len();
      |                           ^~~~~~~~~~~~~~
      |                           client_key
/Users/tim/esp/esp-box/examples/factory_demo/managed_components/espressif__esp_rainmaker/src/core/esp_rmaker_client_data.c:133:27: error: 'esp_rmaker_mqtt_conn_params_t' has no member named 'client_cert_len'; did you mean 'client_cert'?
  133 |         mqtt_conn_params->client_cert_len = esp_rmaker_get_client_cert_len();
      |                           ^~~~~~~~~~~~~~~
      |                           client_cert
cc1: some warnings being treated as errors

I'll see if I can figure this out on my own. Maybe it's resolved or at least reported on the rainmaker repository.

closed 2023-04-18 by corneo

espressif2022, 2023-04-19:
> I think this is why the boot button did not work... from from https://github.com/espressif/esp-rainmaker/tree/master/examples/switch#readme ---
>
> `This example uses the BOOT button and RGB LED on the [...]
git pull the new version, it has been fixed now, they changed the api name.