espressif/esp-box #66: 我想把uac，hid的代码移植到我其他的项目上，需要把哪些部分移植过去呢？

weiqifa0, 2023-04-08:
我尝试直接拷贝headset下面的核心代码过去，发现有非常多的错误。

weiqifa0, 2023-04-08:
https://github.com/weiqifa0/esp-box

我把代码放到了这个位置

weiqifa0, 2023-04-08:
在headset例子程序里面

weiqifa0, 2023-04-08:
[image]
我已经把需要的放到这个位置了，但是编译的时候还是提示找不到文件。

[8/19] Building C object esp-idf/main/CMakeFiles/__idf_main.dir/src/page_fft.c.objFAILED: esp-idf/main/CMakeFiles/__idf_main.dir/src/page_fft.c.obj
/Users/crisqifawei/.espressif/tools/xtensa-esp32s3-elf/esp-2022r1-11.2.0/xtensa-esp32s3-elf/bin/xtensa-esp32s3-elf-gcc -DLV_CONF_INCLUDE_SIMPLE -DMBEDTLS_CONFIG_FILE=\"mbedtls/esp_config.h\" -DUNITY_INCLUDE_CONFIG_H -I/Users/crisqifawei/esp/esp-box/examples/usb_headset/build/config -I/Users/crisqifawei/esp/esp-box/examples/usb_headset/main/include -I/Users/crisqifawei/esp/esp-idf/components/newlib/platform_include -I/Users/crisqifawei/esp/esp-idf/components/freertos/FreeRTOS-Kernel/include -I/Users/crisqifawei/esp/esp-idf/components/freertos/esp_additions/include/freertos -I/Users/crisqifawei/esp/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/include -I/Users/crisqifawei/esp/esp-idf/components/freertos/esp_additions/include -I/Users/crisqifawei/esp/esp-idf/components/esp_hw_support/include -I/Users/crisqifawei/esp/esp-idf/components/esp_hw_support/include/soc -I/Users/crisqifawei/esp/esp-idf/components/esp_hw_support/include/soc/esp32s3 -I/Users/crisqifawei/esp/esp-idf/components/esp_hw_support/port/esp32s3/. -I/Users/crisqifawei/esp/esp-idf/components/esp_hw_support/port/esp32s3/private_include -I/Users/crisqifawei/esp/esp-idf/components/heap/include -I/Users/crisqifawei/esp/esp-idf/components/log/include -I/Users/crisqifawei/esp/esp-idf/components/soc/include -I/Users/crisqifawei/esp/esp-idf/components/soc/esp32s3/. -I/Users/crisqifawei/esp/esp-idf/components/soc/esp32s3/include -I/Users/crisqifawei/esp/esp-idf/components/hal/esp32s3/include -I/Users/crisqifawei/esp/esp-idf/components/hal/include -I/Users/crisqifawei/esp/esp-idf/components/hal/platform_port/include -I/Users/crisqifawei/esp/esp-idf/components/esp_rom/include -I/Users/crisqifawei/esp/esp-idf/components/esp_rom/include/esp32s3 -I/Users/crisqifawei/esp/esp-idf/components/esp_rom/esp32s3 -I/Users/crisqifawei/esp/esp-idf/components/esp_common/include -I/Users/crisqifawei/esp/esp-idf/components/esp_system/include -I/Users/crisqifawei/esp/esp-idf/components/esp_system/port/soc -I/Users/crisqifawei/esp/esp-idf/components/esp_system/port/include/private -I/Users/crisqifawei/esp/esp-idf/components/xtensa/include -I/Users/crisqifawei/esp/esp-idf/components/xtensa/esp32s3/include -I/Users/crisqifawei/esp/esp-idf/components/lwip/include -I/Users/crisqifawei/esp/esp-idf/components/lwip/include/apps -I/Users/crisqifawei/esp/esp-idf/components/lwip/include/apps/sntp -I/Users/crisqifawei/esp/esp-idf/components/lwip/lwip/src/include -I/Users/crisqifawei/esp/esp-idf/components/lwip/port/esp32/include -I/Users/crisqifawei/esp/esp-idf/components/lwip/port/esp32/include/arch -I/Users/crisqifawei/esp/esp-box/examples/usb_headset/managed_components/espressif__es7210/include -I/Users/crisqifawei/esp/esp-idf/components/driver/include -I/Users/crisqifawei/esp/esp-idf/components/driver/deprecated -I/Users/crisqifawei/esp/esp-idf/components/driver/esp32s3/include -I/Users/crisqifawei/esp/esp-idf/components/esp_pm/include -I/Users/crisqifawei/esp/esp-idf/components/esp_ringbuf/include -I/Users/crisqifawei/esp/esp-box/examples/usb_headset/managed_components/espressif__esp-box/include -I/Users/crisqifawei/esp/esp-idf/components/spiffs/include -I/Users/crisqifawei/esp/esp-idf/components/esp_partition/include -I/Users/crisqifawei/esp/esp-box/examples/usb_headset/managed_components/espressif__esp_codec_dev/include -I/Users/crisqifawei/esp/esp-box/examples/usb_headset/managed_components/espressif__esp_codec_dev/interface -I/Users/crisqifawei/esp/esp-box/examples/usb_headset/managed_components/espressif__esp_codec_dev/device/include -I/Users/crisqifawei/esp/esp-box/examples/usb_headset/managed_components/espressif__esp_lvgl_port/include -I/Users/crisqifawei/esp/esp-idf/components/esp_lcd/include -I/Users/crisqifawei/esp/esp-idf/components/esp_lcd/interface -I/Users/crisqifawei/esp/esp-box/examples/usb_headset/managed_components/lvgl__lvgl -I/Users/crisqifawei/esp/esp-box/examples/usb_headset/managed_components/lvgl__lvgl/src -I/Users/crisqifawei/esp/esp-box/examples/usb_headset/managed_components -I/Users/crisqifawei/esp/esp-box/examples/usb_headset/managed_components/espressif__esp-dsp/modules/dotprod/include -I/Users/crisqifawei/esp/esp-box/examples/usb_headset/managed_components/espressif__esp-dsp/modules/support/include -I/Users/crisqifawei/esp/esp-box/examples/usb_headset/managed_components/espressif__esp-dsp/modules/windows/include -I/Users/crisqifawei/esp/esp-box/examples/usb_headset/managed_components/espressif__esp-dsp/modules/windows/hann/include -I/Users/crisqifawei/esp/esp-box/examples/usb_headset/managed_components/espressif__esp-dsp/modules/windows/blackman/include -I/Users/crisqifawei/esp/esp-box/examples/usb_headset/managed_components/espressif__esp-dsp/modules/windows/blackman_harris/include -I/Users/crisqifawei/esp/esp-box/examples/usb_headset/managed_components/espressif__esp-dsp/modules/windows/blackman_nuttall/include -I/Users/crisqifawei/esp/esp-box/examples/usb_headset/managed_components/espressif__esp-dsp/modules/windows/nuttall/include -I/Users/crisqifawei/esp/esp-box/examples/usb_headset/managed_components/espressif__esp-dsp/modules/windows/flat_top/include -I/Users/crisqifawei/esp/esp-box/examples/usb_headset/managed_components/espressif__esp-dsp/modules/iir/include -I/Users/crisqifawei/esp/esp-box/examples/usb_headset/managed_components/es [... cut ...] er. For more information run 'idf.py docs -sp api-guides/build-system.html'.
Also, please check if the header file has been removed, renamed or relocated - refer to the migration guide for more information.
ninja failed with exit code 1, output of the command is in the /Users/crisqifawei/esp/esp-box/examples/usb_headset/build/log/idf_py_stderr_output_35331 and /Users/crisqifawei/esp/esp-box/examples/usb_headset/build/log/idf_py_stdout_output_35331
(base) crisqifawei@CRISQIFAWEI-MC0 usb_headset %

ESP-Mars, 2023-06-30:
Hi, if you don't have any questions, please close this issue. Thank you!

lijunru-hub, 2023-06-30:
你的 lvgl 目录下面有文件吗？大概率是没有拉取子模块。
使用 git submodule update --init --recursive 拉取子模块

closed 2023-09-04 by ESP-Mars