espressif/esp-box #68: Feature request - Matter

a-d-r-i-a-n-d, 2023-04-17:
Hi there,

I think the ESP32-S3-BOX could be an excellent device if it supported Matter. Imagine being able to commission and control all your Matter-compatible devices from one centralized hub - that would be fantastic! I realize this could be a significant change in terms of demand and the types of buyers, as it would appeal not only to developers and enthusiasts, but also to regular consumers.

My question is: does the ESP32-S3-BOX have enough resources to function as a Matter server, controller, and commissioner?

Thank you for your consideration!

espressif2022, 2023-04-18:
> Hi there,
>
> I think the ESP32-S3-BOX could be an excellent device if it supported Matter. Imagine being able to commission and control all your Matter-compatible devices from one centralized hub - [...]

Thanks for your suggestion.
 We have a Matter project internally, and it will be released in the near future.

a-d-r-i-a-n-d, 2023-04-18:
Great news, looking forward to it.

ESP-Mars, 2023-08-17:
@a-d-r-i-a-n-d Hi we have release a matter switch demo for user to enjoy! https://github.com/espressif/esp-box/tree/master/examples/matter_switch

closed 2023-09-04 by ESP-Mars