espressif/esp-box #69: Cannot update firmware

nicoolaro, 2023-04-19:
I just received ESPBox today, at the begining following user guide it syas to opne "network" settings , bu i had no netowrk setings no matter where i clicked on the screen. So i thought to update firmware, i followed guide to update to v0.30 by using ESPLaunchpad, Launchpad stuck on 7% ... i rebooted and tried again Launchpad stuck at 1%

```
esptool.js
Serial port WebSerial VendorID 0x303a ProductID 0x1001
Connecting....
Detecting chip type... ESP32-S3
Chip is ESP32-S3
Features: Wi-Fi,BLE
Crystal is 40MHz
MAC: 7c:df:a1:e7:9b:c0
Uploading stub...
Running stub...
Stub running...
Changing baudrate to 921600
Changed
Not changing the image
Compressed 14876672 bytes to 8457110...
Writing at 0x0... (0%)
Writing at 0x2062d... (0%)
Writing at 0x2a3da... (0%)
Writing at 0x38b74... (0%)
Writing at 0x4373e... (0%)
Writing at 0x4d001... (1%)
```

Then i tried windows Flash tools and it did not work, as soon as i click on START button changes to FINISHED, ESP BOX is not on black screen.

I also tried boot mode by holding BOOT button while plugging USB cable,still black screen,

Can somone help setting the device?

he88, 2023-05-13:
我与你遇到了相同的问题。请问你解决了吗？

nicoolaro, 2023-05-13:
> 我与你遇到了相同的问题。请问你解决了吗？

I used esptool-v3.3.3-win64 to flash

he88, 2023-05-14:
> 

I also solved this problem by using the third-party library esptools in Python and successfully flashed the firmware in CMD

ESP-Mars, 2023-06-30:
Hi, if you don't have any questions, please close this issue. Thank you!

closed 2023-06-30 by nicoolaro