espressif/esp-box #72: claim init failed. please reboot device

meeech, 2023-05-17:
Hi

Received device. Is nice. Doing setup. Updated the firmware to 0.3. That went well. However, when using the app on iphone to try to setup, when I get to step to try to connect to wifi, i always get message "claim init failed. please reboot device". Any advice please?

HelloFOFO, 2023-05-25:
I got the same error.

At first, I used iPhone X  +  iOS 16.1.1   + ESP Box V1.1.0.
I tried to clear the network info but failed again and again.

Then I used Android(Huawei Mate 30) and succeeded.

espressif2022, 2023-06-02:
> @ESP-Mars phone is latest iphone se. ios 16.3.1. ESP Box V1.1.0
> 
> Unfortunately, I dont have an android to work around the problem 🗡️

After Provisioning started end.  Adding the following steps , it can temporarily fix this issue.
`ble_gap_set_prefered_default_le_phy(BLE_HCI_LE_PHY_1M_PREF_MASK, BLE_HCI_LE_PHY_1M_PREF_MASK);`

dustins, 2023-06-12:
Same issue. iPhone 11, iOS 16.5, ESP Box v1.1.0

RobG-git, 2023-06-12:
Same issue: ESP32-S3-BOX 01.01 with firmware v0.3.0 from ESP-BOX app on iPhone 11 IOS 16.5.  I also tried flashing the box with firmware v0.2.1 but same claim init issue.

Please could you explain about the text below:

> After Provisioning started end. Adding the following steps , it can temporarily fix this issue.
> `ble_gap_set_prefered_default_le_phy(BLE_HCI_LE_PHY_1M_PREF_MASK, BLE_HCI_LE_PHY_1M_PREF_MASK);`

espressif2022, 2023-06-13:
> Same issue. ESP32-S3-BOX was already on FW version 0.3 when I tried to set it up.
> 
> iPhone 11, iOS 16.5, ESP Box v1.1.0

can you try build the fw version of master. and try again?

espressif2022, 2023-06-13:
> Same issue: ESP32-S3-BOX 01.01 with firmware v0.3.0 from ESP-BOX app on iPhone 11 IOS 16.5. I also tried flashing the box with firmware v0.2.1 but same claim init issue.
>
> Please could you explain [...]

> Same issue: ESP32-S3-BOX 01.01 with firmware v0.3.0 from ESP-BOX app on iPhone 11 IOS 16.5. I also tried flashing the box with firmware v0.2.1 but same claim init issue.
>
> Please could you explain [...]

I may have made a mistake. 
I tried to reproduce this issue in the master version, and it appeared. So I modify it here, and it's alright. 
Later, I couldn't reproduce this problem in the master version, even without the above changes, only appeared in 0.3.0.

RobG-git, 2023-06-13:
I have now downloaded esp-idf and esp-box, done "idf.py build flash monitor" for the factory demo, and the device now connects with the ESP-BOX app on my iPhone 11 and is connected to WiFi. When I look at the "About us" screen on the device it says "Software Ver: v0.4.0" (NOTE: version four) and "ESP-IDF Ver: v5.2-dev-1010-g213504238"

But when I look in the ESP-BOX app under Device Settings it says "Firmware ver: v0.3.0-75-g7e5256f"

I then ran "python3 -m esptool --chip esp32s3 write_flash 0x0 Downloads/ESP-BOX_Demo_EN_V0.3.0.bin", removed the device from the ESP-BOX app and tried to add it again and got "Claim init failed..." In the "About us" screen the device reports "Software Ver: v0.3.0" (NOTE: version three) and "ESP-IDF Ver: v4.4-264-g2bdea81b2a"

I then did "idf.py flash monitor" for the factory demo and the ESP-BOX app connected with no problem. But it shows thre devices (light, switch, fan) whilst the device shows four (light, switch, fan, air).  The app shows the version as v0.3.0-75... whilst the device shows the version as v0.4.0. The app shows "IP address: --" and "Config Time: 2020-03-20"

Please may you update the ESP-BOX_Demo_EN_V0.3.0.bin with a more recent version of the binary. Please may you update the "factory demo" example code and ESP-BOX app so that they show the same number of devices, and the same version of the firmware, and the app displays the IP address and has the right Config Time?

keith721, 2023-06-17:
Received my two ESP32-S3-BOX devices from Digi-Key today.  Experienced the failure to connect 2.4GHz wifi, same messages: claim init failed. please reboot device.  Happened twice, looking for updated firmware to flash.

espressif2022, 2023-06-19:
Due to recent certificate adjustments, the IOS APP can't add devices. 
We updated last week and are ready to release.

ESP-Mars, 2023-06-30:
@keith721 @RobG-git @dustins @HelloFOFO Hey friends, we have solved the titled issue, but here is a packaging error during the process of uploading APP to Apple Store, we have to wait for the official solution from Apple Developer (https://developer.apple.com/forums/thread/727564). Until then, please use ESP-BOX for Android to experience the function. Thank you for your patience and understanding.

keith721, 2023-07-03:
It appears a fix/solution/workaround was posted by user vadimwe on the first page:
```
Workaround is to update all the generated ...-frameworks.sh files to add the -f flag to the call to readlink. In other words, need to replace:

    source="$(readlink "${source}")"

with

    source="$(readlink -f "${source}")"

Seems this is CocoaPods issue...

ESP-Mars, 2023-07-11:
> It appears a fix/solution/workaround was posted by user vadimwe on the first page:
>
> ```
> Workaround is to update all the generated ...-frameworks.sh files to add the -f flag to the call to readlink. [...]

The bug has been fixed, please update your ESP-BOX from App Store.

ESP-Mars, 2023-07-12:
Thank you all for your testing. I will now close this issue. If you have any further questions or concerns, feel free to reopen it.

closed 2023-07-12 by ESP-Mars

DheeranYuvaraj, 2023-08-01:
@ESP-Mars 

Same error in ESP-BOX App in Android version 13.