espressif/esp-box #76: 【求助】关于ESP-DL

fatsheep2020, 2023-05-27:
DL的官方安装和使用例子是基于linux的：https://github.com/espressif/esp-dl/tree/master/examples/human_face_detect
有没有基于windows的，还是只能在linux下使用？
谢谢！

alibukharai, 2023-06-14:
> DL的官方安装和使用例子是基于linux的：https://github.com/espressif/esp-dl/tree/master/examples/human_face_detect 有没有基于windows的，还是只能在linux下使用？ 谢谢！
The provided example should be compatible with both Linux and Windows operating systems, However If you could specify the required commands, I would be able to provide you with more details.

fatsheep2020, 2023-06-15:
hi, Ali:
Thanks for your answer!  i prefer to use it in windows, since my IDF was run in windows. could you give some guide?

fatsheep2020, 2023-06-17:
it can works in my windows VSCode! thanks for your reminder!

closed 2023-06-17 by fatsheep2020