espressif/esp-box #78: 求助，安装MicroPython以后”嘀“，”嘀“不停

jack7927, 2023-06-06:
我在thonny上配置MicroPython解释器，下载ESP32S3芯片bin
擦除以后
开始报错，一连接ESP32-S3-Box，电脑音箱嘀嘀响个不停，更换了电脑，数据线都响
我看了Thonny上的错误日志：

11:17:36.299 ERROR   thonny.plugins.micropython.mp_back: Exception in MicroPython main method
Traceback (most recent call last):
  File "C:\Program Files (x86)\Thonny\lib\site-packages\thonny\plugins\micropython\mp_back.py", line 139, in __init__
    self._process_until_initial_prompt(
  File "C:\Program Files (x86)\Thonny\lib\site-packages\thonny\plugins\micropython\bare_metal_backend.py", line 268, in _process_until_initial_prompt
    self._process_output_until_active_prompt(
  File "C:\Program Files (x86)\Thonny\lib\site-packages\thonny\plugins\micropython\bare_metal_backend.py", line 844, in _process_output_until_active_prompt
    self._interrupt()
  File "C:\Program Files (x86)\Thonny\lib\site-packages\thonny\plugins\micropython\mp_back.py", line 337, in _interrupt
    self._write(INTERRUPT_CMD)
  File "C:\Program Files (x86)\Thonny\lib\site-packages\thonny\plugins\micropython\mp_back.py", line 466, in _write
    return self._connection.write(data)
  File "C:\Program Files (x86)\Thonny\lib\site-packages\thonny\plugins\micropython\serial_connection.py", line 89, in write
    size = self._serial.write(data)
  File "C:\Program Files (x86)\Thonny\lib\site-packages\serial\serialwin32.py", line 325, in write
    raise SerialTimeoutException('Write timeout')
serial.serialutil.SerialTimeoutException: Write timeout

求助，应该怎么解决？

ESP-Mars, 2023-06-06:
建议通过 u0 rx 和 u0 tx 连接 [TTL 转usb 模块](https://www.google.com/search?q=TTL+%E8%BD%ACusb&sxsrf=APwXEde_5EhsPjmtR5XoiCyGCAvL6qNSTg:1686046243330&source=lnms&tbm=isch&sa=X&ved=2ahUKEwi3gKLbs67_AhX-ilYBHduqDp4Q_AUoAXoECAEQAw&biw=1440&bih=688&dpr=2#imgrc=m-iDbbAstkkf4M)，通过串口烧录程序。

当前你是通过 USB 去烧录，芯片被擦除后，Flash 内无代码 USB 重复连接导致 ‘电脑音箱嘀嘀’ 是[正常现象](https://espressif-docs.readthedocs-hosted.com/projects/espressif-esp-faq/zh_CN/latest/software-framework/peripherals/usb.html#esp32-c3-esp32-s3-usb-serial-jtag-controller-pc-usb-usb)。建议提前烧录稳定的固件。

jack7927, 2023-06-06:
非常感谢！您提到的“通过 u0 rx 和 u0 tx 连接TTL 转usb 模块”，是否有什么文档或者资料我能进一步学习？再次感谢您的解惑！

jack7927, 2023-06-06:
这个问题解决了，非常感谢！

closed 2023-06-06 by jack7927