espressif/esp-box #81: box changed color in sunlight

mars3142, 2023-06-13:
The ESP-Box is on my desk and the sun is shining into my room directly. Now after some time, the box changed color. This isn't good. What's the material? It should get a revision ;)

[image]
[image]

ESP-Mars, 2023-06-14:
Thanks for your feedback, the observed discoloration issue with this material we will consider making improvements in the future. This material is photosensitive resin, suitable for general product casing verification and common assembly parts, offering a high cost-performance ratio.

ESP-Mars, 2023-06-30:
Hi, if you don't have any questions, please close this issue. Thank you!

closed 2023-07-11 by ESP-Mars