espressif/esp-box #82: First page is empty at startup for M5Stack CoreS3 ESP32S3

lboue, 2023-06-14:
First page is empty for M5Stack CoreS3 ESP32S3.
What is the purpose of having an empty page?

[image]

imliubo, 2023-06-14:
Hi @lboue , I open the issue on CoreS3 repo, so you can ask there. And the hardware is a little different, so we stiil need do  some work on this. I will check this. Thanks for report. :)

lboue, 2023-06-14:
> Hi @lboue , I open the issue on CoreS3 repo, so you can ask there. And the hardware is a little different, so we stiil need do some work on this. I will check this. Thanks for report. :)

Thanks

lboue, 2023-06-14:
@imliubo I am sorry but I can't find your issue here:
1. https://github.com/m5stack/CoreS3-Box/issues
2. https://github.com/m5stack/M5CoreS3/issues

imliubo, 2023-06-14:
Related discussion moved to [here](https://github.com/m5stack/CoreS3-Box/issues/1).

ESP-Mars, 2023-06-30:
Close the issue since it moved to another external project.

closed 2023-06-30 by ESP-Mars