espressif/esp-box #83: [need help] about ESP-DL error

fatsheep2020, 2023-06-16:
when running ESP-DL example as guide: https://github.com/espressif/esp-dl/blob/master/docs/zh_CN/get_started.md
when run command: idf.py flash monitor, system build and flash, after successful, reset system there will be error logs from monitor terminal:
Rebooting...
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0xc (RTC_SW_CPU_RST),boot:0xa (SPI_FAST_FLASH_BOOT)
Saved PC:0x40375bec
0x40375bec: esp_restart_noos at C:/Users/rzlw3b/esp/esp-idf/components/esp_system/port/soc/esp32s3/system_internal.c:151 (discriminator 1)

SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3810,len:0x17b4
load:0x403c9700,len:0xd7c
load:0x403cc700,len:0x300c
.entry 0x403c992c
I (24) boot: ESP-IDF v5.0.2 2nd stage bootloader
I (24) boot: compile time 10:33:50
I (24) boot: chip revision: v0.1
I (26) qio_mode: Enabling default flash chip QIO
I (31) boot.esp32s3: Boot SPI Speed : 80MHz
I (36) boot.esp32s3: SPI Mode       : QIO
I (41) boot.esp32s3: SPI Flash Size : 4MB
I (45) boot: Enabling RNG early entropy source...
I (51) boot: Partition Table:
I (54) boot: ## Label            Usage          Type ST Offset   Length
I (62) boot:  0 factory          factory app      00 00 00010000 003b6000
I (69) boot:  1 nvs              WiFi data        01 02 003c6000 00004000
I (77) boot:  2 fr               Unknown data     01 06 003ca000 00020000
I (84) boot: End of partition table
I (88) esp_image: segment 0: paddr=00010020 vaddr=3c050020 size=30614h (198164) map
I (127) esp_image: segment 1: paddr=0004063c vaddr=3fc92700 size=028c0h ( 10432) load
I (129) esp_image: segment 2: paddr=00042f04 vaddr=40374000 size=0d114h ( 53524) load
I (143) esp_image: segment 3: paddr=00050020 vaddr=42000020 size=4e9fch (322044) map
I (192) esp_image: segment 4: paddr=0009ea24 vaddr=40381114 size=01560h (  5472) load
I (200) boot: Loaded app from partition at offset 0x10000
I (200) boot: Disabling RNG early entropy source...
E (212) quad_psram: PSRAM ID read error: 0x00ffffff, PSRAM chip not found or not supported, or wrong PSRAM line mode
E (213) cpu_start: Failed to init external RAM!

abort() was called at PC 0x4037551d on core 0
0x4037551d: call_start_cpu0 at C:/Users/rzlw3b/esp/esp-idf/components/esp_system/port/cpu_start.c:394 (discriminator 3)

Backtrace: 0x40375f7e:0x3fceb220 0x4037aa5d:0x3fceb240 0x40380566:0x3fceb260 0x4037551d:0x3fceb2d0 0x403cdb56:0x3fceb330 0x403cdf5d:0x3fceb380 0x403c9995:0x3fceb4b0 0x40045c01:0x3fceb570 |<-CORRUPTED
0x40375f7e: panic_abort at C:/Users/rzlw3b/esp/esp-idf/components/esp_system/panic.c:423

0x4037aa5d: esp_system_abort at C:/Users/rzlw3b/esp/esp-idf/components/esp_system/esp_system.c:153

.0x40380566: abort at C:/Users/rzlw3b/esp/esp-idf/components/newlib/abort.c:38

0x4037551d: call_start_cpu0 at C:/Users/rzlw3b/esp/esp-idf/components/esp_system/port/cpu_start.c:394 (discriminator 3)
[image]

espressif2022, 2023-06-16:
Which module is yours? Specifically, choose "Octal Mode PSRAM" for R8 series, and "Quad Mode PSRAM" for others.

fatsheep2020, 2023-06-17:
ok, thanks!

closed 2023-06-17 by fatsheep2020