espressif/esp-box #84: audio output distorted in usb_headset example

mjhewitt1, 2023-06-21:
When using the example 'usb_headset' (unchanged) the audio output is distorted (sounds like it is skipping/cutting out). Any similar issues/ideas how to fix this? 
Many thanks

leeebo, 2023-06-29:
Hi @mjhewitt1 Would you please try with the prebuilt binary from [esp-box launchpad](https://espressif.github.io/esp-launchpad/?flashConfigURL=https://raw.githubusercontent.com/espressif/esp-box/master/launch.toml), choose usb_headset then flash to give it a try

mjhewitt1, 2023-07-03:
Hi, thanks for your response. I have given it a go and still sounds the same unfortunately

lijunru-hub, 2023-07-05:
Are you using a windows computer? What is the version of windwos

mjhewitt1, 2023-07-05:
Ah so, I have tried on a Windows 10 computer and it works! But no luck on Ubuntu 20.04

mjhewitt1, 2023-07-05:
Hi there, I have fixed the issue :) 

I used the esp-box launchpad site to connect and flash the USB headset example to the ESP-BOX. The audio output was fine when I did this on a Windows PC, but not on Ubuntu. 

I fixed the issue with the audio output on Ubuntu by adding the following lines to /etc/pulse/daemon.conf : 
default-sample-rate = 48000
alternate-sample-rate = 48000

Which was suggested here -- https://askubuntu.com/questions/1399384/ubuntu-audio-crackling-using-logitech-usb-audio-even-after-formatting 

Thanks a lot for your responses

closed 2023-07-05 by mjhewitt1