espressif/esp-box #85: esp-box-lite is experiencing an I2C issue, could it be due to the touchscreen problem? However, the provided sample README indicates that it supports the ESP-Box-lite.

ZeusFunk, 2023-06-23:
the error is like：
`E (768) lcd_panel.io.i2c: panel_io_i2c_rx_buffer(135): i2c transaction failed
E (768) TT21100: esp_lcd_touch_tt21100_read_data(173): I2C read error!
E (768) TT21100: esp_lcd_touch_new_i2c_tt21100(103): TT21100 init failed
E (768) TT21100: Error (0xffffffff)! Touch controller TT21100 initialization failed!
I (768) gpio: GPIO[3]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0
ESP_ERROR_CHECK failed: esp_err_t 0xffffffff (ESP_FAIL) at 0x42008102
0x42008102: bsp_display_indev_init at C:/Users/GCLuis/Desktop/esp-box-new/examples/image_display/managed_components/espressif__esp-box/esp-box.c:306 (discriminator 1)

file: "./managed_components/espressif__esp-box/esp-box.c" line 306
func: bsp_display_indev_init
expression: bsp_touch_new(((void *)0), &tp)`
and my environment is configured according to the latest settings which is under the ESP-IDF master and ESP-Box master environments.

espressif2022, 2023-06-25:
The latest master is not supported yet.
We implement BSP support for box_lite internally, and wait for a detailed test and it will be released. 

ESP-Box-lite does not have TP, the error reported here is caused by calling the BSP of ESP-Box

ZeusFunk, 2023-06-25:
ok thanks

closed 2023-06-25 by ZeusFunk