espressif/esp-box #86: Request to integrate ChatGPT Example from ESPBox?

Mohit-Ak, 2023-07-07:
Hi,
There is a very interesting example that was added to the ESP-BOX demo and almost looks like a flagship example for it. I was wondering if we can implement the same in Core-S3?

ESP-BOX URL
https://github.com/espressif/esp-box/tree/master/examples/chatgpt_demo

I tried comparing the changes made in this repo for other examples and tried replicating it but with no success yet especially with the microphone sensor. Can we have this as an issue or possibly a PR please?

espressif2022, 2023-07-10:
> Hi, There is a very interesting example that was added to the ESP-BOX demo and almost looks like a flagship example for it. I was wondering if we can implement the same in Core-S3?
>
> ESP-BOX URL [...]

What is your situation? Did not  read a valid recording audio, or can't drive the microphone ?

Mohit-Ak, 2023-07-10:
Apologies, I wanted to raise the ticket in https://github.com/m5stack/CoreS3-Box which is a fork of this repo but lacks this example.

closed 2023-07-10 by Mohit-Ak