espressif/esp-box #87: How to set data for ref channel？

RealNee, 2023-07-20:
In the factory demo example, the speech recognition part,
  [app_sr.c](https://github.com/espressif/esp-box/blob/master/examples/factory_demo/main/app/app_sr.c)   audio_feed_task(void *arg):
```
        /* Channel Adjust */
        for (int  i = audio_chunksize - 1; i >= 0; i--) {
            audio_buffer[i * 3 + 2] = 0;
            audio_buffer[i * 3 + 1] = audio_buffer[i * 2 + 1];
            audio_buffer[i * 3 + 0] = audio_buffer[i * 2 + 0];
        }
```
Why set the data of the third channel to 0？
In the [sr document](https://docs.espressif.com/projects/esp-sr/en/latest/esp32s3/audio_front_end/README.html#afe-dual-mic), the third channel should be the REF channel, how should this channel be set?

espressif2022, 2023-07-20:
The third channel is used for AEC (`afe_config.aec_init = false`), We didn't get the desired effect, so we didn't enable this function.

RealNee, 2023-07-20:
> The third channel is used for AEC (`afe_config.aec_init = false`), We didn't get the desired effect, so we didn't enable this function.

If I want to use AEC, how should the data of the REF channel be set up, is it copied from one of the recording channels?

espressif2022, 2023-07-20:
> > The third channel is used for AEC (`afe_config.aec_init = false`), We didn't get the desired effect, so we didn't enable this function.
> 
> If I want to use AEC, how should the data of the REF channel be set up, is it copied from one of the recording channels?

No, It's meaningless to copy the data of these two channels.

The third channel is the hardware acquisition channel, which is to filter out the audio signal generated by itself during playback.
The logic is that the sound recorded delete the sound played by itself is the effective audio input of the environment. You can refer to the hardware schematic diagram.

ZorkyZorky, 2023-08-15:
我想在box上增加回声消除，但是现在唤醒都不能使用了。做了以下修改：
1. ret |= es7210_write_reg(ES7210_SDP_INTERFACE2_REG12, 0x02);
2..bits_per_sample        = I2S_BITS_PER_SAMPLE_32BIT  //在bsp_i2s.c，I2S_CONFIG_DEFAULT()
3.afe_config.aec_init = true;
4.i2s_read(I2S_NUM_0, feed_buffer, chunksize * 4 * sizeof(int16_t),
            &bytes_read, portMAX_DELAY);//读取4个通道
5.    for (int i = 0; i < chunksize; i++) {
        int16_t ref = feed_buffer[4 * i + 0];
        feed_buffer[3 * i + 0] = feed_buffer[4 * i + 1];
        feed_buffer[3 * i + 1] = feed_buffer[4 * i + 3];
        feed_buffer[3 * i + 2] = ref;
    }//adjust channel

还有其他需要修改的地方吗？

ZorkyZorky, 2023-08-15:
我想在box上增加回声消除，但是现在唤醒都不能使用了。做了以下修改：
1. ret |= es7210_write_reg(ES7210_SDP_INTERFACE2_REG12, 0x02);
2..bits_per_sample        = I2S_BITS_PER_SAMPLE_32BIT  //在bsp_i2s.c，I2S_CONFIG_DEFAULT()
3.afe_config.aec_init = true;
4.i2s_read(I2S_NUM_0, feed_buffer, chunksize * 4 * sizeof(int16_t),&bytes_read, portMAX_DELAY);//读取4个通道
5.for (int i = 0; i < chunksize; i++) {
        int16_t ref = feed_buffer[4 * i + 0];
        feed_buffer[3 * i + 0] = feed_buffer[4 * i + 1];
        feed_buffer[3 * i + 1] = feed_buffer[4 * i + 3];
        feed_buffer[3 * i + 2] = ref;
    }//adjust channel

还有其他需要修改的地方吗？

espressif2022, 2023-08-15:
> 我想在box上增加回声消除，但是现在唤醒都不能使用了。做了以下修改：
>
> 1. ret |= es7210_write_reg(ES7210_SDP_INTERFACE2_REG12, 0x02);
>    2..bits_per_sample        = I2S_BITS_PER_SAMPLE_32BIT  //在bsp_i2s.c，I2S_CONFIG_DEFAULT() [...]

bits_per_sample  是不变的

ZorkyZorky, 2023-08-16:
但是在skainet里bits_per_sample是32bit，而且skainet换成16bit就唤醒不了了

ZorkyZorky, 2023-08-21:
查看了一下es7210的user guide，里面看到如果只用一根数据线传输两个麦克风和回声通道的值需要启用TDM，看skainet里对es7210的寄存器配置是启用的1XFS TDM的形式，但是ESP32S3端的I2S好像没有配置TDM，也就是#if SOC_I2S_SUPPORTS_TDM,请问这个需要设置吗？是在哪里设置？

RealNee, 2023-08-21:
> 我想在box上增加回声消除，但是现在唤醒都不能使用了。做了以下修改：
>
> 1. ret |= es7210_write_reg(ES7210_SDP_INTERFACE2_REG12, 0x02);
>    2..bits_per_sample        = I2S_BITS_PER_SAMPLE_32BIT  //在bsp_i2s.c，I2S_CONFIG_DEFAULT() [...]

[image]
在box中好像只有3个通道是有数据的，及1.2.3通道，第四个通道是悬空的，第三通道是反馈的通道。至于怎么开启和读取这个通道数据还没有试验过.

ESP-Mars, 2023-11-09:
Thank you all for your testing. I will now close this issue. If you have any further questions or concerns, feel free to reopen it.

closed 2023-11-09 by ESP-Mars