espressif/esp-box #89: USB连接问题

undefinedcodezhong, 2023-08-28:
Windows10系统，esp32s3box-lite 使用USB连接后显示未识别设备，设备管理器的端口哪里也没有显示COM端口，应该怎么处理？

espressif2022, 2023-08-28:
> Windows10系统，esp32s3box-lite 使用USB连接后显示未识别设备，设备管理器的端口哪里也没有显示COM端口，应该怎么处理？

烧录的哪个 demo ？

undefinedcodezhong, 2023-08-28:
ESP-BOX-Lite_Demo_CN_VO_5_0这个demo
在线固件烧录工具 ESP Launchpad的显示情况
[image]
设备管理器界面：
[image]
linux系统也找不到串口设备
[image]

alibukharai, 2023-08-28:
@undefinedcodezhong I recommend attempting to enter download mode manually by holding down the Boot button and simultaneously pressing the EN button.

espressif2022, 2023-08-28:
前面有烧写 usbheadset 之类占用 USB 的固件？

undefinedcodezhong, 2023-08-28:
> @undefinedcodezhong I recommend attempting to enter download mode manually by holding down the Boot button and simultaneously pressing the EN button.

Thank you for your reply. I have tried the manual mode. It doesn't work. I think the problem is my PC could not recognize the USB device, rather than the inability to download firmware via USB.

undefinedcodezhong, 2023-08-28:
@espressif2022 这个不确定，因为是淘的板子，别人开发一半不做了，我接了板子想搞自己的程序。我尝试按照这里的[Windows系统更新固件方法](https://github.com/espressif/esp-box/blob/master/docs/firmware_update_cn.md#windows-%E7%B3%BB%E7%BB%9F%E6%9B%B4%E6%96%B0%E5%9B%BA%E4%BB%B6%E8%AF%B4%E6%98%8E)，不过使用usb2uart模块来代替typeC，但是卡在了“等待上电同步”这里了。

设备管理器能看到usb2uart的com口：
[image]
flash_download_tool下载图示：
[image]

undefinedcodezhong, 2023-08-28:
@espressif2022 换了台电脑，还是用USB转串口的方式，现在能下载固件了，但是连接USB不能正确识别的问题还是存在，这个有没有解决方案，毕竟有USB开发也会方便很多，串口烧录程序总是麻烦一些。

espressif2022, 2023-08-29:
> @espressif2022 换了台电脑，还是用USB转串口的方式，现在能下载固件了，但是连接USB不能正确识别的问题还是存在，这个有没有解决方案，毕竟有USB开发也会方便很多，串口烧录程序总是麻烦一些。

我看你这识别的是 USB 设备，不是串口设备。USB转串口，你这怎么接的？

undefinedcodezhong, 2023-08-29:
@espressif2022 最后是换了一个CH340 的USB转TTL串口，通过uart下载的，能成功下载固件的那次忘记保留端口设备截图了。Type-C直连识别不了，会显示未识别设备。

undefinedcodezhong, 2023-08-29:
@espressif2022 ok了，换了条线，终于能typeC直连了

closed 2023-08-30 by undefinedcodezhong