espressif/esp-box #91: can box connect with USB device (u-disc)? how to do that?

fatsheep2020, 2023-08-28:
the question as title, i want save data into USB storage.
thanks!

espressif2022, 2023-08-29:
you can refer to this: https://github.com/espressif/esp-dev-kits/tree/master/esp32-s3-lcd-ev-board/examples/usb_msc_file_sys

alibukharai, 2023-08-29:
@fatsheep2020 Additionally, you can explore this example implementation that utilizes TinyUF2. https://github.com/espressif/esp-box/tree/master/examples/chatgpt_demo/factory_nvs

fatsheep2020, 2023-08-30:
> 

thanks!  but what used is Arduino environment, is there any example on that?

ESP-Mars, 2023-09-04:
> thanks! but what used is Arduino environment, is there any example on that?

USB has not been ported to the Arduino environment yet.

ESP-Mars, 2023-11-09:
We are temporarily closing the issue, and any updates related to this requirement will be communicated here. If you have further questions or concerns, please feel free to reopen the issue.

closed 2023-11-09 by ESP-Mars