espressif/esp-box #92: [Help]how to download/upload file from SPIFFS

fatsheep2020, 2023-08-31:
when i want upload many files to SPIFFS and download it to check, how to do that?

Thanks!

espressif2022, 2023-09-01:
> when i want upload many files to SPIFFS and download it to check, how to do that?
> 
> Thanks!

Can you give a more detailed description of the requirements?
What interface, USB, wifi?

fatsheep2020, 2023-09-04:
yes, the communication interface is usb or wifi, both are ok. just use SPIFFS(the flash file system in Box) to save files, and also can use PC to download check the files content.
Thanks!

espressif2022, 2023-09-04:
> yes, the communication interface is usb or wifi, both are ok. just use SPIFFS(the flash file system in Box) to save files, and also can use PC to download check the files content. Thanks!

You can refer to this two demos in ESP-IDF:
Wi-Fi: esp-idf/examples/protocols/http_server/file_serving
USB: esp-idf/examples/peripherals/usb/device/tusb_msc
If you want to verify whether the file is correct, can't you do a local MD5 verification in the box?

ESP-Mars, 2023-11-09:
Temporarily closed due to the extended period of time with no response. Should you have additional questions or concerns, don't hesitate to reopen the issue.

closed 2023-11-09 by ESP-Mars