espressif/esp-box #94: 编译espbox-lite 工程的mp3 modle出现下面的错误怎么解决？

zhoaji, 2023-09-03:
I (218) gpio: GPIO[3]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:2 
E (218) lcd_panel.io.i2c: panel_io_i2c_rx_buffer(135): i2c transaction failed
E (218) TT21100: esp_lcd_touch_tt21100_read_data(198): I2C read error!
E (218) TT21100: esp_lcd_touch_new_i2c_tt21100(110): TT21100 init failed
E (218) TT21100: Error (0xffffffff)! Touch controller TT21100 initialization failed!
I (218) gpio: GPIO[3]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0

espressif2022, 2023-09-04:
Use menuconfig select the right board, the path is as follows:
`"TOP-> HMI Board Config-> ESp32-S3-BOX-Lite"`

ESP-Mars, 2023-11-09:
Temporarily closed due to the extended period of time with no response. Should you have additional questions or concerns, don't hesitate to reopen the issue.

closed 2023-11-09 by ESP-Mars