espressif/esp-box #95: esp32 S3 BOX在编译下载chatgpt的demo后，touch初始化报错重启

MichaelDu9226, 2023-09-04:
esp32 S3 BOX在编译下载chatgpt的demo后，touch初始化报错重启，非BOX 3
LOG如下：

[10:38:42.701]收←◆[0;31mE (1529) lcd_panel.io.i2c: panel_io_i2c_rx_buffer(135): i2c transaction failed[0m
[0;31mE (1535) TT21100: esp_lcd_touch_tt21100_read_data(198): I2C read error![0m
[0;31mE (1542) TT21100: esp_lcd_touch_new_i2c_tt21100(110): TT21100 init failed[0m
[0;31mE (1549) TT21100: Error (0xffffffff)! Touch controller TT21100 initialization failed![0m
[0;32mI (1558) gpio: GPIO[3]| InputEn: 0| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0 [0m
ESP_ERROR_CHECK failed: esp_err_t 0xffffffff (ESP_FAIL) at 0x42039993
file: "./components/espressif__esp-box/esp-box.c" line 424
func: bsp_display_indev_init
expression: bsp_touch_new(((void *)0), &tp)

[10:38:42.775]收←◆
abort() was called at PC 0x40380def on core 0

Backtrace: 0x40375f82:0x3fcac1a0 0x40380df9:0x3fcac1c0 0x40388656:0x3fcac1e0 0x40380def:0x3fcac250 0x42039993:0x3fcac280 0x420399e5:0x3fcac2d0 0x4200a0cf:0x3fcac310 0x42118aa7:0x3fcac330

espressif2022, 2023-09-04:
首先不是 esp-box-lite ？
断电重启下看下，TP 正常的情况下，可能是 IIC 总线死锁了

MichaelDu9226, 2023-09-04:
确定是 esp box，21年12月份购买的
我将与touch相关的BSP_NULL_CHECK(bsp_display_indev_init(disp), NULL);注释以后，编译下载可以联网，但是仍然报错：app_sr: AFE Fetch Fail

espressif2022, 2023-09-04:
> 

1：先试一下，[出厂固件](https://espressif.github.io/esp-launchpad/?flashConfigURL=https://raw.githubusercontent.com/espressif/esp-box/master/launch.toml)

2：因为这一路 IIC 总线挂着触摸还有 语音相关的 codec adc 器件，从 log 是否能看出这两边的有无初始化错误？
3：还有，你可以在器件执行初始化前，先扫描下这一路总线下的 器件地址，看下能找到几个设备？

MichaelDu9226, 2023-09-04:
BOX的出厂固件测试没问题，触摸也是可以使用的，所以感觉是固件相关的问题
我是按照https://github.com/espressif/esp-box/tree/master/examples/chatgpt_demo redme，执行的git checkout 53ff7d43dbff642d831a937b066ea0735a6aca24 && git pull && git submodule update --init --recursive
想确认一下BOX有不同的touch ic吗？

espressif2022, 2023-09-04:
> BOX的出厂固件测试没问题，触摸也是可以使用的，所以感觉是固件相关的问题 我是按照https://github.com/espressif/esp-box/tree/master/examples/chatgpt_demo redme，执行的git checkout 53ff7d43dbff642d831a937b066ea0735a6aca24 && git pull && git submodule update --init --recursive 想确认一下BOX有不同的touch ic吗？

所有 TP 只有一个地址，是同一个批次的 `#define ESP_LCD_TOUCH_IO_I2C_TT21100_ADDRESS (0x24)`
上述操作流程没问题，注意下这个是在 ESP-IDF 的仓库中操作的，不是 esp-box 仓库。

你用 factory_demo 编译烧录也是 OK 的？还有如果你是拉取的最新代码？
这次发布我们在 组件 `espressif__esp-box`  加了个 `BSP_LCD_SUB_NAME` 的补丁，看下是否选择了 `esp32-s3-box-3`

MichaelDu9226, 2023-09-04:
确实是BSP_LCD_SUB_NAME的问题，修改为esp32-s3-box-3即可使用，谢谢。
另外还有关于中英选择，及tts合成结果超时的问题，我提问在另外一个问题里了，还请帮忙解答一下，谢谢
https://github.com/espressif/esp-box/issues/96

espressif2022, 2023-09-04:
好的，如果问题已经解决，麻烦关闭 ISSUE

closed 2023-09-04 by MichaelDu9226

lomon-emerson, 2023-12-08:
> 确实是BSP_LCD_SUB_NAME的问题，修改为esp32-s3-box-3即可使用，谢谢。 另外还有关于中英选择，及tts合成结果超时的问题，我提问在另外一个问题里了，还请帮忙解答一下，谢谢 #96

我也是这个问题，BSP_LCD_SUB_NAME是在factory_nvs的idf.py menuconfig进行设置，还是在里面的json文档里面？谢谢。